earlephilhower/arduino-pico #564: Setting configSUPPORT_STATIC_ALLOCATION to 1 results in a compile error

ogatatsu, 2022-04-29:
Setting configSUPPORT_STATIC_ALLOCATION in FreeRTOSConfig.h to 1 will result in a compile error.

```
C:\Users\ogata\Documents\ArduinoData\packages\rp2040\hardware\rp2040\2.0.0\libraries\FreeRTOS\src\variantHooks.cpp:317:6: error: conflicting declaration of C function 'void vApplicationGetIdleTaskMemory(StaticTask_t**, StackType_t**, uint16_t*)'
  317 | void vApplicationGetIdleTaskMemory( StaticTask_t ** ppxIdleTaskTCBBuffer,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\ogata\Documents\ArduinoData\packages\rp2040\hardware\rp2040\2.0.0\libraries\FreeRTOS\src\task.h:1,
                 from C:\Users\ogata\Documents\ArduinoData\packages\rp2040\hardware\rp2040\2.0.0\libraries\FreeRTOS\src\variantHooks.cpp:41:
c:\users\ogata\documents\arduinodata\packages\rp2040\hardware\rp2040\2.0.0\libraries\freertos\lib\freertos-kernel\include\task.h:1861:10: note: previous declaration 'void vApplicationGetIdleTaskMemory(StaticTask_t**, StackType_t**, uint32_t*)'
 1861 |     void vApplicationGetIdleTaskMemory( StaticTask_t ** ppxIdleTaskTCBBuffer,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\ogata\Documents\ArduinoData\packages\rp2040\hardware\rp2040\2.0.0\libraries\FreeRTOS\src\variantHooks.cpp:336:6: error: conflicting declaration of C function 'void vApplicationGetTimerTaskMemory(StaticTask_t**, StackType_t**, uint16_t*)'
  336 | void vApplicationGetTimerTaskMemory( StaticTask_t ** ppxTimerTaskTCBBuffer,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\ogata\Documents\ArduinoData\packages\rp2040\hardware\rp2040\2.0.0\libraries\FreeRTOS\src\timers.h:1,
                 from C:\Users\ogata\Documents\ArduinoData\packages\rp2040\hardware\rp2040\2.0.0\libraries\FreeRTOS\src\variantHooks.cpp:42:
c:\users\ogata\documents\arduinodata\packages\rp2040\hardware\rp2040\2.0.0\libraries\freertos\lib\freertos-kernel\include\timers.h:1358:10: note: previous declaration 'void vApplicationGetTimerTaskMemory(StaticTask_t**, StackType_t**, uint32_t*)'
 1358 |     void vApplicationGetTimerTaskMemory( StaticTask_t ** ppxTimerTaskTCBBuffer,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

It seems that the types of `pulIdleTaskStackSize` and `pulTimerTaskStackSize` are different.

earlephilhower, 2022-04-29:
With the Arduino code there are lots of dynamic allocations, so the static allocation was not tried nor supported.  If it's important top you, a PR would be much appreciated!  But, again, there are tons of dynamic allocations so static task allocation don't really make sense here...

earlephilhower, 2022-04-29:
Also, how did you actually set that `#define`?  If you set it in your code it will not be set while compiling the FreeRTOS core.  You'd need to hack the actual FreeRTOS source files in the core to have any chance of having it work.

ogatatsu, 2022-04-30:
I checked by editing `FreeRTOSConfig.h` directly, and PR the fix for the compile error.

earlephilhower, 2022-05-01:
Fixed by #566

closed 2022-05-01 by earlephilhower