earlephilhower/arduino-pico #567: Multicore-FreeRTOS.ino example function setup1 and loop1 not started

janczeresnia, 2022-04-30:
Hi
I try to run the example Multicore-FreeRTOS.ino but setup1 and loop1 never run. PS shows that Core1 is always Ready. Never Running. What am I doing wrong?

The same is the case with the example of Multicore.ino until I add the rp2040.restartCore1 command to the setup, the setup1 and loop1 functions do not start.

earlephilhower, 2022-04-30:
Are you running with `picoprobe`?    #402 notes an issue when running thru the debugger.  I think that particular thing is out of my control and needs something done to openOCD to get it working...

janczeresnia, 2022-04-30:
Yes a use picoprobe to uploading sketches.

janczeresnia, 2022-04-30:
Thanks as I loaded the program directly, everything works great :)

closed 2022-04-30 by janczeresnia