earlephilhower/arduino-pico #571: Picoprobe upload issues through platformio

tkanarsky, 2022-05-02:
Hi! Trying to setup my platformio workflow: target Pico connected via USB for power, other Pico flashed with latest Picoprobe firmware connected to SWD headers on target Pico. 

I'm having issues with uploading via picoprobe, so I haven't gotten to testing the debugging part. 

My platformio.ini is as follows:

```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 1m
lib_deps = 
	; adafruit/Adafruit SSD1306@^2.5.3
	; adafruit/Adafruit BusIO@^1.11.4
	; jandrassy/WiFiEspAT@^1.3.2
	https://github.com/tkanarsky/Pico-PIO-USB.git
platform_packages =
    maxgerhardt/framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
	maxgerhardt/toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.3-a/x86_64-linux-gnu.arm-none-eabi-ed6d983.220212.tar.gz
    platformio/tool-openocd-raspberrypi@https://gitee.com/flyboard2040/pio-openocd-picoprobe-linux.git
debug_tool = picoprobe
upload_protocol = picoprobe
monitor_speed = 115200
``` 

The other parts of the platformio.ini, other than the line ` platformio/tool-openocd-raspberrypi@https://gitee.com/flyboard2040/pio-openocd-picoprobe-linux.git` and `debug_tool/upload_protocol`, work fine and result in properly uploaded firmware. I wanted to bite the bullet and set up a proper debugger because printf'ing everything was driving me mad.

I get the following output when starting an upload:

```
Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; board: pico; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.6.0+sha.d20e2ed) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (picoprobe) External (cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico 1.20000.0+sha.55394a7 
 - tool-mklittlefs 1.203.210628 (2.3) 
 - tool-openocd-raspberrypi 2.1100.0+sha.af6aec9 
 - tool-rp2040tools 1.0.2 
 - toolchain-pico 5.100300.220212 (10.3.0)
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium size: 1044480 Flash Length: 1044480 EEPROM Start: 270528512 Filesystem start 269479936 Filesystem end 270528512
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 19 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Pico-PIO-USB> 0.0.0+20220502022230.sha.1d60795
Building in release mode
Compiling .pio/build/pico/src/usb/test.cpp.o
Linking .pio/build/pico/firmware.elf
Generating UF2 image
elf2uf2 ".pio/build/pico/firmware.elf" ".pio/build/pico/firmware.uf2"
Retrieving maximum program size .pio/build/pico/firmware.elf
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium size: 1044480 Flash Length: 1044480 EEPROM Start: 270528512 Filesystem start 269479936 Filesystem end 270528512
Checking size .pio/build/pico/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   4.2% (used 11044 bytes from 262144 bytes)
Flash: [=         ]   7.2% (used 75652 bytes from 1044480 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picoprobe, picotool, raspberrypi-swd
CURRENT: upload_protocol = picoprobe
Uploading .pio/build/pico/firmware.elf
Open On-Chip Debugger 0.11.0+dev-g0628da7-dirty (2021-10-29-19:23)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

Warn : Transport "swd" was already selected
adapter speed: 5000 kHz

target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
** Programming Started **
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
Warn : no flash bank found for address 0x20000000
Warn : no flash bank found for address 0x20010e40
** Programming Finished **
** Verify Started **
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
Error: checksum mismatch - attempting binary compare
embedded:startup.tcl:542: Error: ** Verify Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 603
at file "embedded:startup.tcl", line 542
*** [upload] Error 1
================================================ [FAILED] Took 2.96 seconds ================================================
The terminal process "platformio 'run', '--target', 'upload'" terminated with exit code: 1.
```
It seems like it's having issues finding the proper offset to write the firmware to, which causes the checksum to fail

I'm not well-versed in this toolchain, so apologies in advance if the error is completely unrelated to your efforts. If so, could you point me in the right direction?

Thanks in advance!

tkanarsky, 2022-05-02:
Okay, so I've been tweaking the openocd command and it seems the following works:

```
./openocd -d2 -s /home/tim/.platformio/packages/tool-openocd-raspberrypi/share/openocd/scripts -f interface/picoprobe.cfg -f target/rp2040.cfg -c "program {firmware.elf} verify reset; shutdown;" 
Open On-Chip Debugger 0.11.0+dev-g0628da7-dirty (2021-10-29-19:23)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselect 'swd'
Warn : Transport "swd" was already selected
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x10000001
Info : rp2040.core0: hardware has 4 breakpoints, 2 watchpoints
Info : rp2040.core1: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
** Programming Started **
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
Info : Found flash device 'win w25q16jv' (ID 0x001540ef)
Info : RP2040 B0 Flash Probe: 2097152 bytes @0x10000000, in 32 sectors

Info : Padding image section 1 at 0x10012840 with 192 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x10012900 .. 0x1001ffff
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
** Programming Finished **
** Verify Started **
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
** Verified OK **
** Resetting Target **
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
shutdown command invoked
```
I removed the `0x10000000` from the openocd command. I will try to find where in the build script the command is called, and edit it out. What's the root cause for that offset (I assume) being there?

tkanarsky, 2022-05-02:
Looks like the platform wants the firmware image to start at 0x10000000.

```
  if "uploadfs" in COMMAND_LINE_TARGETS:
        # filesystem upload. use FS_START.
        openocd_args.extend([
            "-c", "program {$SOURCE} ${hex(FS_START)} verify reset; shutdown;"
        ])
    else:
        # normal firmware upload. flash starts at 0x10000000
        openocd_args.extend([
            "-c", "program {$SOURCE} %s verify reset; shutdown;" %
            board.get("upload.offset_address", "0x10000000")
        ])
    openocd_args = [
        f.replace("$PACKAGE_DIR", platform.get_package_dir(
            "tool-openocd-raspberrypi") or "")
        for f in openocd_args
    ]
    # use ELF file for upload, not bin (target_firm). otherwise needs
    # offset 0x10000000
    #upload_source = target_elf
    env.Replace(
        UPLOADER="openocd",
        UPLOADERFLAGS=openocd_args,
        UPLOADCMD="$UPLOADER $UPLOADERFLAGS")
    if not board.get("upload").get("offset_address"):
        upload_source = target_elf
    upload_actions = [env.VerboseAction("$UPLOADCMD", "Uploading $SOURCE")]
```
Why is that? It runs just fine when uploading to the default offset.

tkanarsky, 2022-05-02:
ahh, maybe the offset is necessary only if uploading a raw binary? looks like the comment says the offset is not necessary when flashing an ELF

tkanarsky, 2022-05-06:
Root cause not in this repo, closing

closed 2022-05-06 by tkanarsky

dlbogdan, 2022-05-06:
Glad I could help.

tkanarsky, 2022-05-06:
For future reference: here is a minimal working platformio project with Picoprobe upload and debug: https://github.com/tkanarsky/pico-arduino-platformio-debug-template