earlephilhower/arduino-pico #575: General purpose clock I/O library and usage examples

brycecherry75, 2022-05-07:
I noted that in the RP2040 Datasheet, Section 1.4.3 "GPIO Functions" has CLOCK GPIN(0-1) for external counting/frequency measurement and CLOCK GPOUT(0-3) for generating external clock signals which I among certain others would find useful and therefore looking for a library and usage examples.
Usage is described in section 2.15 "Clocks" and code for an RP2040-based frequency counter is at https://github.com/ingbird64/Pi-Pico_FrequencyCounter/blob/main/src/freq-simple.c

earlephilhower, 2022-05-08:
@brycecherry75 it looks like it's relatively simple, why not try making a PR or even a completely separate library for it?  Looking at the pico-sdk sources you have probably 80% of the work already there...

earlephilhower, 2022-08-06:
This really doesn't seem like it's going to be added, so closing for now.  If someone wants to submit a PR adding this feature, I'd be very happy to have it, of course!

closed 2022-08-06 by earlephilhower