earlephilhower/arduino-pico #582: Add PIO USB support

sfranzyshen, 2022-05-14:
Add support for https://github.com/sekigon-gonnoc/Pico-PIO-USB

with the latest pull request to Pico-PIO-USB ... TinyUSB support has been added
https://github.com/sekigon-gonnoc/Pico-PIO-USB/pull/22 

and there is a active pull request to TinyUSB  itself ...
https://github.com/hathach/tinyusb/pull/1412

earlephilhower, 2022-05-15:
As in #575, a PR would be appreciated.  Although I think that in this case it might make sense to make a separate Arduino library so you could use if here or in the MBED version.  I'm not really seeing any changes to the core needed for this at first glance.

sfranzyshen, 2022-05-17:
I guess the changes really should be applied to the Adafruit_TinyUSB_Arduino project and then updated here ...

kilograham, 2022-05-20:
I added a PR https://github.com/hathach/tinyusb/pull/1467

sfranzyshen, 2022-06-16:
A new release of Adafruit_TinyUSB_Arduino has been released with "1.11.0 - Updated tinyusb to support rp2040 pio usb host"
https://github.com/adafruit/Adafruit_TinyUSB_Arduino/releases/tag/1.11.0

It's not clear to me how the Adafruit_TinyUSB_Arduino library is pulled into this project  ... or if any other changes would be needed to update it within this project ... 

So I don't know what would be needed to create a pull-request for this ... support?

sfranzyshen, 2022-06-19:
sweet! thank you

earlephilhower, 2022-06-19:
Let's wait on the PR merge before closing...

closed 2022-06-19 by earlephilhower

sfranzyshen, 2022-06-24:
any chance we can get a bump (update) again on the Adafruit_TinyUSB_Arduino again? Not much has changed but an [DualRole](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/examples/DualRole/device_info_rp2040) example has been added ...

PR #639 (2022-06-19): Update to latest Adafruit_TinyUSB
Fixes #582
commit 803184c1 (2022-06-19): Update to latest Adafruit_TinyUSB (#639)

Fixes #582