earlephilhower/arduino-pico #585: i2c slave .onReceive() failure/error when 'restart' rather than 'stop' is transmitted by i2c master

DWiskow, 2022-05-19:
Wire.cpp does not generate a "Wire.onReceive()" interrupt for an i2c Slave following transmission of a 'restart' rather than a 'stop' by an i2c Master. This situation can occur frequently when a master sends a register address followed by a read request for the contents of that register.

This is particularly important in a Multi Master i2c system, where the Master should not allow another Master to obtain control of the bus between its write of the required register and the read of the register content [and therefore explicitly uses a 'restart' rather than a 'stop' follow by a 'start' between it's write and read over the bus.

I have written a short Arduino C++ program to illustrate this issue [i2cmemory.ino -(https://github.com/earlephilhower/arduino-pico/files/8733993/i2cmemory.ino.txt)]

The error is that wire.cpp in arduino-pico/libraries/Wire/src only checks the IC_INTR_STAT register bit 10 (R_START_DET - a 'start' condition) to generate it's interrupt . . . it should also check and act on bit 12 (E_RESTART_DET - i.e. a 'restart' condition)

The attached image shows the required changes to wire.cpp to correc tthis [image]

DWiskow, 2022-05-19:
The i2cmemory.ino code, referenced in my previous post, needs to be compiled and uploaded to a Raspberry Pi Pico and GPIO 20 & 21 (SDA & SCL) connected to a suitable master device [e.g. a Raspberry Pi].

i2cTools under the Pi-OS can then be used to reliably repeat the issue . . . and ultimately prove the proposed fix . . . an example of this is shown in the attached text file 
[i2cTools .txt](https://github.com/earlephilhower/arduino-pico/files/8734071/i2cTools.txt)

Optionally this can also be demonstrated using the attached python program which is designed to run on a Raspberry Pi  
[i2cTest.py.txt](https://github.com/earlephilhower/arduino-pico/files/8734078/i2cTest.py.txt)

DWiskow, 2022-05-19:
> Very good debug. Why not submit a PR since you seem to have a tested fix?

I would . . . but I am not very familiar with github/PR process :-)

It surprises me that this has not been raised before  given that "i2c.writeto_then_readfrom" is the preferred approach in python [rather than  i2c.writeto with stop=false] . . . I guess not that many people are writing i2c slave programs using your arduino-pico library/implementation, yet.

earlephilhower, 2022-05-19:
No worries about the pull request.  If you could eyeball #586 and see that it matches your proven good code then we're all set.  Thx!

DWiskow, 2022-05-19:
Code matches, all correct . . . that’s a very quick fix for this issue 😇

closed 2022-05-19 by earlephilhower

DWiskow, 2022-05-20:
updated i2cTools.txt & i2cTest.py.txt [as typo's present in both files]

[i2cTools .txt](https://github.com/earlephilhower/arduino-pico/files/8736946/i2cTools.txt)
[i2cTest.py.txt](https://github.com/earlephilhower/arduino-pico/files/8736948/i2cTest.py.txt)

ashishagarwal14, 2023-03-27:
I was using older version, and no knowing this issue has kept me up for 2 weeks, finally landed here! thanks for the debug.

PR #586 (2022-05-19): Fix Wire I2C restart handling
Fixes #585

On an I2C bus restart, call the onReceive callback and clear the buffer.

Thanks to @DWiskow for the debug and patch!
commit a3a5abac (2022-05-19): Fix Wire I2C restart handling (#586)

Fixes #585

On an I2C bus restart, call the onReceive callback and clear the buffer.

Thanks to @DWiskow for the debug and patch!