earlephilhower/arduino-pico #590: EEPROM.begin(size) seems not to work within a eeprom wrapper class

elral, 2022-05-23:
We have a project working with Arduinos under PlatformIO, I transferred this to use on a raspberry pico. Thanks to this work only very less changes were required.
But I am wondering about one topic. We have set up a wrapper class for some eeprom functionalities, and within the constructor the eeprom is is initialized with EEPROM.begin(size) (which is not required for the Arduinos). Unfortenutely this seems not to work as I do not get back any valid data which is stored in the eeprom. I have to use a seperate init() function within my class and have to call this function within my setup(). Then everything is working fine.

Is there any reason to do it this way?

Thanks and regards

Ralf

earlephilhower, 2022-05-23:
See https://arduino-pico.readthedocs.io/en/latest/eeprom.html

There is no EEPROM on the Pico so it's simulated.  You need to use EEPROM.commit to actually write to flash.  But a better way would probably be to use LittleFS.

earlephilhower, 2022-05-23:
Ah, okay.  Can you please make an MCVE so I can see what you're doing and reproduce the failure simply?

elral, 2022-05-24:
Sorry for asking, what is MCVE?
I will prepare a small program to reproduce this "failure".

elral, 2022-05-25:
I have setup a working variant under: https://github.com/elral/eeprom_test

Within the branch https://github.com/elral/eeprom_test/tree/class I have moved the eeprom functions into a class like we use in our project (https://github.com/elral/MobiFlight-FirmwareSource/blob/additionalHW_support/src/MF_Modules/MFEEPROM.cpp).

The interesting point now is, in our program read and write is working. Here in this branch only reading in working!?
Uncommenting `#define WORKING` leads to a version where reading and writing does not work like in our project.

I hope it is small enough, otherwise I will try to reduce it further.

earlephilhower, 2022-05-25:
That looks good, thanks.  I'll take a look at it this week.  Might be something related to the ctor() initialization during the runtime startup, but need to trace through to see for certain.

elral, 2022-05-25:
It is not urgent and definitely out of my knowledge ;) 
Thanks a lot!

closed 2022-05-25 by earlephilhower