earlephilhower/arduino-pico #598: Raspberry pi PİCO Board SP1 for SD Card

Auxut, 2022-05-27:
Hi,
Im beginner at coding, so please be patient with me.
I have a raspberry pi pico board, i need to use both of the SPI connections.(SPI0 and SPI1)
I have 2 modules first one is a RF module and second one is the sd card adapter.
Both of the modules are working perfectly if they connected to SPI0 seperately but i want to use both at the same board.
I want to use RF module at SPI0 and SD card at SPI1. But I cannot manage to initialize SD card with using SP1.
I found this post here: https://github.com/earlephilhower/arduino-pico/issues/290 
And got confused, RP pico has already SPI1 pins defined at pin definitions header file. 

I believe i need to do this definitions to make it work with SPI1:

#define SDCARD_SPI SP1
#define SDCARD_SS_PIN 13 
(And probably rx,tx,sck pin definitions too?)

The problem is I got confused and i dont know which driver header i need to add these definitions.
I only tried to run example codes given in this repo, so im not including any of them here.
If this is the wrong place to ask this question im sorry(im also new to github), feel free to warn me and i remove this post (if i can).

Thanks in advance, Have good day/night/afternoon.

M.D.Çerçi

Auxut, 2022-06-02:
Problem solved with editing sdFatConfig.h file.
I added defines for the all SPI1 pins in to that.

closed 2022-06-02 by Auxut

NatanielSlo, 2022-12-08:
If i can ask, did you use the built in SD library or the SDfat one? I have the same issue so if you could share your code i would be very thankfull. I have no clue how to use the SD card on the second SPI.