earlephilhower/arduino-pico #612: Update PlatformIO builder script + docs

maxgerhardt, 2022-06-07:
Commits
* https://github.com/earlephilhower/arduino-pico/commit/420d669a4b82d5fdd2eeab4a856b6eec6929ea45
* https://github.com/earlephilhower/arduino-pico/commit/e3053f69f824398b3d615648d5e35ff244bf8378

require an update of docs in regards to 
* update toolchain link downloads for the new version
* how to activate stack protector

require an update of the builder script in regards to
* correctly linking against `-lstdc++` or `-lstdc++-exc` depending on whether exceptions are enabled (`-fexceptions` found in build flags)

earlephilhower, 2022-06-07:
Argh, sorry for breaking your build.  I need to do a Platform.IO job as part of CI. :(

maxgerhardt, 2022-06-07:
Pretty sure it did not break the build (e.g., [`-lstdc++`](https://github.com/earlephilhower/arduino-pico/blob/master/tools/platformio-build.py#L96-L99) is sitll linked against), we just need to expose the new features to PlatformIO projects too by expanding the builder script.

ivankravets, 2022-06-09:
Hi @maxgerhardt and @earlephilhower,

We have received 35 packages for review in the PlatformIO registry. What is the further fate of these packages?

https://github.com/platformio/platform-raspberrypi/issues/13 - Could you provide a PR with support for this core?

Thanks!

maxgerhardt, 2022-06-09:
Ideally we'd want to have the pico-toolchain, Raspberrypi's tool-openocd version and `framework-arduinopico` in the registry, then I can reference stable versions in https://github.com/maxgerhardt/platform-raspberrypi and create the PR.

ivankravets, 2022-06-09:
All packages are approved => https://registry.platformio.org/search?q=owner%3Aearlephilhower

Please tell us which packages are not required, we will remove them to keep the registry clean.

ivankravets, 2022-06-09:
Do we need a custom ARM toolchain and OpenOCD?

maxgerhardt, 2022-06-09:
Hmm I think I had problems with the tool-openocd-raspberrypi version tha PlatformIO had ([1 year old](https://registry.platformio.org/tools/platformio/tool-openocd-raspberrypi/versions)) when trying PicoProbe etc. support, so a newer build is needed.

And as far as I understand it, this core builds on custom configuration + features of the ARM toolchain, e.g. with exception code etc. (linking against `stdc++-exc` if exceptions are enabled etc), see e.g. [here](https://github.com/earlephilhower/pico-quick-toolchain/commits/master).

ivankravets, 2022-06-09:
Thanks for the provided information. We would be thankful for the PR. I've just done a quick look at your fork and it looks good. The only note is to use `env.Exit(1)` in SCons scripts where you need to stop the building process.

maxgerhardt, 2022-06-09:
@ivankravets PR created per above, more PRs for docs and builder script + doc update here will follow.

closed 2022-06-17 by earlephilhower