earlephilhower/arduino-pico #623: Platformio build not defaulting to -Os

jhmaloney, 2022-06-12:
I hope I am reporting this issue the right place...

I believe the convention for both the Arduino IDE and Platformio is to provide a default compiler optimization setting of -Os. Max had made that work but when I recently upgraded that no longer appears to be the case. (I noticed because the timing of my NeoPixel code changed.)

I'm working around the problem by adding:
`build_flags = -Os`
to the Pico entry in my platformio.ini file, but it would be great to make that the default so people get better code even if they don't specify any build_flags.

Here is the entry I am using:
`[env:pico-mac]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
board_build.filesystem_size = 1.5m
lib_ldf_mode = chain+
lib_ignore = WiFi
build_flags = -Os
board_build.core = earlephilhower
; Core package not yet registered with PlatformIO, so pull from git
; Note: Toolchain download link is specific to OS! See https://github.com/earlephilhower/pico-quick-toolchain/releases.
platform_packages =
	maxgerhardt/framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
	maxgerhardt/toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.3-a/x86_64-apple-darwin14.arm-none-eabi-ed6d983.220212.tar.gz
`

earlephilhower, 2022-06-12:
Already fixed and will be in the next release: https://github.com/earlephilhower/arduino-pico/pull/619

closed 2022-06-12 by earlephilhower

jhmaloney, 2022-06-12:
Quick work! :-) Thanks!