earlephilhower/arduino-pico #624: Platformio toolchain for M1 Mac Powerbook?

jhmaloney, 2022-06-12:
I am migrating from an Intel-based Mac Powerbook to a new M1-based Mac Powerbook Pro. I'm using the same version of Platformio (PlatformIO Core, version 6.0.2) and the same arduino-pico framework on both computers. The Pico build works on the the Intel Powerbook but not on M1 Powerbook. The error message reports "extra junk" at the end of one of the assembly files. Here is the output:

`Processing pico-mac (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; board: pico; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.3-a/x86_64-apple-darwin14.arm-none-eabi-ed6d983.220212.tar.gz
Unpacking  [####################################]  100%          
Tool Manager: toolchain-pico@5.100300.220212 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.7.0+sha.1b8624b) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.20003.0+sha.c352105 
 - tool-rp2040tools @ 1.0.2 
 - toolchain-pico @ 5.100300.220212 (10.3.0) 
 - toolchain-rp2040-earlephilhower @ 5.100300.220608 (10.3.0)
Converting vm.ino
Flash size: 2.00MB
Sketch size: 0.50MB
Filesystem size: 1.50MB
Maximium Sketch size: 520192 EEPROM start: 0x101ff000 Filesystem start: 0x1007f000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ soft
Found 14 compatible libraries
Scanning dependencies...
Dependency Graph
|-- LittleFS @ 0.1.0
|-- Servo(rp2040) @ 1.0.0
|-- SPI @ 1.0
|-- Wire @ 1.0
Building in release mode
Compiling .pio/build/pico-mac/FrameworkArduinoBootloader/boot2_w25q080_2_padded_checksum.S.o
Compiling .pio/build/pico-mac/src/dataPrims.c.o
Compiling .pio/build/pico-mac/src/filePrims.cpp.o
Compiling .pio/build/pico-mac/src/fileTransfer.cpp.o
Compiling .pio/build/pico-mac/src/interp.c.o
Compiling .pio/build/pico-mac/src/ioPrims.cpp.o
Compiling .pio/build/pico-mac/src/mem.c.o
Compiling .pio/build/pico-mac/src/miscPrims.c.o
Compiling .pio/build/pico-mac/src/netPrims.cpp.o
Compiling .pio/build/pico-mac/src/outputPrims.cpp.o
Compiling .pio/build/pico-mac/src/persist.c.o
Compiling .pio/build/pico-mac/src/persistFile.cpp.o
Compiling .pio/build/pico-mac/src/radioPrims.c.o
Compiling .pio/build/pico-mac/src/runtime.c.o
Compiling .pio/build/pico-mac/src/sensorPrims.cpp.o
Compiling .pio/build/pico-mac/src/serialPrims.cpp.o
Compiling .pio/build/pico-mac/src/tftPrims.cpp.o
Compiling .pio/build/pico-mac/src/tinyJSON.c.o
/var/folders/59/dvn8w71s54s1643_7pr2bx400000gn/T//cciLkwKo.s: Assembler messages:
/var/folders/59/dvn8w71s54s1643_7pr2bx400000gn/T//cciLkwKo.s:29: Error: junk at end of line, first unrecognized character is `n'
*** [.pio/build/pico-mac/src/outputPrims.cpp.o] Error 1
Compiling .pio/build/pico-mac/src/varPrims.c.o
====================================================== [FAILED] Took 7.28 seconds ======================================================

Environment    Status    Duration
-------------  --------  ------------
pico-mac       FAILED    00:00:07.279
================================================ 1 failed, 0 succeeded in 00:00:07.279 ================================================
`

Here is the entry I am using:
`[env:pico-mac]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
board_build.filesystem_size = 1.5m
lib_ldf_mode = chain+
lib_ignore = WiFi
build_flags = -Os
board_build.core = earlephilhower
; Core package not yet registered with PlatformIO, so pull from git
; Note: Toolchain download link is specific to OS! See https://github.com/earlephilhower/pico-quick-toolchain/releases.
platform_packages =
	maxgerhardt/framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
	maxgerhardt/toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.3-a/x86_64-apple-darwin14.arm-none-eabi-ed6d983.220212.tar.gz
`

The ArduinoIDE successfully compiles the same code. It is using this compiler:

~/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.3.3-a-ed6d983/bin/arm-none-eabi-g++

Should I be using a different toolchain?

earlephilhower, 2022-06-12:
Can you include the generated `.s` file so we can see exactly what the assembler is complaining about?

jhmaloney, 2022-06-12:
That would be super helpful, of course.

Unfortunately, the .s file is written into a temporary folder which is apparently later deleted by Platformio. I tried adding the -S switch to the build switches but that also did not work (unrelated issue). Is there another way to get the .s files?

M1 Macs run 64-bit Intel binaries using the Rosetta emulator so, in theory, if the assembler is a 64-bit executable the x86_64-apple-darwin14 toolchain should work.

It is interesting that the Arduino IDE is using pqt-gcc/1.3.3-a-ed6d983/bin/arm-none-eabi-g++. I'm not sure what the "pqt" abbreviation means-- perhaps "Python QT"?

earlephilhower, 2022-06-12:
If you can run PIO in verbose mode to get the actual commands executed, then you would be able to run them in another terminal and re-generate the output file.

At least one other user is building on M1 without issue, so it's kind of odd to see an issue in the assembler here.  Do you have any raw assembly in the source that is failing to build (i.e. an ``__asm`` block?)

The `pqt-` is `pico-quick-toolkit` which is the GCC 10.3 toolchain.

jhmaloney, 2022-06-12:
Thanks to your helpful  hints, I tracked the problem down to this nanosecond delay function:

`static inline void __not_in_flash_func(picoDelay(int n)) {
	volatile int dummy;
	for (int i = n; i > 0; i--) {
		dummy = i;
	}
}
`

The problem arises from the combination of "inline", "__not_in_flash_func()", and the volatile local variable. Removing the "__not_in_flash_func()" fixes the problem. That declaration isn't actually needed since the function is inlined anyhow. Oddly, this combination didn't cause any problems with the same platformIO toolchain on my Intel Mac so there is probably some subtle behavior difference in some system library function or perhaps in the Intel emulator...

This is a pretty obscure construct so I doubt others will run into this exact problem.

This isn't a bug in the Arduino-Pico framework, so sorry to bother you. But thank you for your help in isolating the problem.

I'm am excited that it looks like PlatformIO will officially supporting the framework. Heartfelt thanks to you and Max for all the hard work you've put into both the Arduino-Pico and the PlatformIO support for it. I use those tools multiple times a week, sometimes quite heavily... 

I really appreciate your efforts!

earlephilhower, 2022-06-13:
OK, great.  Looks like we can close this issue then.  Good luck w/the M1

closed 2022-06-13 by earlephilhower