earlephilhower/arduino-pico #626: Reason for SdFat8266 over SdFat

savejeff, 2022-06-13:
I'm currently converting to this core. 

I'm using the vanilla SdFat library but I'm getting compilation errors as the Core has a variant of the SdFat library inbuild. 
Is there a reason for the use of SdFat8266 over the vanilla SdFat library? 
As far as I can see the SdFat8266 just contains some backward compatibility hacks and fixes.

savejeff, 2022-06-13:
Ah okay thx for the Info. 

What do you mean by "not using the SdFat Lib directly" ? is this due to the integration into the Pico Core ? I'm using SdFat on the a daily basis and if found it to be the most reliable library for SD Card interaction out there

earlephilhower, 2022-06-13:
Use `SD.begin()` and `SD.open()` or `SDFS.begin()` and `SDFS.open()` and not any SdFat calls directly is what I mean.  You will still have the SDFat code running underneath, but it will work with the standard `File` we have for the core that's used in `LittleFS` (and `SPIFFS` in the ESP8266, too).  So you can easily change from SD to internal flash (or both if needed).

For example, the `SpeedTest` example in `LittleFS` can run on an SD card with a simple 1-line change thanks to this.