earlephilhower/arduino-pico #630: Problem using TinyUSB library with platformio

risq, 2022-06-15:
Hello,

First I'm sorry if that's an issue with platformio and not arduino-pico. But this was working before I switched to arduino-pico core and I can't find a way to make it work since.

Here is my `platformio.ini`file :
```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
framework = arduino
board = pico
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
platform_packages =
    maxgerhardt/framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
    maxgerhardt/toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.4-d/x86_64-apple-darwin14.arm-none-eabi-46ea946.220608.tar.gz
lib_compat_mode = off
lib_deps = 
	fortyseveneffects/MIDI Library@^5.0.2
	adafruit/Adafruit MPR121@^1.1.1
	adafruit/Adafruit TinyUSB Library@^1.10.3
```

I got the following issue when trying to build : 
```

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; framework: arduino; board: pico)
---------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.7.0+sha.8d7a063) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.20003.0+sha.3fb8cbf 
 - tool-rp2040tools @ 1.0.2 
 - toolchain-pico @ 5.100300.220608 (10.3.0) 
 - toolchain-rp2040-earlephilhower @ 5.100300.220608 (10.3.0)
Flash size: 2.00MB
Sketch size: 1.50MB
Filesystem size: 0.50MB
Maximium Sketch size: 1568768 EEPROM start: 0x101ff000 Filesystem start: 0x1017f000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ off
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- MIDI Library @ 5.0.2
|-- Adafruit MPR121 @ 1.1.1
|   |-- Adafruit BusIO @ 1.11.6
|   |   |-- Wire @ 1.0
|   |   |-- SPI @ 1.0
|   |-- Wire @ 1.0
|   |-- SPI @ 1.0
|-- Wire @ 1.0
Building in release mode

[...]

src/main.cpp:3:10: fatal error: Adafruit_TinyUSB.h: No such file or directory

**************************************************************************
* Looking for Adafruit_TinyUSB.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_TinyUSB.h"
* Web  > https://registry.platformio.org/search?q=header:Adafruit_TinyUSB.h
*
**************************************************************************

    3 | #include <Adafruit_TinyUSB.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/pico/src/main.cpp.o] Error 1
================================================== [FAILED] Took 3.76 seconds ==================================================
```

As you can see TinyUSB is not in the dependency graph and then header file can not be found.
I thought it was an issue with platformio thinking it was not compatible with pico board so I added `lib_compat_mode = off` in my platformio.ini file but it does not change anything.
Using platformio default pico config, this worked well and I could include and use TinyUSB.

Thank you very much for your help.

maxgerhardt, 2022-06-15:
I will have a look at this.

maxgerhardt, 2022-06-17:
You must activate TinyUSB with `build_flags` as stated in the docs: https://arduino-pico.readthedocs.io/en/latest/platformio.html#usb-stack

Also the core has the Adafruit Tiny USB built-in, try using that first. Use the `platformio.ini`:

```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
framework = arduino
board = pico
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
build_flags = -DUSE_TINYUSB
lib_compat_mode = off
lib_deps = 
	fortyseveneffects/MIDI Library@^5.0.2
	adafruit/Adafruit MPR121@^1.1.1
```
delete the `.pio` folder of the project prior to building again.

Based on your usage of `platform_packages` you also seem to be using an outdated version of the integration, please use the [CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and execute `pio platform update https://github.com/maxgerhardt/platform-raspberrypi.git` to update.

risq, 2022-06-19:
Thanks a lot, and sorry I missed it in the docs.
Is `platform_packages` not needed in my platformio.ini ?
Also I updated my platfom as you asked, and now it compiles properly.

But unfortunately I still can't make usb MIDI to work... The usb device is not recognized by my computer, and it looks that it can not pass Tinyusb initialization in the code.

Here is a simple example which works on platformio base raspberry pico platform : 
```cpp
#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

Adafruit_USBD_MIDI usb_midi;
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

void setup()
{
    // LED
    pinMode(LED_BUILTIN, OUTPUT);
    digitalWrite(LED_BUILTIN, LOW);

    // Manual begin() is required on core without built-in support for TinyUSB such as mbed rp2040
    TinyUSB_Device_Init(0);

    // Initialize MIDI, and listen to all MIDI channels
    // This will also call usb_midi's begin()
    MIDI.begin(MIDI_CHANNEL_OMNI);

    // wait until device mounted
    while (!TinyUSBDevice.mounted())
        delay(1);

    digitalWrite(LED_BUILTIN, HIGH);
    delay(500);
}

void loop()
{
    // Send Note On for current position at full velocity (127) on channel 1.
    MIDI.sendNoteOn(40, 127, 1);

    digitalWrite(LED_BUILTIN, HIGH);
    delay(2000);

    // Send Note Off for previous note.
    MIDI.sendNoteOff(40, 0, 1);

    digitalWrite(LED_BUILTIN, LOW);
    delay(4000);
}
```

with the following `platformio.ini` file : 
```
[env:pico]
platform = raspberrypi
board = pico
framework = arduino
lib_deps = 
	fortyseveneffects/MIDI Library@^5.0.2
	adafruit/Adafruit TinyUSB Library@^1.10.3
```

To switch to earlephilhower pico core, I updated my `platformio.ini` as you showed me : 
```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
build_flags = -DUSE_TINYUSB
lib_deps = 
	fortyseveneffects/MIDI Library@^5.0.2
```

After switching the device is not recognized anymore. The led never lights up so I assumed it is stuck during tinyusb initialization. Even when adding `Serial.begin(115200);` as stated in the docs to allow automatic upload I can not upload code again without rebooting the board pressing bootsel button.
I tried with and without `TinyUSB_Device_Init(0);` line because I'm not sure it's needed in this situation.

I also tried `midi_test` example from arduino IDE with earlephilhower board core, uploading via arduino IDE worked, using the same code from platformio I got the same issues, no compilation problem but device not recognized, not passing setup code, impossible to reupload without rebooting pressing bootsel button.

Thanks again for your help, and sorry if I missed something...

maxgerhardt, 2022-06-19:
One tiny thing before I try this: Can you add
```ini
lib_archive = no
```
to the `platformio.ini` and rebuild? There might an undocumented linker quirk here.

risq, 2022-06-19:
You are right, it worked. Thanks a lot.
First I tried with the code I showed you before and it did not worked, then I tried using midi_test example and it worked (was not working without `lib_archive = no`, I double checked).
I finally found out that my `TinyUSB_Device_Init(0);` line made it fail during initialization. Just remove it from my code and it worked again.

Thanks again for your help !

maxgerhardt, 2022-06-19:
Thanks for the confirmation, I will try and put this in the PlatformIO builder script so that the Adafruit TinyUSB library is automatically compiled with lib_archive = no if its in the detected libraries, users should not have to worry about these internals.

PR #658 (2022-06-26): Make TinyUSB linking work automatically with PIO
Closes #630.

As written in https://github.com/earlephilhower/arduino-pico/issues/630#issuecomment-1159775571, the TinyUSB library only works with PlatformIO during runtime if its object files are linked individually in the firmware instead of archiving them into a `.a` file and linking that.

So that users don't have to write
```ini
lib_archive = no
```
into their `platformio.ini` ([docs](https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-archive)) and to make the docs valid (they say only `build_flags = -DUSE_TINYUS [... cut]
closed 2022-06-26 by earlephilhower