earlephilhower/arduino-pico #632: Deadlock with Multithreading and ADC DMA

Zitt, 2022-06-16:
I'm attempting to port ADC DMA demos to earlephilhowe / arduino-pico but am experienced what appears to be a blocking hang when combined with https://github.com/Zitt/FastLED.

This simple project [
adc_fft.zip](https://github.com/earlephilhower/arduino-pico/files/8922064/adc_fft.zip)
runs the ADC as a dma transaction then supposed to run a https://github.com/mborgerding/kissfft to find the power of the audio via a microphone.  The port is from:
https://github.com/AlexFWulff/awulff-pico-playground

The symptom is that I see the loop() fire via "loop0()" being printed to the serial monitor. I also see loop1() fire every 500 ms. 
What I'm not seeing is the WS2812 led changing color. 
I see it changed from Red to Green at the start of the program so it fires at least twice. But then never changed to Blue or any other colors. 

I know the fastled code works because if I comment out the sample() call in loop() the leds function as expected.

Unsure how to debug as I haven't tried to debug on this microcontroller.

earlephilhower, 2022-06-16:
If you're seeing `loop0` and `loop1` printing out, then it's not a deadlock.  A deadlock would completely stop one or both cores.

I suggest breaking down your code into either the DMA read or the LED write and get them running separately.  Because you're not using the core for much (i.e. manually using the SDK `adc_`* routines, etc.) there's not much we can do in this repo.

Another suggestion would be to triple-check your DMA setup.  Make sure both ping and pong buffers are pointing to each other and are set to immediately begin once triggered from EOD on the other one.  The I2S code here has some examples of this (but for an N-entry loop).

Zitt, 2022-06-17:
@earlephilhower  - thanks.
I was able to do further debug and found that the original code had a bug where it wasn't using the global varable `dma_chan` in the setup routine. Fixing that I was able to get into the actual FFT routines.

However, now I'm seeing very clear hangs of Core1 after executing the 
kiss_fftr(kcfg , fft_in, fft_out);
routine a couple of times.  The attached revision [adc_fft.zip](https://github.com/earlephilhower/arduino-pico/files/8931199/adc_fft.zip) shows loop1() stops responding after the second execution of the `kiss_fftr` method.

output:
loop1()
every 500ms()
loop1()
loop1()
[INFO] C:\EE\microController\arduino-2.0.0rc6\Projects\awulff-pico-playground-main\adc_fft\adc_fft.ino:102 hello from setup()
loop1()
every 500ms()
loop1()
[INFO] C:\EE\microController\arduino-2.0.0rc6\Projects\awulff-pico-playground-main\adc_fft\adc_fft.ino:147 @40c pre kiss_fftr()
[INFO] C:\EE\microController\arduino-2.0.0rc6\Projects\awulff-pico-playground-main\adc_fft\adc_fft.ino:149 @40c post kiss_fftr()
Greatest Frequency Component: 200.0 Hz
[INFO] C:\EE\microController\arduino-2.0.0rc6\Projects\awulff-pico-playground-main\adc_fft\adc_fft.ino:147 @73c pre kiss_fftr()
[INFO] C:\EE\microController\arduino-2.0.0rc6\Projects\awulff-pico-playground-main\adc_fft\adc_fft.ino:149 @73c post kiss_fftr()
Greatest Frequency Component: 0.0 Hz

Any thoughts as to how to debug why loop1() stops firing?

earlephilhower, 2022-06-18:
You could have any number of issues, from code problems to core problems.  You should get `gdb` running with picoprobe.  It makes figuring out what happened in cases like this very easy!  See the readme or docs for info on running it.

earlephilhower, 2022-06-20:
Please try the latest git head (note you will need the updated toolchain, too, via get.py).  It includes big multicore stability fixes

Zitt, 2022-06-21:
I haven't tried the tip yet; just got setup with gdb and initially; this is what GDB said:
`Thread 2 received signal SIGTRAP, Trace/breakpoint trap.
[Switching to Thread 2]
isr_hardfault () at /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/src/rp2_common/pico_standard_link/crt0.S:98
98      /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/src/rp2_common/pico_standard_link/crt0.S: No such file or directory.`
that said; I'm a noob at gdb.

Zitt, 2022-06-21:
No joy with 2.2.0  as installed via the arduino 2.0rc6 IDE.
Anything compiled seems to create problem with USB device. 
Going back to 2.1.0 seems to fix whatever issue there is with the usb device.

Zitt, 2022-06-21:
I seem to have accidentally found a workaround.
I have to enable "stack protector" 
[image]
and then things seem to work better.

Zitt, 2022-07-01:
[earlephilhower](https://github.com/earlephilhower)
I'm not having any luck with this request.
Tonight I updated to 2.2.2 via the Arduino 2.0.0rc6 interface.
Compiling my code now results in a "usb device not found" with a yellow bang in device manager.

earlephilhower, 2022-07-01:
I think you have a corrupted install somehow.

As of now, the multithreaded stuff is all good in release 2.2.2 (and a few prior) , so I suggest removing every trace and reinstalling from the IDE or GIT from scratch.  To debug your crash, you really do need to get PicoDebug and GDB running.  They will tell you exactly where the core is at, and give you tools to single step through the app from the beginning thru the failure.

Zitt, 2022-07-01:
IF I had a corrupted install; then why would 2.1.0 work where 2.2.x doesn't?
IE to get to working state; all I had to do was "install" 2.1.0 from the 2.0.0rc6 menu.

[image]

Not trying to be aggressive; just pointing out that there is something amiss with the 2.2.x code bases which seems to be breaking the serial port implementation.

And yes; When I initially had problems with 2.2.0 after your request; I did delete the entire 
C:\Users<username>\AppData\Local\Arduino15\packages\rp2040
folder and re-download.  If you really think it will help; I can try again.

earlephilhower, 2022-07-01:
> Not trying to be aggressive; just pointing out that there is something amiss with the 2.2.x code bases which seems to be breaking the serial port implementation.

2.2.x has been out for a bit without other folks hitting this, even under Platform.IO.  So, I think something got weird on your install.  It happens, no worries.  The serial port for upload and the onboard USB serial stuff hasn't been touched except to make `Serial::write` return the right value 2 days ago (and it was verified by the user who found the problem.)

Also, one thing hits me now.  If you used the MSFT store to install the IDE, uninstall it and instead use the https://Arduino.cc download version.  The packaged version doesn't seem to allow any COMx: port access at all "for secutiry" but basically breaking it for any real use.

Zitt, 2022-07-01:
this install is from arduino direct; I only use windows store for Microsoft specific installs.
I'll try to duplicate with rc8 as a clean download/install.

closed 2022-07-21 by earlephilhower