earlephilhower/arduino-pico #637: Sleep on core 1 seems to negatively impact execution on core 0

JonnyHaystack, 2022-06-17:
Hi, I am experiencing odd behaviour when using delayMicroseconds() or sleep_us() on core 1. It seems to somehow interrupt the execution on core 0. I'm not entirely sure in what way it's interrupting it. I was observing that signals that were being sent by core 0 were sometimes cutting out half way, or sometimes nothing was being sent at all in response to received commands (context: communicating with a Nintendo GameCube console). At first I thought I must be doing something that isn't thread safe, but after commenting out everything else that I was doing from core1, I realised that the problem was caused simply by doing a sleep on core 1.

I made this minimal example to test and demonstrate the problem:
```
#include <Arduino.h>

bool gp1 = false;

void setup() {
  pinMode(1, OUTPUT);
}

void loop() {
  gp1 = !gp1;
  digitalWrite(1, gp1);
}

void setup1() {}

void loop1() {
  delayMicroseconds(50);
}
```

My platformio.ini:
```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi
framework = arduino
board = pico
board_build.core = earlephilhower
```

I'm just toggling GPIO pin 1 on and off as fast as possible on core 0, and doing 50us sleeps on core 1. Logic analyzer trace below shows the problem. Every time core 1 comes out of a sleep, core 0 seems to stall(?) for a while.

[image]

I also tested turning another GPIO pin on before the sleep (on core 1) and off after it. The result seems to indicate the effect on core 0 happens slightly before it comes out of sleep.

[image]

Code used for second test:
```
#include <Arduino.h>

bool gp1 = false;

void setup() {
  pinMode(1, OUTPUT);
}

void loop() {
  gp1 = !gp1;
  digitalWrite(1, gp1);
}

void setup1() { pinMode(2, OUTPUT); }

void loop1() {
  digitalWrite(2, 1);
  delayMicroseconds(50);
  digitalWrite(2, 0);
}
```

This behaviour seems very wrong, and from looking at [cores/rp2040/main.cpp](https://github.com/earlephilhower/arduino-pico/blob/master/cores/rp2040/main.cpp) I can't understand what would cause this. It's unfortunately making multicore pretty unusable for me so I'm hoping there's a simple solution.

JonnyHaystack, 2022-06-17:
Appears to be reproducible even on pure Pico SDK
```
#include <pico/multicore.h>
#include <pico/stdlib.h>

bool gp1;

void blah();

int main(void) {
    gpio_init(1);
    gpio_init(2);
    gpio_set_dir(1, GPIO_OUT);
    gpio_set_dir(2, GPIO_OUT);

    multicore_reset_core1();
    multicore_launch_core1(blah);

    while (true) {
        gp1 = !gp1;
        gpio_put(1, gp1);
    }
}

void blah() {
    while (1) {
        gpio_put(2, 1);
        sleep_us(50);
        gpio_put(2, 0);
    }
}
```

earlephilhower, 2022-06-17:
If it shows in the plain SDK, you should really open up an issue there https://github.com/raspberrypi/pico-sdk/issues .  We just use the SDK here whenever we can, so if there's an underlying problem, it needs to be solved upstream.

Looking at the SDK code all I see is a busywait on the core which should not affect anything on the other core.  Could you have an analog issue (i.e. power droop or some interaction w/the logic probe)?  I know my own no-name 8-bit clone "logic analyzer" sometimes has issues with noise on its inputs or even the phase of the moon. 😭

JonnyHaystack, 2022-06-17:
It's definitely not a noise issue, it has been causing me real problems and I only brought out the logic analyzer so I could do testing and provide evidence of it. Yeah I'll open an upstream issue, sorry I should probably have tested with the SDK alone before opening an issue here. Though I am pretty surprised that no one else has noticed this considering the scope of the problem is that wide.

Andy2No, 2022-06-18:
@JonnyHaystack Have you checked what happens in your example if you swap the roles of the cores?  I'm under the impression that core0 has to deal with interrupts that core1 doesn't.

As a possible workaround, instead of delaying for 50us you could try looping while comparing micros() to an end time.  Something like:

```

  utime = micros();

Andy2No, 2022-06-18:
@JonnyHaystack Discussion #583 looks like it might be relevant (on why things slow down, or don't keep consistent timing), and @earlephilhower  gave the link for the SDK documentation there (https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf).  

On page 30 of that PDF, there's a section called "3.1.4. I/O Using software control of GPIOs ("bit-banging")", containing the words ""Bit-Banging" (i.e. using the processor to hammer out the protocol via the GPIOs) is very hard. The processor isn’t really designed for this. It has other work to do… for slower protocols you might be able to use an IRQ...".

It goes on to talk about using the PIO.  You might also consider using a UART, if the protocol is suitable for using one.  You'd then get good timing during sending a short message, though there might be less predictable latency to start sending one.

JonnyHaystack, 2022-06-18:
@Andy2No will look into this tomorrow. It's not going to be feasible for me to replace all delays with workarounds because of calls to Arduino libraries which use delay functions internally.

JonnyHaystack, 2022-06-18:
Looks like swapping the cores has does still have the problem but the behaviour isn't quite the same. There is a disruption but it's somewhat more... erratic?

[image]

Code:
```
#include <pico/multicore.h>
#include <pico/stdlib.h>

bool gp1;

void blah();

int main(void) {
    gpio_init(1);
    gpio_init(2);
    gpio_set_dir(1, GPIO_OUT);
    gpio_set_dir(2, GPIO_OUT);

    multicore_reset_core1();
    multicore_launch_core1(blah);

    while (true) {
        gpio_put(2, 1);
        sleep_us(50);
        gpio_put(2, 0);
    }
}

void blah() {
    while (1) {
        gp1 = !gp1;
        gpio_put(1, gp1);
    }
}
```

earlephilhower, 2022-06-18:
I just actually looked at the code and think there are a few things you might not be taking into account:
* There is a shared XIP cache.  Calling code on one core can cause a code spill and astall on the other core
* Interrupts aren't disabled here, so your code can be interrupted for things like USB/etc.
* Even RAM is a shared resource with 1 port, so a core could stall the other with a simple global/heap read or write.

I'll close this here and let you finish up on the SDK site, but the code behavior seems expected here...

closed 2022-06-18 by earlephilhower

JonnyHaystack, 2022-06-18:
FYI I have run much heavier code on core 1 before, while running this same code on core 0, and it didn't cause issues. Seems it only happens when using sleep, but pretty much anything else is fine. So it doesn't particularly appear to be related to your points above. USB interrupts and whatever else is there by default haven't been giving me trouble either, and like I said this happens even when doing literally nothing on the second core except sleeping repeatedly, even with no GPIO use.

Although this is clearly not an arduino-pico specific problem, I think it might be worth adding some way to make `delay()`/`delayMicroseconds()` use `busy_wait_ms()`/`busy_wait_us()` instead of `sleep_ms()`/`sleep_us()`, so that this type of issue can be avoided. Because as it is currently, if I want to use an Arduino library that uses delay, it's going to cause problems. A compiler flag to select which implementation to use for delay would be great.

JonnyHaystack, 2022-06-20:
That seems to always be defined when using arduino-pico so it doesn't work, but I found `PICO_SDK_VERSION_STRING` and `ARDUINO_ARCH_RP2040`, which I think should work fine.