earlephilhower/arduino-pico #642: TinyUSB Web Serial Output buffering/delay on PlatformIO

zjwhitehead, 2022-06-20:
When using PlatfomIO and the latest RP2040 core/packages WebUSB now shows up in the browser but is painfully slow and much slower than when using the ArduinoIDE. 

Code is taken from the Adafruit web usb serial example and modified for PlatformIO 
https://github.com/zjwhitehead/PIO-WebUSB-Testing

- Note: I tried setting `-O3` and similar flags but I get the same results. 
- Cant tell if its related to https://github.com/earlephilhower/arduino-pico/issues/630 

Hopefully this is the right repo for the issue, and its easily reproducible for others. 

Heres what it looks like when running the platform via Aruduino IDE
[image]
And here is when running the same code via PlatformIO

[image]

maxgerhardt, 2022-06-20:
Thanks for the report, I'll look into it. Your `platformio.ini` looks good though.

earlephilhower, 2022-06-21:
For 2.2.0 I brought in the latest version of the Adafruit_TinyUSB library.  Possibly something's broken there?  I don't think there's much in the core related to the codepaths you're hitting (or at least the paths used in the core haven't recently changed...2.2.0 was more about multicore libc and FreeRTOS fixes than anything else).

zjwhitehead, 2022-06-22:
I was just able to reproduce my original findings with the latest updates (2.2.0). The Arduino IDE flashing of the same code works as expected, while the PlatformIO seems to miss the line breaks and is much slower when using web_usb serial output. 
I'm going to push on reproducing the issue in Arduino IDE now to be broken like you @maxgerhardt

zjwhitehead, 2022-06-22:
I figured it out, its with the TinyUSB library version. Im working backwards now to determine the version that broke. 
2 reasons why it was working on my ArduinoIDE 
- My TinyUSB library wasnt updated (on 1.6, latest is 1.12)
- The Ardunio IDE for some reason is preferring the Arduino Library installed version of TinyUSB vs whats in the board package for the pico 
relevant Arduino IDE output below
```
Using board 'adafruit_feather' from platform in folder: C:\Users\zwhit\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.2.0
Using core 'rp2040' from platform in folder: C:\Users\zwhit\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.2.0
Detecting libraries used...
"C:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.4.0-b-cb31b54/bin/arm-none-eabi-g++" -c -Werror=return-type -DUSE_TINYUSB "-IC:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0/libraries/Adafruit_TinyUSB_Arduino/src/arduino" -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x239a -DUSB_PID=0x80f1 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather RP2040\"" -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions "-iprefixC:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0/" "@C:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0/lib/platform_inc.txt" -fno-rtti -std=gnu++17 -g -w -x c++ -E -CC -DSERIALUSB_PID=0x80f1 -DUSBD_MAX_POWER_MA=250 -DF_CPU=125000000L -DARDUINO=10816 -DARDUINO_ADAFRUIT_FEATHER_RP2040 "-DBOARD_NAME=\"ADAFRUIT_FEATHER_RP2040\"" -DARDUINO_ARCH_RP2040 -DDEBUG_RP2040_PORT=Serial -Os "-IC:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0\\cores\\rp2040" "-IC:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0\\variants\\adafruit_feather" "C:\\Users\\zwhit\\Documents\\Arduino\\builds\\sketch\\webusb_serial.ino.cpp" -o nul
Alternatives for Adafruit_TinyUSB.h: [Adafruit_TinyUSB_Library@1.7.1 Adafruit_TinyUSB_Arduino@1.11.0]
ResolveLibrary(Adafruit_TinyUSB.h)
  -> candidates: [Adafruit_TinyUSB_Library@1.7.1 Adafruit_TinyUSB_Arduino@1.11.0]
"C:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.4.0-b-cb31b54/bin/arm-none-eabi-g++" -c -Werror=return-type -DUSE_TINYUSB "-IC:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0/libraries/Adafruit_TinyUSB_Arduino/src/arduino" -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x239a -DUSB_PID=0x80f1 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather RP2040\"" -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions "-iprefixC:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0/" "@C:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0/lib/platform_inc.txt" -fno-rtti -std=gnu++17 -g -w -x c++ -E -CC -DSERIALUSB_PID=0x80f1 -DUSBD_MAX_POWER_MA=250 -DF_CPU=125000000L -DARDUINO=10816 -DARDUINO_ADAFRUIT_FEATHER_RP2040 "-DBOARD_NAME=\"ADAFRUIT_FEATHER_RP2040\"" -DARDUINO_ARCH_RP2040 -DDEBUG_RP2040_PORT=Serial -Os "-IC:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0\\cores\\rp2040" "-IC:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0\\variants\\adafruit_feather" "-IC:\\Users\\zwhit\\Documents\\Arduino\\libraries\\Adafruit_TinyUSB_Library\\src" "C:\\Users\\zwhit\\Documents\\Arduino\\builds\\sketch\\webusb_serial.ino.cpp" -o nul
"C:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.4.0-b-cb31b54/bin/arm-none-eabi-g++" -c -Werror=return-type -DUSE_TINYUSB "-IC:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0/libraries/Adafruit_TinyUSB_Arduino/src/arduino" -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x239a -DUSB_PID=0x80f1 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather RP2040\"" -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions "-iprefixC:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0/" "@C:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0/lib/platform_inc.txt" -fno-rtti -std=gnu++17 -g -w -x c++ -E -CC -DSERIALUSB_PID=0x80f1 -DUSBD_MAX_POWER_MA=250 -DF_CPU=125000000L -DARDUINO=10816 -DARDUINO_ADAFRUIT_FEATHER_RP2040 "-DBOARD_NAME=\"ADAFRUIT_FEATHER_RP2040\"" -DARDUINO_ARCH_RP2040 -DDEBUG_RP2040_PORT=Serial -Os "-IC:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0\\cores\\rp2040" "-IC:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0\\variants\\adafruit_feather" "-IC:\\Users\\zwhit\\Documents\\Arduino\\libraries\\Adafruit_TinyUSB_Library\\src" "C:\\Users\\zwhit\\Documents\\Arduino\\libraries\\Adafruit_TinyUSB_Library\\src\\arduino\\Adafruit_TinyUSB_API.cpp" -o nul
"C:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.4.0-b-cb31b54/bin/arm-none-eabi-g++" -c -Werror=return-type -DUSE_TINYUSB "-IC:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0/libraries/Adafruit_TinyUSB_Arduino/src/arduino" -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x239a -DUSB_PID=0x80f1 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather RP2040\"" -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sect [... cut ...] \zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0\\cores\\rp2040" "-IC:\\Users\\zwhit\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.2.0\\variants\\adafruit_feather" "-IC:\\Users\\zwhit\\Documents\\Arduino\\libraries\\Adafruit_TinyUSB_Library\\src" "C:\\Users\\zwhit\\Documents\\Arduino\\libraries\\Adafruit_TinyUSB_Library\\src\\arduino\\Adafruit_USBD_Device.cpp" -o nul
```

Will update when I figure out which version of the library broke this example

zjwhitehead, 2022-06-22:
So from testing various release from https://github.com/adafruit/Adafruit_TinyUSB_Arduino/releases

My initial guess is that its further upstream from here aka https://github.com/adafruit/Adafruit_TinyUSB_Arduino/pull/167

[Commits since 10.0.1](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/compare/1.10.1...master)

| Version |  Status|
| ----------- | ----------- |
| 1.7.1 | ✅|
| 1.9.4 | ✅|
| 1.10.0 |✅ |
| 1.10.2 | ❌|
| 1.11.0 |❌ |

maxgerhardt, 2022-06-22:
Then I think the best way to go about it is to open an issue at https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues -- I'll try to do so later in the day.

maxgerhardt, 2022-06-22:
This issue is now tracked in https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/176.

zjwhitehead, 2022-07-06:
After returning to test this theres good news and bad news. 

It seems like https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/176 fixed the issue with the line breaks not being respected in `.println()`

But the bad news is I cant consistently get one line to be posted to the webusb serial output at a time. Meaning, we still are getting a buffering issue where 3+ `.println("foo")` appear in chunks vs individually and in a timey manner like with other boards/platforms.

Calling `.flush()` doesnt seem to impact the performance.

Im going to dig in a bit more and post further findings but Im currently at a loss.

maxgerhardt, 2022-07-06:
I think the TinyUSB devs want other devs to counteract this by accumulating data into a buffer and then splitting what can be split and keeping the rest until new data arrives that can eventually be split.

zjwhitehead, 2022-07-06:
@maxgerhardt I hope thats not the case.
If so we should change the method from `.print()` to `.addToBuffer()` etc. because the functionality between the normal Serial console output and the webusb.print is very different now.
I can confirm by inspecting the incoming Chrome serial data that its a `ArrayBuffer(64)` every time. 😞 

For background info I've ben using https://github.com/bblanchon/ArduinoJson (awesome lib btw) to create and parse JSON data that is sent to a simple webapp. Previously I could just send over a serialized 1 time JSON object and parse it but since I dont always know the string sizes etc it breaks up and invalidates the JSON

zjwhitehead, 2022-07-06:
I'm open to a workaround btw. 

Here's another good example of why this isn't ideal: You want to tell a website when you press a button. 
Modify the loop with the below code which is taken partially from [here](https://github.com/earlephilhower/arduino-pico/blob/master/libraries/rp2040/examples/Bootsel/Bootsel.ino#L12)
```
void loop() {
    // from WebUSB to both Serial & webUSB
  if (usb_web.available()) echo_all(usb_web.read());

  // From Serial to both Serial & webUSB
  if (Serial.available())   echo_all(Serial.read());
  

  if (BOOTSEL) {
    usb_web.println("You pressed BOOTSEL");
    // Wait for BOOTSEL to be released
    while (BOOTSEL) {
      delay(1);
    }
  }
}
```
You will **never** get notified when pressing the button (once) because it just so happens that the message string is less than the 64 threshold. 
Press it again? You'll get notified twice. Then you probably go down the rabbit hole of trying to debounce the button etc.

maxgerhardt, 2022-07-06:
Hmm I see what you mean. So when you print a short message and after that delay forever, the message gets never flushed out? Even if you call `.flush()`?

zjwhitehead, 2022-07-06:
Yep. Calling `.flush()` has no impact as far as I can tell.
Updated the example above to include it.

Perhaps this core isnt implementing `.flush()` properly?

maxgerhardt, 2022-07-06:
No, the `.flush()` method of the web serial is implemented in TinyUSB and was also altered in the PR to call this `tud_flush()`. Maybe a non-functioning flush should be reported there with a minimal example.

zjwhitehead, 2022-07-06:
Ok sorry for the confusion, I realized my TinyUSB library wasnt fully up to date in my PlatformIO. 
As you suggested, calling `.flush()` now makes it send immediately.

zjwhitehead, 2022-07-08:
yeah I'll close this and will open an issue if I run into other problems. 
Thanks!

closed 2022-07-08 by zjwhitehead

earlephilhower, 2022-07-08:
@maxgerhardt I can do a new release if this is the only thing holding up the PlatformIO merge.  OTW, I might wait a bit as there were only a couple new boards and this update as changes so far since 2.2.2

maxgerhardt, 2022-07-09:
The merge is currently held up by some company-political issues, not due to the content of the PR itself :/ I'm trying to get it through.

PR #672 (2022-07-08): Update TinyUSB to 1.13.3
Fix errors in WebUSB buffering
Fixes #642
commit 8eb030ed (2022-07-08): Update TinyUSB to 1.13.3 (#672)

Fix errors in WebUSB buffering
Fixes #642