earlephilhower/arduino-pico #643: Dependency Generation When Compiling

VisualMicro, 2022-06-20:
When compiling with this package there are no Dependency (*.d) files generated, which is useful for the Arduino IDE and Visual Micro to allow correct caching and quicker subsequent builds.

These can be added by users in platform.local.txt or a local board.txt (as shown in our [documentation here](https://www.visualmicro.com/page/Arduino-Compiler-Dependencies-Caching.aspx)), however it would improve build speed for all users if this was added into the packages' platform.txt, for both compiler.c.flags and compiler.cpp.flags (as used in the [Arduino MBED Package](https://github.com/arduino/ArduinoCore-mbed)).

earlephilhower, 2022-06-20:
Thanks for the pointers.  I thought it was handled internally by `arduino-builder` but I guess not!

closed 2022-06-20 by earlephilhower

VisualMicro, 2022-06-21:
Brilliant, thanks for the quick turnaround and release, tested on 2.2.0 and looks much better.

PR #644 (2022-06-20): Add dependency file generation to build
Fixes #643 .   Thanks @VisualMicro
commit a0060d9c (2022-06-20): Add dependency file generation to build (#644)

Fixes #643