earlephilhower/arduino-pico #648: Document quirks on Adafruit TinyUSB Examples

maxgerhardt, 2022-06-21:
I just went to try out the very first Adafruit USB example linked in the library, which is https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/9c03087bcdbe1dc4faff3e2a41c3bc109ee2a554/examples/CDC/cdc_multi/cdc_multi.ino. The example is supposed to create two USB serial devices.

To my surprise this did not work at all -- reason being, that the Adafruit USB CDC class checks the config

```cpp
void Adafruit_USBD_CDC::begin(uint32_t baud) {
  (void)baud;

  // already called begin()
  if (isValid()) {
    return;
  }

  // too many instances
  if (!(_instance_count < CFG_TUD_CDC)) {
    return;
  }
```

and `CFG_TUD_CDC` is the maximum allowed number of CDC devices, which in the RP2040 configuration

https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/9c03087bcdbe1dc4faff3e2a41c3bc109ee2a554/src/arduino/ports/rp2040/tusb_config_rp2040.h#L58

is *1*. Changing this to `2` and reuploading made the two serial devices appear properly.

Maybe this will confuse some other people too that at least one example of that library is not executable on the RP2040 with the default configuration and documentation should be added to it?

Another would be to make the USB configuration file, or at least the number of maximum devices, configurable with a macro / custom user config header file.

earlephilhower, 2022-06-21:
Should we maybe open a PR on the upstream lib?  I don't do anything other than include it (and allow it to hook into the Serial stack).

maxgerhardt, 2022-06-22:
Mhmm actually I think I'm in the wrong here, the sketch itself says that it needs this configuration change.

https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/24ddb004df4acc6edf264f4c67273c250ca83230/examples/CDC/cdc_multi/cdc_multi.ino#L16-L18

Maybe we can just add a tiny note in https://github.com/earlephilhower/arduino-pico/blob/master/docs/usb.rst regards to where this config file is **if** it needed changing for specific sketches.

maxgerhardt, 2022-06-23:
In PlatformIO this is easy: Since the TinyUSB library checks for the existance of a macro

https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/7264c1492a73d9a285512752b03f2550841c06bc/src/tusb_option.h#L172-L177

We can do 
```ini
build_flags = 
   -DCFG_TUSB_CONFIG_FILE=\"custom_tusb_config.h\"
   -Iinclude/ 
```
and then in the `include/custom_tusb_config.h` file we can paste in a copy of https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/src/arduino/ports/rp2040/tusb_config_rp2040.h but with the needed changes.

In the Arduino IDE we would have add a [`platform.local.txt`](https://arduino.github.io/arduino-cli/0.21/platform-specification/) in the core folder and modify compiler commands with this `-D` inclusion..
```

JonnyHaystack, 2022-06-24:
> In PlatformIO this is easy: Since the TinyUSB library checks for the existance of a macro [...]

Do you know if there's a way to get multiple HID endpoints like this? I tried setting `CFG_TUD_HID` to 2 and creating two instances of `Adafruit_USBD_HID` and calling `begin()` on both of them, but I can't get both to work properly together.

maxgerhardt, 2022-06-24:
The library doesn't really seem to support this, it has a global variable holding a reference to "the" HID instance.

https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/7264c1492a73d9a285512752b03f2550841c06bc/src/arduino/hid/Adafruit_USBD_HID.cpp#L36-L37

But maybe https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/HID/hid_composite/hid_composite.ino is what you want? Multiple HID descriptors so that can do mouse and keyboard

JonnyHaystack, 2022-06-25:
> The library doesn't really seem to support this, it has a global variable holding a reference to "the" HID instance. [...]

Ah, that's annoying. Yes I have just been using multiple reports on one interface so far, but it's rather limiting because I can't see a way to have multiple libraries e.g. one for gamepad, one for keyboard, and use them in parallel without either having to know the other exists. Then I saw this the other day and thought maybe it would solve that whole problem.

earlephilhower, 2022-06-25:
Yes, TinyUSB gives you one HID.  You need to generate the  descriptors/etc. to stitch together whatever libraries are used.  See how https://github.com/earlephilhower/arduino-pico/blob/master/cores/rp2040/RP2040USB.cpp works with potentially one keyboard and one mouse.

PR #659 (2022-06-26): Add notes on TinyUSB documentation and quirks
Closes #648 by documentating said findings.

PlatformIO specific instructions were tested locally and found to work.
commit baf65f51 (2022-06-26): Add notes on TinyUSB documentation and quirks (#659)

Fixes #648
closed 2022-06-26 by earlephilhower