earlephilhower/arduino-pico #651: ISR function is not executing

drakejest, 2022-06-24:
I was revisiting an old code, and weirdly enough it is not functioning as i remember it. I have narrowed the culprit down, and it would seem that my ISR function is not executing although the signal in the pin is there.

On the [screenshot ](https://aws1.discourse-cdn.com/arduino/original/4X/6/6/8/668064e1217e69b9da049cb5f777077f57e6eabf.png) 
digitalRead is changing state but the pressEvent is not set. No matter how many times the interrupt signal is sent i am not able to trigger the ISR routine

Im using core v2.2.0 

```
bool pressEvent = false;
const int interruptPin = 9;

void switchPressed() {
  pressEvent = true;
}

void setup() {
  Serial.begin(9600);
  while (!Serial);
  Serial.println("Starting");

  pinMode(interruptPin, INPUT);
  attachInterrupt(digitalPinToInterrupt(interruptPin), switchPressed, RISING);

}

void loop() {
  Serial.print(digitalRead(interruptPin));
  Serial.print("\t");
  Serial.println(pressEvent);
}
```

drakejest, 2022-06-24:
I rolled back to version 1.13.0 and it works there no problem.

earlephilhower, 2022-06-24:
I've been trying things and it looks like the pico SDK develop branch might have an issue.  2.1.1 works, 2.2.0 does not, but there is no code change in the core.  The pico-sdk branch *was* updated to include some mandatory fixes...

earlephilhower, 2022-06-24:
https://github.com/raspberrypi/pico-sdk/pull/850

Changed the way gpio interrupts are handled, but kept the same API.  Ouch.

earlephilhower, 2022-06-24:
We'll give it a few days, and if the fix doesn't get merged I'll make a new release off of my own branch.

PR #653 (2022-06-24): Update to latest pico-sdk develop branch
Fixes #651
closed 2022-06-24 by earlephilhower