earlephilhower/arduino-pico #660: initial upload to sparkfun thing plus fails: need to select a serial port

wez, 2022-06-27:
I have a sparkfun thing plus that I've had for a while but never programmed before.
Trying to upload a sketch fails because there isn't a port selected.
However, there isn't a port to select: when the board comes up in UF2 mode, dmesg doesn't show any serial ports, so I'm a bit stuck.

Is there something I can do to bootstrap this board?
I'm wondering if perhaps I need to upload a newer UF2 that enables the serial port?

earlephilhower, 2022-06-27:
You should be able to hold the bootsel button while plugging it in and upload that way.  See the readme.

wez, 2022-06-27:
I found the uf2 file created by the build in the arduino temp dir and manually copied it over to bootstrap it to get the serial port present when bootsel is activated.  Without that, I could do nothing.

Things were made more complicated because I'm running Fedora and it installs Arduino from a flatpak and that also messes with its ability to detect the device.

After uninstalling that and manually installing Arduino, things work a bit more sanely.

earlephilhower, 2022-06-27:
Thanks for the update.  Last I checked, the Microsoft Store packaged Arduino has the same problem....doesn't allow access to the serial ports or root drives making it pretty useless.  There might be compelling reasons for self-contained packages, but they always seem to be unfit for purpose in practice!

closed 2022-06-27 by earlephilhower

VarLad, 2022-07-29:
@wez Did you check its permissions via Flatseal?

wez, 2022-07-29:
No, I downloaded the working program directly from Arduino and moved on with my life :-p