earlephilhower/arduino-pico #661: Error: Could not find the package with 'earlephilhower/...... requirements for your system 'darwin_arm64'

kikolobo, 2022-06-27:
Hello. I was compiling well on my M1 Mac using the original isntructions where I used:

platform_packages = 
	maxgerhardt/framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
	maxgerhardt/toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.3-a/x86_64-apple-darwin14.arm-none-eabi-ed6d983.220212.tar.gz

I followed the instructions, removed the lines and ran :
pio pkg update -g -p https://github.com/maxgerhardt/platform-raspberrypi.git

But now when I compile I get the error with:
Error: Could not find the package with 'earlephilhower/toolchain-rp2040-earlephilhower @ ~5.100300.0' requirements for your system 'darwin_arm64'

My guess is that I need to change something to use roseta instead of native darwin_arm64, but I am clueless.

Any idea on how to fix this?
I am running a generic board I made. This is my .ini:

-----

earlephilhower, 2022-06-27:
@maxgerhardt would you know if people publish Mac X86-64 binaries under the `darwin_arm64` architecture, or if there is some way to make Platform.IO fall back to the x86-64 binaries on the M1 Macs?

I can't imagine everyone has rebuilt their toolchains for the new Mac architectures yes, and AFAIK there's not a working cross-compiler available yet under Linux.

kikolobo, 2022-06-27:
> @maxgerhardt would you know if people publish Mac X86-64 binaries under the `darwin_arm64` architecture, or if there is some way to make Platform.IO fall back to the x86-64 binaries on the M1 Macs?  

Having at least a way to fallback would be good.. Or if you have a process where I can revert back to my original setup because, changing back .ini file won't fix it! And I have a big project 😊 

Any help welcomed!

maxgerhardt, 2022-06-28:
>@maxgerhardt would you know if people publish Mac X86-64 binaries under the darwin_arm64 architecture, or if there is some way to make Platform.IO fall back to the x86-64 binaries on the M1 Macs?

My bad I didn't look at this, *indeed* that is what supposed to be done for Macs. PlatformIO staff-uploaded packages do this for all Mac binaries, they compile it for x86-64 and then declare in the `package.json` to be compatible with both `darwin_x86_64` and `darwin_arm64`. Compare e.g. 

[image]

from https://api.registry.platformio.org/v3/packages/platformio/tool/toolchain-gccarmnoneeabi.

maxgerhardt, 2022-06-28:
@kikolobo 

Can you try the `platformio.ini`

```ini
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
platform_packages =
   toolchain-rp2040-earlephilhower@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.4.0-b/x86_64-apple-darwin14.arm-none-eabi-cb31b54.220619.tar.gz
```

sstaub, 2022-06-28:
Followed the instructions on the docs and it works on my macMini M1. The only problem I have that the first installation get stuck with rebuild intellisense. I have to first change the .ini to another board, then restart VSC.
After this procedure it works.

kikolobo, 2022-06-28:
> @kikolobo
> 
> Can you try the `platformio.ini`
> 
> ```ini
> [env:pico]
> platform = https://github.com/maxgerhardt/platform-raspberrypi.git
> board = pico
> framework = arduino
> board_build.core = earlephilhower
> board_build.filesystem_size = 0.5m
> platform_packages =
>    toolchain-rp2040-earlephilhower@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.4.0-b/x86_64-apple-darwin14.arm-none-eabi-cb31b54.220619.tar.gz
> ```
> 
> If this does **not** work, locally download the toolchain link listed above to some folder, go in the `package.json` and add `darwin_arm64` to the systems array and then use
> 
> ```ini
> platform_packages =
>    toolchain-rp2040-earlephilhower@symlink://<path to where you extracted it>
> ```
> 
> in the `platformio.ini`.

Let me try all this in a few hours and get back to you.

Thanks for the quick response!

kikolobo, 2022-06-28:
Allright, this are my findings...

1) If I run Visual Studio, ticking the : Run as Roseta option, everything works. (CMD+I in the VSCode App Icon + Ticking the 'Open as Roseta') Option.  But this is sub optimal because VSCode runs as roseta, and it does feel noticeably slower.

This option also works with this more standard .ini file (without the platform_packages option set):

```ini
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
```

2) Following your instructions, I turned back off the open as roseta (which was working before all this) and used this ini file:

> ```ini
> [env:pico]
> platform = https://github.com/maxgerhardt/platform-raspberrypi.git
> board = pico
> framework = arduino
> board_build.core = earlephilhower
> board_build.filesystem_size = 0.5m [...]

This does not work. It returns this warning during compilation: 
**arm-none-eabi-gcc: command not found**

And then this error during compilation:
*** [.pio/build/pico/FrameworkArduinoBootloader/boot2_generic_03h_4_padded_checksum.S.o] Error 127

I am guessing I am missing the 'arm-none-eabi-gcc' but I am a little bit assistant to try and install tools at this point since I have this important project to deliver. So I will continue with this research in a few days after I finish.

3) To your last suggestion.. I am doing that test right now.. Will let you know how it goes (will edit this comment) soon.

earlephilhower, 2022-06-28:
Thanks for the reports and info!

I'll be manually adding the `darwin_arm64` tag to the package and re-uploading later today which should "solve" this problem w/o needing to manually edit `ini` files.  The toolchain `makefile` also has been updated to do this automatically on the next release.  So give it a few days and all should be up and running...

maxgerhardt, 2022-06-28:
>This does not work. It returns this warning during compilation:
>arm-none-eabi-gcc: command not found

Yeah I think this is because the download still declares only `darwin_x86_64` in its manifest and PlatformIO still checks after it downloads it (even though its an explicit downlink link instruction). This is then fixed with the third method.

If you work in a team and everyone's working on Mac and need a fix now-now, then you can also include the toolchain folder that you have now extracted and modified in the project itself, then refer to it with a path-relative symlink (`@symlink://toolchain_mac_arm64` for to refer to `<project folder>/toolchain_mac_arm64`).

@earlephilhower Already fixed this in https://github.com/earlephilhower/pico-quick-toolchain/commit/c61c1288f4e87fc7cefded2d0f4fe6b294ff724e, it just needs to rebuild and upload to the registry + be approved.

kikolobo, 2022-06-28:
Thank you all for your help. Will wait for the next release in a few days.

i am impressed and grateful for the speed of response from the team.

closed 2022-06-28 by kikolobo