earlephilhower/arduino-pico #666: REQ: Add support for Raspberry Pi Pico W

Martin-HiPi, 2022-06-30:
Hi!

I just wanted to let you know what the PID:VID of the new Pico W (AKA PiCow): 2e8a:f00a

[Raspberry Pi Pico W](https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#raspberry-pi-pico-w-2)
[Raspberry Pi Pico W Datasheet](https://datasheets.raspberrypi.com/picow/pico-w-datasheet.pdf)

The onboard LED has moved to WL_GPIO0 on the wireless chip.

earlephilhower, 2022-07-01:
Looks like some assumptions I made about pin numbering are no longer valid and we're going to have to move to weak `pinMode` and `digitialWrite/Read` (besides all the new fun WiFi stuff!)

It doesn't look like any availability here in the US right now.  We'll see in a couple weeks if the usual spots have stock...

Martin-HiPi, 2022-07-01:
@earlephilhower DM me, I might be able to help ;-)

I have one on hand, I work for PiShop.us

FeuerSturm, 2022-07-02:
> Plus, this is probably more like a marathon than a sprint. Adding a WiFi class and associated helpers feels like it's going to take some time...

@earlephilhower can that WiFi class pretty please be "compatible" with the functions of the usual ones like WiFININA etc.,
that way we could easily adopt our existing projects to the new Pico W, that would be amazing!

Thanks for all the hard work you're doing!

earlephilhower, 2022-07-02:
@FeuerSturm I was planning on "borrowing" the `WIFININA` classes, in fact, and porting the implementation.  :)  Along with a lot of the work we've done on the ESP8266 WiFi helper classes (i.e. web server, http(s) client, etc.).  Probably want to name it as `WiFi` or `WIFI` to avoid any confusion w/the Arduino Nano Connect libs, of course.

And it turns out MicroCenter doesn't have any in stock yet, so we'll see how availability looks next week.

FeuerSturm, 2022-07-02:
@earlephilhower That's great news on the compatibility with exist code :)
I've take a quick look at the driver in PicoSDK and it seems like there's no UDP related functions in it (or I just overlooked them) which would be pretty sad as most of my stuff relies on that.

Bad news about the availability, my local electronics store didn't have any either, so I ordered some, won't be with me before next week either.

I'm still a bloody beginner, but if there's anything I can do to help to make that Wifi class happen, I'd be glad to do so, even if it's just testing it.
I'd be glad to donate some bucks as well.

PR #670 (2022-07-15): Add Pico W WiFi support
Add support for the WiFi chip on the Pico W board.

Works well enough for experimental use

To Do:
`[X]` Update LWIP options (now it is the default example one
`[X]` Add WiFi class
`[X]` Add WiFi client class
`[X]` Add support classes (DNS, etc.)
`[X]` Add WiFi server class
`[X]` Add AP mode
`[X]` Examples
`[X]` Docs (such as they are...)

* USB interrupt now no longer hard coded (conflicted with the WiFi IRQ).
* Add in Pico W board to makeboards.py
* Add in GPIO and variant support
* Initialize WiFi in the Variant
* Use manual LWIP, fix size accounting
* Remove the SDK [... cut]
closed 2022-07-15 by earlephilhower