earlephilhower/arduino-pico #667: PlatformIO instructions point to an older version of pico-quick-toolchain

skybrian, 2022-07-02:
I noticed that on this page: https://arduino-pico.readthedocs.io/en/latest/platformio.html

It recommends adding lines to your platformio.ini file that point to version 1.3.1-a of pico-quick-toolchain. But on the [releases page](https://github.com/earlephilhower/pico-quick-toolchain/releases/),  the latest version is 1.4.0-b. Should the number be bumped up?

(Also, in each release note for pico-quick-toolchain, the message has a broken link back to the arduino-pico repo.)

skybrian, 2022-07-02:
Never mind, reading more carefully, those instructions are deprecated now. I will update my platformio files.

(The release note link is still broken, though.)

earlephilhower, 2022-07-02:
Thx.  Fixed the note in the release.  There will be a real Platform.IO update any day now...

closed 2022-07-02 by earlephilhower