earlephilhower/arduino-pico #668: No more able to program my RP2040

jfl-207, 2022-07-02:
Hello,

First of all a big thank you for your work.

i'm actually no more able to program my RP2040.
Can not be sure at 100% but seems to be since the update a few day ago.

"Resetting COM10
Converting to uf2, output size: 114176, start address: 0x2000
Unable to build drive list
Une erreur est survenue lors du transfert du croquis
"
if using the "Mbed version" it's ok but no more with yours.
tried with 2 different board ans same result.

Any advice on what to check ?

regards

Jean-Francois

earlephilhower, 2022-07-02:
I assume you're running from the Board Manager install, yes?

Can you install version 2.0.2 and try again?  There was one change where instead of waiting 10 seconds and then checking for the Pico, the uploader instead tried over and over for 10 seconds.  This makes it much faster normally, but maybe there are corner cases under Windows when the drive is in the middle of appearing that the MSFT tools we use to scan report errors.

If you're using the `git` version, try changing:
https://github.com/earlephilhower/arduino-pico/blob/5be405308e31b864f0d8f213fdf266344ed45bce/tools/uf2conv.py#L243-L245
to:
````
    except:
        r = ""
````

jfl-207, 2022-07-02:
Hi, wend down to 2.0.2.
Still no luck.
message slightly different now
"Resetting COM10
Caught exception during reset!
Converting to uf2, output size: 157696, start address: 0x2000
Unable to build drive list
Une erreur est survenue lors du transfert du croquis
"
Other strange thing : flash drive of RP2040 stay visible in the explorer after the crash

earlephilhower, 2022-07-02:
Maybe this is related to the label of one of your drives.  We used to have an issue w/non straight ASCII names, but that was fixed a while ago.  Can you run and report the output from:
````
wmic PATH Win32_LogicalDisk get DeviceID,VolumeName,FileSystem,DriveType
````
and
````
powershell -NonInteractive -Command Get-WmiObject -class Win32_LogicalDisk | Format-Table -Property DeviceID, DriveType, Filesystem, VolumeName
````

earlephilhower, 2022-07-02:
Also, are you running the downloaded installer IDE or the Windows Store version?  The Windows Store version may not have enough privileges to actually get drive list/etc. which is needed to actually do the upload.  (And in some cases I've seen the Windows Store version not even allow COM: access, making it completely useless for almost any board!)

jfl-207, 2022-07-02:
[image]
if no mistake, it's the dowloaded one.
as for the 2 command, could you explain me how to lauch them ?

FeuerSturm, 2022-07-02:
I had a similar problem some time ago,
unplug the Pico, hold BOOTSEL button while plugging it in again, copy https://datasheets.raspberrypi.com/soft/flash_nuke.uf2 to flash drive and wait.
After that I could use mine again with Arduino IDE

earlephilhower, 2022-07-02:
To run the commands I listed above start a `cmd` window (hit Windows key, then type "cmd" and hit return, and then run the cmds).

Also, since it *used* to work fine and not even older versions do not, it seems like something has changed on your setup.  Did you recently update Windows, have a USB stick plugged in or CD loaded, etc.?

jfl-207, 2022-07-04:
Hello, sorry for delay in reply.
i'm not able to run both command "command not recognized"
no change on pc , so don't know what'a happening.
will try to install Arduino ide on another PC tommorow

Andy2No, 2022-07-05:
@jfl-207  It looks like you need a caret (^) before the pipe (|) in the second command (before Format-Table) - or at least you do in Windows 7, which is what I've just tried it on, so I expect it's the same:

So, instead of

```
powershell -NonInteractive -Command Get-WmiObject -class Win32_LogicalDisk | Format-Table -Property DeviceID, DriveType, Filesystem, VolumeName
```

Try

```
powershell -NonInteractive -Command Get-WmiObject -class Win32_LogicalDisk ^| Format-Table -Property DeviceID, DriveType, Filesystem, VolumeName
```

Use the Copy symbol that appears if you hover the mouse near the right hand end of each of those boxes, to place one of those lines on the clipboard, then right click in the command window, and choose Paste, then press Enter.

Here's what I got - first the wmic command worked, listing my drive, then the powershell line failed - apparently because the pipe gets processed by the command shell instead of powershell.  Finally, the powershell line again, with the caret, which worked: 

```
Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

C:\Users\andy>wmic PATH Win32_LogicalDisk get DeviceID,VolumeName,FileSystem,Dri
veType
DeviceID  DriveType  FileSystem  VolumeName
C:        3          NTFS        Tosh_3as2TB
D:        3          NTFS        Toshiba12TB
E:        3          NTFS        Element_5TB
W:        5

C:\Users\andy>powershell -NonInteractive -Command Get-WmiObject -class Win32_Log
icalDisk | Format-Table -Property DeviceID, DriveType, Filesystem, VolumeName
'Format-Table' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\andy>powershell -NonInteractive -Command Get-WmiObject -class Win32_Log
icalDisk ^| Format-Table -Property DeviceID, DriveType, Filesystem, VolumeName

DeviceID                      DriveType Filesystem          VolumeName
--------                      --------- ----------          ----------
C:                                    3 NTFS                Tosh_3as2TB
D:                                    3 NTFS                Toshiba12TB
E:                                    3 NTFS                Element_5TB
W:                                    5
```

jfl-207, 2022-07-05:
@Andy2No 
Thanks for your help.
i'm now able to run the powershell script.
No luck with "WMIC", but it's seems it's because Microsoft remove it from win 11.

jfl-207, 2022-07-05:
@earlephilhower 
Hi,
Was able to run the script. here is the result.
C:\Users\jfl_2>powershell -NonInteractive -Command Get-WmiObject -class Win32_LogicalDisk ^| Format-Table -Property DeviceID, DriveType, Filesystem, VolumeName

DeviceID DriveType Filesystem VolumeName
-------- --------- ---------- ----------
C:               3 NTFS       Windows
D:               3 NTFS       Data
G:               3 FAT32      Google Drive.

On other usefull info. i'm able to program the pico if i put it manually in program mode by unplug - hold bootsel - replug.
Not as user friendly as before but at least i can program it

regards
jf

Bodmer, 2022-07-19:
There's definitely a problem with COM ports with Windows with recent board package builds. The Pico seems to be able to lock up the Arduino IDE until the Pico is manually reset or disconnected. It then appears that a COM port that does not exists is trying to be acessed.  Closing the IDE then using the Arduno mbed board package works fine.

So ditto I cannot program my Pico anymore using this board package.

Bodmer, 2022-07-19:
Board package 2.2.2 works fine. As soon as I load 2.3.0 or later I have to nuke tha flash on the Pico after an upload otherwise it freezes the IDE when I try to open a COM port.

earlephilhower, 2022-07-19:
Absolutely nothing related to the upload process was changed between 2.2.2 and 2.3.0.  You can check the commit log, or git diffs:

````
earle@amd:~/Arduino/hardware/pico/rp2040$ git diff 988940b25e020cb6e22e7aedb5ca90ba63f2eb7d cfaae84813f7a2f1dab48d0dceb723e6c7a6590f tools/uf2conv.py
earle@amd:~/Arduino/hardware/pico/rp2040$ 
````
````
commit 988940b25e020cb6e22e7aedb5ca90ba63f2eb7d (tag: 2.3.0)
Author: Earle F. Philhower, III <earlephilhower@yahoo.com>
Date:   Fri Jul 15 16:49:36 2022 -0700

    Update version

commit abf2c586c7993c59c15ac04011f9c8e013d2b116
Author: Earle F. Philhower, III <earlephilhower@yahoo.com>
Date:   Fri Jul 15 16:47:53 2022 -0700

    Add Pico W WiFi support (#670)
    
    * Add support for the WiFi chip on the Pico W board.
    * USB interrupt now no longer hard coded (conflicted with the WiFi IRQ).
    * Add in Pico W board to makeboards.py
    * Add in GPIO and variant support
    * Initialize WiFi in the Variant
    * Use manual LWIP, fix size accounting
    * Remove the SDK WiFi overrides
    * Pulling in work done in the ESP8266 core.
    * Make IPAddress support IPv6
    * Build LWIP with IPv4 and IPv6 support
    * Use proper MAC
    * Avoid cyw_warn crash.  Make macro to a comment while building
    * Add WiFiServer
    * Add WiFiUdp
    * Move LWIP-specific support files to LWIP_Ethernet
    * Add WiFi::ping (ICMP ping)
    * Move ICMP echo (ping) to LWIPIntfDev
    * Move hostByName to LwipIntfDev
    * Add AP mode with simple DHCP server
    * Add some examples and basic ESP8266 compat hacks
    * Update Adafruit TinyUSB to fix crash
    * Set DHCP hostname
    * Make Wifi.begin() return CONNECTED with link + IP
    * Return connected() on WiFi::begin
    * Fix spurious TCP retransmission
    * Protect LWIP from reentrancy
    
    The Pico SDK calls "sys_check_timeouts() from inside a periodic interrupt.
    This appears unsafe, as the interrupt could happen while already in the
    (non-reentrant) LWIP code.
    
    Block the interrupt from calling sys_check_timeouts by using a global flag
    manually set via an RAII recursive lock.
    
    Add interrupt protection macros around critical sections inside LWIP via
    the standard defines.
    
    These two changes should make LWIP significantly more stable and long
    running.
    
    * Support disconnecting and reconnecting WiFi
    * Add WiFiServer simple example
    * Update documentation
    
    Fixes #666
    Fixed #665

commit c025c4a1f8d52157443474351cecacbb8fe8a11e
Author: Earle F. Philhower, III <earlephilhower@yahoo.com>
Date:   Thu Jul 14 09:04:02 2022 -0700

    Fix ::printf/etc. due to mutex size conflict (#677)
    
    When real multicore/lockign support was added to newlib, there was an opaque
    field in FILE that was used as a mutex, but was only 4 bytes in size.  The
    recursive mutexes on the RP2040 are 8 bytes.  This mismatch caused corruption
    of the FILE structure and crashes of the system when ::printf/::puts/etc. were
    run.
    
    Adjust the lock field size in FILE to 8 bytes and rebuild the toolchain to
    fix.

commit fb05d969794885e04e07247d146eca45acaaa234 (usbvid)
Author: Earle F. Philhower, III <earlephilhower@yahoo.com>
Date:   Thu Jul 14 07:43:27 2022 -0700

    Add delay to 2nd core start for Picoprobe (#676)
    
    No idea why, but when a Picoprobe upload and reset is used, the 2nd core
    does not start w/o a __wfe() call on the first one before launch.
    
    Fixes #674
    Fixes #402

commit 8eb030ed899a509a6460a45751b229b9c2e46420
Author: Earle F. Philhower, III <earlephilhower@yahoo.com>
Date:   Fri Jul 8 13:45:01 2022 -0700

    Update TinyUSB to 1.13.3 (#672)
    
    Fix errors in WebUSB buffering
    Fixes #642

commit 5be405308e31b864f0d8f213fdf266344ed45bce
Author: Pontus Oldberg <pontus.oldberg@invector.se>
Date:   Thu Jun 30 19:16:44 2022 +0200

    Add support for new SD/RTC and SubGHz boards (#663)

commit af0c1c87f51056b74b65bdca33b238d587e4d5d5 (t1)
Author: Earle F. Philhower, III <earlephilhower@yahoo.com>
Date:   Wed Jun 29 12:50:42 2022 -0700

    Automatically update the JSON file in makeboards (#664)
    
    Keep the Arduino Board Manager list up-to-date from makeboards.py

commit cfaae84813f7a2f1dab48d0dceb723e6c7a6590f (tag: 2.2.2)
Author: Earle F. Philhower, III <earlephilhower@yahoo.com>
Date:   Tue Jun 28 15:07:55 2022 -0700

    Update version
````

Bodmer, 2022-07-19:
Upload seems to work fine.

This is the simplest program that seems to freeze and freeze the Arduino IDE too:
```
void setup() {
  Serial.begin(115200);
  delay(5000);
  Serial.println("Just booted!");
  delay(10000);
  while (!Serial);
}

void loop() {
 Serial.println(1234.56);
 delay(1000);
}
```
If I open the COM port monitor window within 5s of the upload then I see "Just Booted". Then 10s later the Pico and IDE freezes.

This happens even if I remove the while(!Serial)

earlephilhower, 2022-07-19:
Are using the built-in USB or the Tools->USB->Tiny USB option w/Adafruit_TinyUSB?  There's still no changes to the core related to the internal serial (nor any TInyUSB change in the SDK), but the AdaFruit_TinyUSB did have its rev bumped up.

jfl-207, 2022-07-19:
> Are using the built-in USB or the Tools->USB->Tiny USB option w/Adafruit_TinyUSB? There's still no changes to the core related to the internal serial (nor any TInyUSB change in the SDK), but the AdaFruit_TinyUSB did have its rev bumped up.

Hi,

On my side, i'm programming via the built in USB.

Regards
Jean-Francois

Bodmer, 2022-07-19:
I am using Pico SDK USB. I get compile errors with the Adafruit TinyUSB.
```
...undefined reference to `_ZN17Adafruit_USBD_CDC5beginEm
...undefined reference to `Serial'
```

The crash seems to happen in the print stream (mabe?) because further tests show both:

Serial.println(1234.56);
and
tft.println(1234.56);

will crash the Pico. This suggests it is not the USB driver.

Bodmer, 2022-07-19:
Peculiar, I delete the line "Serial.println(1234.56);" and the Adafruit USB driver compiles OK.

earlephilhower, 2022-07-19:
@Bodmer that's very different than this issue, but interesting as there was a toolchain update related to printing floats (and other `stdio` things related to FILE).  Are you running the `git` version?  If so, you also need to to ./tools/get.py to install the proper toolchain.  The busted toolchain would crash on your code, the new one was tested fine.    #677

Bodmer, 2022-07-19:
I'm just using the Arduino IDE to swap the board package versions via the manager.

Bodmer, 2022-07-19:
Yes, it does appear that my problems with 2.3.x are all related to floats in the print stream!

earlephilhower, 2022-07-19:
Good debug.  Can you open a new issue to track?  I must have introduced a regression somewhere, because #614 was all about printing floats...

earlephilhower, 2022-07-21:
#689 took this one over

closed 2022-07-21 by earlephilhower