earlephilhower/arduino-pico #669: Feasibility of FreeRTOS integration as ESP32?

taskx6004, 2022-07-02:
Hi,
I am trying to port some ESP32 PIO project to Pico, after some trail and error, I found it's not a straight forward job. Some OS calls like `QueueHandle_t` or virtual member functions like `setInterruptAction()` (supposedly) is from ESP32's integrated FreeRTOS implementaion
(or not).

My question is, if it's true, is there any easy way I can build FreeRTOS inside the PIO project, or, any chances there's a plan for implementing these OS integration in pico's Arduino core?

Thanks for your effort in this project.

earlephilhower, 2022-07-02:
Basic upstream FreeRTOS is already integrated.  Use `#inlcude <FreeRTOS.h>` to enable it in a sketch and the link/etc. will "magically" work.

I'm not an expert. but IIRC the ESP32 basically went SMP FreeRTOS before there was actually support for it, so they forked and made quite a bunch of changes themselves to make it work.  If they have implemented custom primitives, they may not port w/the upstream FreeRTOS SMP (i.e. the `setInterruptAction`).  Actually, I'm not sure what you mean about virtual fcns, as FreeRTOS is very much straight C code so no virtual fcns, classes, etc.

That said, the QueueHandle_t should be supported. You may need to `#include <queue.h>` in your code in addition to `#include <FreeRTOS.h>`

taskx6004, 2022-07-02:
Thanks, by including those two .h files the error in queue part is disappeared. However the `attachInterruptArg` part still prompting errors, by any chance there are also some magic include will work?

earlephilhower, 2022-07-02:
I'm not familiar with that call.  There is an Arduino standard `attachInterruptParam` which we do implement.  You can call it directly, or just add the param to the attachInterrupt call:

https://github.com/earlephilhower/arduino-pico/blob/master/cores/rp2040/api/Interrupts.h

taskx6004, 2022-07-02:
Thanks again, I will do some reading before asking so intimidatingly again, and more tests.

closed 2022-07-04 by earlephilhower