earlephilhower/arduino-pico #674: Second Core not starting after pico probe upload

savejeff, 2022-07-13:
Hi,

For about 2 weeks now, when I upload code through the pico probe and I perform no power cycle the second core is not starting.
I have to to a complete power cycle. then the code runs fine as always.

I use PlatformIO and I don't know if this is a problem with the arduino-pico part or the platformio part. 
@maxgerhardt has something changed in the pico probe mechanism recently?

this is the config I'm using
```
[env:RP2040_BMS_1_ALT]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
platform_packages = framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
board = pico
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
upload_protocol = picoprobe
debug_tool = picoprobe
```

maxgerhardt, 2022-07-13:
I haven't changed anything in the way openocd is invoked, and also the OpenOCD version should have been stable for a while.

I will check what's going on here, I may be either missing a reset command , or the reset command only affects the first core by default, or OpenOCD is missing this functionality, in which case I'd have to open an issue in Raspberry Pi's openocd fork.

savejeff, 2022-07-13:
Thx for the fast response. I think its just that the second core is not restarted after programming. Even when the previously programmed code had the second core running it is not after uploading new code. 

Could you give me a status on the release of the fixes related to the mulicore fixes ? i have tried to follow the release conversation on platform-raspberrypi but i was not transparent to me if the changes are released yet

maxgerhardt, 2022-07-13:
The deadlocks and race conditions were fixed with the latest toolchain version fixes which implement them in the compiler / libc level. If you updated my platform (which you have per comment above), you already have that fixed version.

savejeff, 2022-07-13:
okay thx for the comfirmation! 
totaly love working with the earlephilower code and platformio in general. problems reports always get a fast response and get fixed timely. thx for that ;]
i had a private debug session with Ivan recently and also told him that.

earlephilhower, 2022-07-13:
What is really odd is that I've been using GDB+PicoProb+OpenOCD pretty extensively and never had an issue w/the 2nd core (esp. when debugging the FreeRTOS SMP stuff).  I've had the 2nd core not respond lots of times, but always when I `thread 2` and `where` I can see it's crashed.  Very odd.

Do you have a simple MCVE that consistently shows this?

savejeff, 2022-07-14:
Here is MCVE in an PIO project
[Demo_RP2040_Core1NotStart.zip](https://github.com/earlephilhower/arduino-pico/files/9109072/Demo_RP2040_Core1NotStart.zip)

this is the output right after upload 
```

 .nnot create std::deque larger than max_size()[1001] main0: loop
[1102] main0: loop
[1203] main0: loop
[1303] main0: loop
[1403] main0: loop
[1503] main0: loop
[1603] main0: loop
[1703] main0: loop
[1803] main0: loop
[1903] main0: loop
[2003] main0: loop
[2103] main0: loop
[2203] main0: loop
[2303] main0: loop
[2403] main0: loop
[2503] main0: loop
[2603] main0: loop
[2703] main0: loop
[2803] main0: loop
[2903] main0: loop
[3003] main0: loop
[3103] main0: loop
[3203] main0: loop
[3303] main0: loop
[3403] main0: loop
[3503] main0: loop
[3603] main0: loop
[3703] main0: loop
[3803] main0: loop
[3903] main0: loop
[4003] main0: loop

```

and this is the output after manual reset
```

 .nnot create std::deque larger than max_size()[1033] main0: loop
[1035] main1: loop
[1135] main0: loop
[1137] main1: loop
[1235] main0: loop
[1237] main1: loop
[1335] main0: loop
[1337] main1: loop
[1435] main0: loop
[1437] main1: loop
[1535] main0: loop
[1537] main1: loop
[1635] main0: loop
[1637] main1: loop
[1735] main0: loop
[1737] main1: loop
[1835] main0: loop
[1837] main1: loop
[1935] main0: loop
[1937] main1: loop
[2035] main0: loop
[2037] main1: loop
[2135] main0: loop
[2137] main1: loop
[2235] main0: loop
[2237] main1: loop

```

also notice the first line that is printed to the RP2040 under test FROM the pico probe. it is just reprinted back in the setup code line 107

That is also new i think

earlephilhower, 2022-07-14:
@savejeff can you try #676 and report back?  I was able to repro your failure using the IDE and picoprobe upload, and for some reason the intercore FIFO does not pump the 2nd core unless there's a __wfe() on the 1st core.  (I loaded GDB and connected to a chip w/the failing sketch and saw core 2 was still in ROM...)

savejeff, 2022-07-14:
any idea where that "nnot create std::deque larger than max_size()" comes from ? Why would the pico probe send text to the RP2040 under test over the serial port?

earlephilhower, 2022-07-14:
The picoprobe may upload a small program running out of SRAM to do the upload, probably, but I have not looked into it at all.

savejeff, 2022-07-14:
this kinda breaks my method of waiting for serial input to start the code. but i guess its fixable by waiting some time and flushing before waiting for input.

commit fb05d969 (2022-07-14): Add delay to 2nd core start for Picoprobe (#676)

No idea why, but when a Picoprobe upload and reset is used, the 2nd core
does not start w/o a __wfe() call on the first one before launch.

Fixes #674
Fixes #402
closed 2022-07-14 by savejeff