earlephilhower/arduino-pico #675: I2C Device issue - works with Arduino Mbed OS RP2040 Boards"

yingnanbao, 2022-07-14:


yingnanbao, 2022-07-14:
I also tried "Wire.setSCL()" and "Wire.setSDA()" to use other pins but no luck.

earlephilhower, 2022-07-14:
One gotcha is that raw #s for pins in this repo correspond to the Pico chip GPIO, not the board GPIO.  So my first suggestion would be to triple check the Nano schematic and veriy the raw GPIO pin you passed into setSDA/setSCL is correct (also that the pin you selected is supported for the I2C interface...see https://cdn-learn.adafruit.com/assets/assets/000/099/339/original/raspberry_pi_Pico-R3-Pinout-narrow.png?1612915004 )  Then a logic analyzer would be handy to see the diffs between here and the other core...

yingnanbao, 2022-07-14:
> One gotcha is that raw #s for pins in this repo correspond to the Pico chip GPIO, not the board GPIO. So my first suggestion would be to triple check the Nano schematic and veriy the raw GPIO pin [...]

Thanks for the suggestion. I literally slept over it and woke up realizing it's probably the library is written for Arduino board so I found this "https://github.com/jrowberg/i2cdevlib/tree/master/RP2040".
Now both ROS and IMU are happy, so am I!

closed 2022-07-14 by earlephilhower