earlephilhower/arduino-pico #678: Problem with Arduino Nano rp2040

tjk144, 2022-07-14:
I installed the device drivers following the README. I tried to write a sketch to the Arduino using your driver and the IDE complains that the arduino no longer is connected. I've tried this on windows and the COM port no longer shows up. On Ubuntu lsusb does not list any devices connected. 

I've been trying to enter the bootloader with no success.

Is there a way to restore the arduino to recognize USB connection again?

Andy2No, 2022-07-14:
@tjk144 Did you try Zadig, in Windows?  It's mentioned in the installation document https://github.com/earlephilhower/arduino-pico/blob/master/docs/install.rst from **Windows 7 Driver Notes** onwards, and there's a closed issue that maybe explains it a bit more:

https://github.com/earlephilhower/arduino-pico/issues/96

earlephilhower, 2022-07-14:
Also, you can always hold BOOTSEL while plugging in the device to have it enter ROM-based firmware upload mode.   See the README.  That is necessary whenever the chip crashes, since the usual software-based upload needs the chip running and responding to USB requests.

closed 2022-07-14 by earlephilhower

tjk144, 2022-07-14:
@Andy2No I just tried Zadig like you suggested. Upon plugging in the Arduino I get the Windows prompt that the device is not recognized. When I navigate through Zadig instead of Board CDC to select I have "Unknown USB Device (Device Descriptor Request Failed)" Driver is (NONE) and it doesn't allow me to change the USB ID from 0000 0002 to the value in the tutorial.

When Zadig fails it gives the output: Driver Installation: FAILED (Resource already exists)

tjk144, 2022-07-14:
In addiition, @earlephilhower I tried holding the BOOTSEL before plugging the device in to the WIndows machine -- windows prompts the device is not recognized. I tried to also do that trick with Zadig to no avail.

earlephilhower, 2022-07-14:
I would suggest trying with a different cable or a different USB port.  If, when you plug in the device while holding down the BOOTSEL button (needs to be held until after you plug it in), the drive shows "unknown" there could be a physical problem with the board, the cable, or the USB port.

Andy2No, 2022-07-17:
@tjk144 @fjansson posted a workaround that seems like it might be related to your problem:

https://github.com/earlephilhower/arduino-pico/issues/20#issuecomment-1186508330