earlephilhower/arduino-pico #685: SPI communication with PSRAM fails

hgs12345, 2022-07-17:
There is a problem with SPI. My application of the RPi PICO communicates with a PSRAM. It works fine with the PicoSPI library (except for the minor issue I reported (#655)).
I now want to use the standard SPI library (instead of PicoSPI), but that doesn't work at all! I wrote a small test program, some values are written into the PSRAM and read again (2 versions: with SPI.h or PicoSPI.h). For the standard library I have tried all the possibilities of parameter variations, but there is no success! Only '0' values are read in!
I hope for a hint!
thank you very much

[PSRam.zip](https://github.com/earlephilhower/arduino-pico/files/9128435/PSRam.zip)

earlephilhower, 2022-07-17:
Since this is HW related, and I don't have your HW, can you get a logic analyzer trace of the two different transfer modes?  There may be timing issues (IIRC, PicoSPI writes immediately with larger delays between bytes, SPI uses the HW FIFO and writes in one burst quickly on endTRansaction, etc.).

Also, can you enable the Tools->Debug Port->Serial and Tools->Debug Level->SPI and capture the output?  That will give a trace of the SPI commands as it understands them.

hgs12345, 2022-07-18:
> ------ Originalnachricht ------
> Von "Earle F. Philhower, III" ***@***.***>
> An "earlephilhower/arduino-pico" ***@***.***>
> Cc "hgs12345" ***@***.***>; "Author"
> Datum 17.07.2022 22:07:35
> Betreff Re: [earlephilhower/arduino-pico] SPI communication with PSRAM
> fails (Issue #685)

Hi,
As you can see in the documentation in the attachment, there is an issue 
concerning CS.
I tried to set CS by software (SPI.begin(); instead of 
SPI.begin(true);), but it did not work.
CS stays permanently LOW despite of "digitalWrite(17, HIGH);"
Also, the CLK frequency is wrong on PicoSPI, but that's another story.
Can you fix the problem with CS?
Thank you,
HGS

***@***.***>
***@***.***>

earlephilhower, 2022-07-18:
Thanks for the detailed report.  I'll take a look and see what I can find...

hgs12345, 2022-07-18:
Documentation 
[PSRam-SPI.pdf](https://github.com/earlephilhower/arduino-pico/files/9133819/PSRam-SPI.pdf)

earlephilhower, 2022-07-18:
Thanks again for the detailed updates!

I've got a chance to look at your original code during lunch, and I think it might just be a misunderstanding about `SPI.begin()`.  
https://github.com/earlephilhower/arduino-pico/blob/53cecae10935ae5b7cea8bbbb05289810113b9a5/libraries/SPI/src/SPI.h#L52-L55

Your original code does `SPI.begin()` (i.e. with no param), therefore you selected software-chip select and it's up to you to toggle things.  The `_cs` pin is completely ignored in the library in this mode.  This is kind of the Arduino standard, as even the Arduino-MBED code only supports SW chip-select mode.

If you wanted hardware chip select, just use `SPI.begin(true)`.  Then the PICO SPI  peripheral *will* use the `CS` pin and do it for you.

The change you did to the SPI library should be a no-op.  The pin is untouched, and it's the program's job to do a `pinMode(CS, OUTPUT);` which is the line you added.  Your code w/the SW chip selects should run fine even w./o that change to SPI.cpp (since you do the pinmode before calling SPI.beginTransaction).

So, in summary, your original code with `SPI.begin(true)` should be fine, or the modified version w/the manual `pinMode(); digitalWrite();` should be fine, too (minus the SPI.cpp changes).

closed 2022-07-19 by earlephilhower

hgs12345, 2022-10-11:
Hi,
As you can see in the documentation in the attachment, there is an issue 
concerning CS.
I tried to set CS by software (SPI.begin(); instead of 
SPI.begin(true);), but it did not work.
CS stays permanently LOW despite of "digitalWrite(17, HIGH);"
Also, the CLK frequency is wrong on PicoSPI, but that's another story.
Can you fix the problem with CS?
Thank you,
HGS

hgs12345, 2022-10-11:
Thanks.
Now things are clearer.
But I have a suggestion:
You should mention in the description of SPI.begin(true) that after the 
transmission of each byte CS is set HIGH.
And not only after the completion of a complete transaction.It depends 
on the SPI slave if this is allowed.
In my case (with PSRam) a running transaction must not be interrupted by 
CS=HIGH!
In any case, everything works now!
I have a final question: I think now the maximum SCK frequency is 
limited to 25MHz (at 133MHz CPU frequency).
Is it possible to increase the SCK frequency to around 30MHz?
Thanks again for the prompt replies,
All the best
HGS
>Message ID: 
***@***.***>
>