earlephilhower/arduino-pico #687: Problem using the Ethernet libraries

sstaub, 2022-07-19:
If I include the Ethernet or Ethernet3 library I get a buch of compiler errors.
I tried with the Arduino IDE 2 and PlatformIO using following code:

```
#include "Ethernet.h"

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly
}
```
Here the compiler errors from Arduino IDE 2
```
Using board 'rpipico' from platform in folder: /Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2
Using core 'rp2040' from platform in folder: /Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2
Verwendete Bibliotheken erkennen...
/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06/bin/arm-none-eabi-g++ -c -Werror=return-type -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x2e8a -DUSB_PID=0x000a "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=0 -DLWIP_IPV6=1 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -iprefix/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/ @/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/lib/platform_inc.txt -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/include -fno-rtti -std=gnu++17 -g -w -x c++ -E -CC -DSERIALUSB_PID=0x000a -DUSBD_MAX_POWER_MA=250 -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -Os -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/cores/rp2040 -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/variants/rpipico /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino-sketch-459C1AE824CA2E8C944D22DD3E7A5FE8/sketch/sketch_jul16a.ino.cpp -o /dev/null
Alternativen für Ethernet.h: [Ethernet@2.0.1]
ResolveLibrary(Ethernet.h)
  -> candidates: [Ethernet@2.0.1]
/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06/bin/arm-none-eabi-g++ -c -Werror=return-type -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x2e8a -DUSB_PID=0x000a "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=0 -DLWIP_IPV6=1 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -iprefix/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/ @/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/lib/platform_inc.txt -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/include -fno-rtti -std=gnu++17 -g -w -x c++ -E -CC -DSERIALUSB_PID=0x000a -DUSBD_MAX_POWER_MA=250 -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -Os -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/cores/rp2040 -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/variants/rpipico -I/Users/sstaub/Documents/Arduino/libraries/Ethernet/src /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino-sketch-459C1AE824CA2E8C944D22DD3E7A5FE8/sketch/sketch_jul16a.ino.cpp -o /dev/null
/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06/bin/arm-none-eabi-g++ -c -Werror=return-type -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x2e8a -DUSB_PID=0x000a "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=0 -DLWIP_IPV6=1 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -iprefix/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/ @/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/lib/platform_inc.txt -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/include -fno-rtti -std=gnu++17 -g -w -x c++ -E -CC -DSERIALUSB_PID=0x000a -DUSBD_MAX_POWER_MA=250 -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -Os -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/cores/rp2040 -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/variants/rpipico -I/Users/sstaub/Documents/Arduino/libraries/Ethernet/src /Users/sstaub/Documents/Arduino/libraries/Ethernet/src/Dhcp.cpp -o /dev/null
Alternativen für SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06/bin/arm-none-eabi-g++ -c -Werror=return-type -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x2e8a -DUSB_PID=0x000a "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=0 -DLWIP_IPV6=1 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -iprefix/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/ @/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/lib/platform_inc.txt -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/include -fno-rtti -std=gnu++17 -g -w -x c++ -E -CC -DSERIALUSB_PID=0x000a -DUSBD_MAX_POWER_MA=250 -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -Os -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/cores/rp2040 -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/variants/rpipico -I/Users/sstaub/Documents/Arduino/libraries/Ethernet/src -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/libraries/SPI/src /Users/sstaub/Documents/Arduino/libraries/Ethernet/src/Dhcp.cpp -o /dev/null
/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06/bin/arm-none-eabi-g++ -c -Werror=return-type -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x2e8a -DUSB_PID=0x000a "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=0 -DLWIP_IPV6=1 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -iprefix/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/ @/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/lib/platform_inc.txt -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/include -fno-rtti -std=gnu++17 -g -w -x c++ -E -CC -DSERIALUSB_PID=0x000a -DUSBD_MAX_POWER_MA=250 -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -Os -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/cores/rp2040 -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/variants/rpipico -I/Users/sstaub/Documents/Arduino/libraries/Ethernet/src -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/libraries/SPI/src /Users/sstaub/Documents/Arduino/libraries/Ethernet/src/Dns.cpp -o /dev/null
/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06/bin/arm-none-eabi-g++ -c -Werror=return-type -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x2e8a -DUSB_PID=0x000a "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=0 -DLWIP_IPV6=1 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -iprefix/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/ @/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/lib/platform_inc.txt -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/include -fno-rtti -std=gnu++17 -g -w -x c++ -E -CC -DSERIALUSB_PID=0x000a -DUSBD_MAX_POWER_MA=250 -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -Os -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/cores/rp2040 -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/variants/rpipico -I/Users/sstaub/Documents/Arduino/libraries/Ethe [... cut ...]
/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/cores/rp2040/api/IPAddress.h:58:18: note: declared private here
   58 |         uint8_t* raw_address() {
      |                  ^~~~~~~~~~~

Bibliothek Ethernet in Version 2.0.1 im Ordner: /Users/sstaub/Documents/Arduino/libraries/Ethernet  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: /Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/2.3.2/libraries/SPI  wird verwendet
exit status 1

Compilation error: exit status 1
```

earlephilhower, 2022-07-19:
For the Pico W LWIP port, I included the ESP8266 changes to make IPAddress IPv6 compatible.  For right now you should be able to edit `cores/rp2040/api/IPAddress.h` to make that `raw_address()` function `public:`.  You can also send in a PR, or I'll get one later tonight.

earlephilhower, 2022-07-19:
So there are two problems here:
1. IPAddress lost the `friend ::DNSServer` which #690 reintroduces
2. The Arduino Ethernet library is accessing private IP address fields, which **do not exist** when LWIP 2.x is running in the mode it's configured now (IPv4/IPv6 dual stack).

For #2, the only fix is to either fix the Ethernet library not to twiddle private fields, or provide a way of downgrading to IPv4 only (I think this will fix, but did not dig much).  So I need to think about this.  Probably another menu item and 2 libpico.a versions to keep up to date.  Will make the lwipopts.h header a lot more involved, too...

sstaub, 2022-07-20:
I think you should make dedicated  IPAddress files for IP6.
For an example see https://github.com/Wiznet/Ethernet/tree/DUAL/src
they use IP6Address.h and IP6Address.cpp for extensions of the Wiznet W6100 chip.

earlephilhower, 2022-07-20:
@sstaub can you give #695 a try and report back?  It built the Ethernet class ChatServer example.  Make sure you're IPv4 only,of course.

sstaub, 2022-07-21:
I can test but how to change this line in the platform.ini ?
```
platform_packages =
   framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#2.2.2
```

earlephilhower, 2022-07-21:
I think you could try replacing `#2.2.2` with `#ipv4ipv6`, I think, but I'm not a P.IO user so take it with a grain of salt...

sstaub, 2022-07-21:
Ok, I can compile and upload to a WizNet W500 EVB Pico via PIO.

commit f5cbbcd5 (2022-07-19): Re-add ArduinoCore-API friend classes for IPAddress

Fixes #687
commit 90534f0f (2022-07-21): Enable IP4v4 or IPv4/IPv6 stacks, Ethernet class

IPv4-only mode saves 20KB+ of flash memory.

Add some backwards compatibility with the global Arduino Ethernet
class when running in IPv4 only mode.

Fixes #687
PR #695 (2022-07-21): Enable IPv4 or IPv4/IPv6 stacks, Ethernet class
IPv4-only mode saves 20KB+ of flash memory.

Add some backwards compatibility with the global Arduino Ethernet
class when running in IPv4 only mode.

Fixes #687
commit 0e18f098 (2022-07-21): Enable IPv4 or IPv4/IPv6 stacks, Ethernet class (#695)

IPv4-only mode saves 20KB+ of flash memory.

Add some backwards compatibility with the global Arduino Ethernet
class when running in IPv4 only mode.

Fixes #687

* Speed P.IO build by not cloning 2GB of sources
* Document P.IO new option
closed 2022-07-21 by earlephilhower