earlephilhower/arduino-pico #688: Upload fails if no serial port is set

fjansson, 2022-07-19:
This is related to #4 and #20. I posted a comment on #20 but could not reopen the issue, so I open this one. At the end is some new analysis of why the problem occurs.

From #20:
I'm meeting what seems to be the same issue on Windows 10: a fresh install of the Arduino editor and arduino-pico cannot program a blank Pico or a Pico connected in BOOTSEL mode. 

This can be somewhat difficult to reproduce, because if you had it working once or Arduino at some point connected to a serial port, the com port setting is remembered in the Arduino preferences file and this makes subsequent Pico programming work. Crucially, this issue will hit new users who install the Arduino editor to program a new Pico. 

## Steps
* Fresh install of Arduino 1.8.19 on Windows 10. Using the Windows .exe installer. Allow all drivers the installer proposes.
* Install arduino-pico according to instructions in this repo. I have version 2.3.0
* Select Tools -> Board -> Raspberry Pi RP2040 Boards(2.3.0) -> Raspberry Pi Pico
* Open Blink example
* Connect a Pico with USB in BOOTSEL mode (holding down BOOTSEL while plugging in the USB cable)
* observe Tools > Port menu is gray (at least for me it is, the computer has no other serial ports and Pico in BOOTSEL mode doesn't have a serial port)
* press Upload
  for me compilation works, upload fails with "Serial port not selected". I cannot select a serial port, because the Port menu is gray.

If you connect a Pico already programmed from Arduino in non-BOOTSEL mode, upload works. After this the Arduino editor remembers the serial port in the preferences file. Subsequent uploads also in BOOTSEL mode will work. One way I found to reproduce the problem is to delete Arduino's preferences file (when Arduino is not running). 
This PC > Windows (C:) > Users > <user name> > AppData > Local > Arduino15 > preferences
Then start Arduino, and fill in the arduino-pico Board Manager URL in File>preferences again. Now the port setting is forgotten, as you can verify in the status bar at the bottom of the window. Just reinstalling Arduino seems insufficient, as that doesn't delete the preferences file.

## Workaround

Program the Pico once by drag-and-drop a .uf2 file:
* Open a new, empty sketch (doesn't work with a read-only example sketch). 
* Compile it by pressing the "Verify" checkmark button. 
* Sketch > Export compiled Binary. Select a location e.g. the desktop. A folder is created there, containing a file ending in .uf2
* Copy this file to the Pico's drive, by drag and drop in the Explorer.
 
The Pico restarts and now now has a serial port. Now the Port menu in Arduino is not gray anymore, select the port there. After this, normal uploading from the Arduino editor should work.

## Analysis

The actual uploading to the Pico happens with uf2conv.py which is part of this repository. The Arduino IDE runs this command on upload. The problem is not in uf2conv but in the Arduino IDE constructing the command to launch it. 

Looking at the source code of Arduino 1.8:

* The message "Serial port not selected." is printed from app/src/processing/app/Editor.java, function run(), if SerialNotFoundException is caught. 

* SerialNotFoundException is thrown by runCommand() in arduino-core/src/cc/arduino/packages/uploaders/SerialUploader.java, if serial.port is part of the command string but is not present in the preferences.

This makes it look like SerialUploader won't work if there is no serial port, as in the case when uploading to a blank Pico or a Pico in BOOTSEL mode (on a computer with no other serial ports). If serial.port *wasn't* part of the command string, programming without a serial port would probably work, but then the auto-reset feature wouldn't work. 

From my testing, it seems the serial port in the preferences doesn't need to exist, it just has to be defined so that the string substitution in the run command can finish. So if Arduino remembers a serial port setting from a previous board, uploading to the Pico will work if the Pico is already in BOOTSEL mode.

earlephilhower, 2022-07-20:
That's a great debug and documentation.  But unfortunately it looks all out of our control.  If the IDE won't call the upload process there's not much we can do.

Would you mind me adding your workaround to the README and docs?

fjansson, 2022-07-21:
Thank you! Please use the material for the README and docs.

Some ideas for how it could be fixed:

* Perhaps there would be interest in the Arduino project for a uf2 uploader. My impression is that several boards use the same pattern of uf2 upload and optional serial port reset. 

* You could define a new board "Raspberry Pi Pico (no serial)" which would not have "{serial.port}" in the command string. Then it should work. Drawbacks are that it's still not obvious to find, and the board list gets even longer if this variant is added for all boards. Perhaps a generic such board would be enough.

* Do you have any control over the Burn Bootloader menu command? Could that be used to upload an .uf2 of an empty sketch, which would enable the serial port?

I don't know what the situation is for Arduino 2.0. Perhaps the uploader logic is different there?

WaGi-Coding, 2022-07-21:
Sorry if thats may not related, but i struggled with uploading code to my Pico W. I then installed the WinUSB Driver for the Pico with a tool called "Zadig", then it worked fine.

Andy2No, 2022-07-21:
@fjansson In Arduino 1.8.19, which I'm still using, the preferences section for the serial port is near the end.   Here's a section of it that's probably the most relevant:

```
serial.port=COM49
serial.port.file=COM49
serial.port.iserial=null
serial.port.label=COM49
serial.port.protocol=serial
```

Maybe just deleting the whole **serial.port...** block would be enough, rather than deleting the whole preferences file.

@WaGi-Coding I gather it's not related to this particular problem, but that is problem I initially had with the Pi Pico too, so anyone reading this while trying to solve an upload problem should definitely try that.  There's a closed issue about it, with instructions - #96 - I don't think it's limited to just Windows 7, as the title suggests.

closed 2022-07-21 by earlephilhower

fjansson, 2022-07-21:
@Andy2No yes deleting just those lines from preferences.txt is enough to reproduce the issue. Conversely, putting those lines in might fix it, even if the port mentioned doesn't exist.

@earlephilhower I just tested Arduino Mbed OS RP2040 under the same conditions, and that board *can* be programmed in BOOTSEL mode even if no serial port exists. The difference is that they use another tool to upload, and don't pass a serial port as an argument. The tool is this: https://github.com/arduino/rp2040tools and picotool. Serial-port reset still works with that board, if the correct port is selected in the menu. I don't understand how the upload tool knows which serial port is selected. But anyway this shows that it should be possible to have serial reset and also work without serial ports.

PR #699 (2022-07-21): Uploading for the first time docs from @fjansson
Fixes #688
commit a97dc38e (2022-07-21): Uploading for the first time docs from @fjansson (#699)

Fixes #688