earlephilhower/arduino-pico #689: Floats in print stream cause processor crash in v2.3.x

Bodmer, 2022-07-19:
Using the print stream with floats causes the processor to crash, for example:

Serial.println(1234.56);

or 

tft.println(1234.56);

This affects v2.3.0, v2.3.1 and v2.3.2. Works OK in v2.2.2

earlephilhower, 2022-07-19:
I am unable to reproduce with `git` version:
````
void setup() {
Serial.begin();
}

void loop() {
  Serial.println(1234.56);
  delay(1000);
}
````
and
````
void setup() {
Serial.begin();
}

void loop() {
  Serial.printf("%f\n", 1234.56);
  delay(1000);
}
````
all work fine, no crashes.  I'll give the boards manager versions a try next...

earlephilhower, 2022-07-19:
I take it back.  I deleted my `dist` directory and the `system` tools dir and re downloaded and installed the tools and am seeing the fault.  The toolchain in question is not the latest one w/the printf mutex fix.

I need to figure out whether this was a `make` problem or a a packaging one...

earlephilhower, 2022-07-19:
I take back my "I take it back" statement.  I still cannot reproduce a failure or a crash in 2.2.2 or 2.3.2 while swapping between them in the boards manager (my problem before was I picked the wrong serial port to watch in the IDE, my PicoProbe, not the downloaded Pico).  I tried both the IDE and using the git version, and running under GDB, and it always seems to work.

It could crash if you have updated header files but older toolchain.  Can you select File->Preferences, and then check "Enable verbose output during compile"?  And then build from scratch and post the compile logs?

Also, I'd recommend deleting all versions and temporary downloaded versions of the core.  Quit Arduino and then deleting the cached downloads in C:\Users\{username}\AppData\Local\Arduino15\staging/packages and the package_rp2040_index.json in C:\Users\{username}\AppData\Local\Arduino15\staging/packages and the installed dir in C:\Users\{username}\AppData\Local\Arduino15\packages/rp2040 , and then restarting and reinstalling the core.

Bodmer, 2022-07-20:
Here's the compile output
[compile_log.zip](https://github.com/earlephilhower/arduino-pico/files/9148576/compile_log.zip)

Bodmer, 2022-07-20:
Cleaned system as described and still get the problem.

Bodmer, 2022-07-20:
I switched on Debug Port: Serial and get this message:

```
Hello

*** PANIC ***

missing double function
```

With this sketch:
```
void setup() {
Serial.begin(115200);
delay(10000);
Serial.println("Hello");
delay(1000);
}

void loop() {
  Serial.println(1234.56);
  delay(1000);
}
```

earlephilhower, 2022-07-20:
The SDK and your version of the chip are not friends, it seems.  I bet you got that chip very early and it has ROM 1.0.  That message is generated by the latest Pico-SDK, not the core.  See `src/rp2_common/pico_double/double_init_rom.c`.

I might be able to add some new `define`s to the build, let me see.

Bodmer, 2022-07-20:
It is indeed an original release RP2-B0 chip. I tried a newer RP2-B2 chip and it is working fine with v2.3.2.

earlephilhower, 2022-07-20:
#693 should get you running again.  Can you give it a try?

Bodmer, 2022-07-20:
Thanks, yesit works and I am up and running again!

PR #693 (2022-07-20): Support original Pico with ROM B0
The SDK disabled float/double support for some functions by default on the
original B0 ROMs.  Manually re-enable it.

Fixes #689
closed 2022-07-20 by earlephilhower