earlephilhower/arduino-pico #691: WiFiClientSecure compatibility

zpm1066, 2022-07-20:
BearSSL WiFiClientSecure & WiFiServerSecure works fine with Pico W. Fantastic progress with porting WiFi libraries to Pico W. Much appreciated!

For those of us who also use the ESP32 WiFiClientSecure, Do you plan to support it for the Pico W? 

It would be good if existing ESP32 code using WiFiClientSecure could be used with the Pico W.

earlephilhower, 2022-07-20:
I've not used the ESP32 WiFi*Secure classes, but it looks like the API could mostly be implemented simply by adding `const char *cert` options to the existing calls which take things like `X509Cert`s.  This would be a good PR for someone to get their feet wet.

Basically they'd need to create temp objects from the char* passed in, and make sure to free them on connection close/etc.  Mostly housekeeping work, really...

zpm1066, 2022-07-21:
Thanks @earlephilhower. I'll review once I get one of my Zephyr projects out of the way.
Great work on the arduino-pico core, including the Pico W support. Only the RSSI seems to be a to-do.

brookers2022, 2022-07-22:
Amazing work on this, I too would be interested in having SP32 WiFiClientSecure working with the Pico W - this would be a huge addition 👍 Thanks for all the work that's  been done so far.

closed 2022-07-23 by earlephilhower

zpm1066, 2022-07-31:
@earlephilhower, Thank you for adding ESP32 WiFiClentSecure support.

commit 24e28767 (2022-07-23): Add basic ESP32 WiFiClientSecure support

Simple sketches should work without modification, but some modes (listed
in the docs) are not possible to support on the Pico W with BearSSL.

Fixes #691
PR #704 (2022-07-23): Add basic ESP32 WiFiClientSecure support
Simple sketches should work without modification, but some modes (listed
in the docs) are not possible to support on the Pico W with BearSSL.

Fixes #691
commit cfc05dab (2022-07-23): Add basic ESP32 WiFiClientSecure support (#704)

Simple sketches should work without modification, but some modes (listed
in the docs) are not possible to support on the Pico W with BearSSL.

Fixes #691