earlephilhower/arduino-pico #703: M1 OpenOCD Upload Issues

eosti, 2022-07-22:
I'm using PlatformIO as described [here](https://github.com/platformio/platform-raspberrypi/pull/36) on an M1 MacBook and run into an issue when trying to upload using the `picoprobe` option. It first cannot find `libhidapi`:

```
Uploading .pio/build/pico/firmware.elf
dyld[50922]: Library not loaded: /usr/local/opt/hidapi/lib/libhidapi.0.dylib
  Referenced from: /Users/eosti/.platformio/packages/tool-openocd-rp2040-earlephilhower/bin/openocd
  Reason: tried: '/usr/local/opt/hidapi/lib/libhidapi.0.dylib' (no such file), '/usr/local/lib/libhidapi.0.dylib' (no such file), '/usr/lib/libhidapi.0.dylib' (no such file)
*** [upload] Error -6
```

I have `libhidapi` installed using brew, which installs to /opt/homebrew. So to link it to a place where OpenOCD will look for it, I ran `sudo ln -s /opt/homebrew/lib/libhidapi.0.dylib /usr/local/lib/libhidapi.0.dylib`. Now, a different error, this time about architecture:

```
dyld[51483]: Library not loaded: /usr/local/opt/hidapi/lib/libhidapi.0.dylib
  Referenced from: /Users/eosti/.platformio/packages/tool-openocd-rp2040-earlephilhower/bin/openocd
  Reason: tried: '/usr/local/opt/hidapi/lib/libhidapi.0.dylib' (no such file), '/usr/local/lib/libhidapi.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/lib/libhidapi.0.dylib' (no such file)
*** [upload] Error -6
```

Finally, when I replace the openocd bin in `~/.platformio/packages/tool-openocd-rp2040-earlephilhower/bin/openocd` with the bin that I compiled separately, it works properly. 

I've also found that by only doing the OpenOCD swap (and not having the `libhidapi` link) everything works as expected. So it looks like the packaged version of OpenOCD is the culprit since using a properly compiled one resolves the `libhidapi` issue.

earlephilhower, 2022-07-22:
You need to install the X86 version of libhidapi.so.  Your symlink isn't going to work because the OpenOCD.exe file is an x86-64 version.  Try something like:  (warning, I don't own or use a Mac)
````
arch -x86_64 brew install libhidapi
````

closed 2022-07-22 by earlephilhower

maxgerhardt, 2022-07-24:
Can we package libhidapi with the Mac OS package? Because looking at https://dl.registry.platformio.org/download/platformio/tool/tool-openocd/2.1100.211028/tool-openocd-darwin_x86_64-2.1100.211028.tar.gz, it has

[image]

it does it like that too.

earlephilhower, 2022-07-24:
I don't build anything on the Mac, and I don't think the cross-compiler can actually do a static link.  For the `dynlib`, do I just need to add a `libexec` directory and the Mac or the IDE handles making sure it's in the LDPATH or whatever the Mac equivalent is?  Or do you know if there's some other magic required?

maxgerhardt, 2022-07-24:
@ivankravets do you know how this works with the Mac's OpenOCD' package, are the dynlibs in `libexec/` automatically loaded without doing anything else?

ivankravets, 2022-07-25:
PlatformIO handles this automatically if compiled libraries are located in the `lib` folder, see https://github.com/platformio/platformio-core/blob/develop/platformio/builder/tools/pioplatform.py#L91

But! This issue has another root. The package is marked compatible with x86/ARM, where sources are compiled for the x86. It means users should have Rosetta installed on their machine. The alternative is to compile separate packages for x86 and ARM.

maxgerhardt, 2022-07-28:
This also [just appeared](https://community.platformio.org/t/how-to-debug-the-basic-blink-program-on-a-rapsberry-pi-pico-using-platformio-and-picoprobe/28981/3?u=maxgerhardt) on a Intel-based Mac book, the binary fails to run because of non-installed dependencies. I guess I'll try and modify https://github.com/earlephilhower/pico-quick-toolchain/ to package OpenOCD with the libexec folder and dylibs like PlatformIO does and hope this problem goes away, with the help of someone with a MacBook to test it..

earlephilhower, 2022-07-28:
@maxgerhardt it's actually easier than that.  The MacOSX OpenOCD binary is a prebuilt blob because the USB dependencies were difficult to get working in a cross -compiled environment.

I can pull down the working official OpenOCD and copy its `libexec` folder, but I can't test it.  And since it's P.IO I would need to do a new complete release, which would break everyone's toolchain (when the current one disappears) so I'd like to test first.

maxgerhardt, 2022-07-28:
If you can get the PIO-packaged `.tar.gz` for  openocd, then I can forward that to the person in the forum who can just use it with 
```ini
platform_packages = tool-openocd-rp2040-earlephilhower@file://<path to .tar.gz>
```

earlephilhower, 2022-07-28:
I've added in the libexec from the darwin-x86 official openocd, attached below (GH doesn't allow .tgz, only .zip):
[osx-openocd-test.zip](https://github.com/earlephilhower/arduino-pico/files/9212898/osx-openocd-test.zip)

fietser28, 2022-09-29:
I've tested the zip and above with PIO (in VSCode) on OS X on M1 without success:

```
Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; board: pico; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing file:///Users/jan/tmp/PicoDebugTest/tools/osx-openocd-test.zip
Unpacking  [####################################]  100%
Tool Manager: tool-openocd-rp2040-earlephilhower@5.100300.220728 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
```
......
```
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picoprobe, picotool, raspberrypi-swd
CURRENT: upload_protocol = picoprobe
Uploading .pio/build/pico/firmware.elf
dyld[63554]: Library not loaded: '/usr/local/opt/hidapi/lib/libhidapi.0.dylib'
  Referenced from: '/Users/jan/.platformio/packages/tool-openocd-rp2040-earlephilhower/bin/openocd'
  Reason: tried: '/usr/local/opt/hidapi/lib/libhidapi.0.dylib' (no such file), '/usr/local/lib/libhidapi.0.dylib' (no such file), '/usr/lib/libhidapi.0.dylib' (no such file)
*** [upload] Error -6
```

Relevant platform.ini lines:
```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino

upload_protocol = picoprobe
debug_tool = picoprobe

platform_packages = tool-openocd-rp2040-earlephilhower@file:///Users/jan/tmp/PicoDebugTest/tools/osx-openocd-test.zip
```

Running openocd from the command line with the libexec path of the package added to DYLD_LIBRARY_PATH works. So all the libraries are there but they are not found automatically. This can be made dynamic with:
```
export DYLD_LIBRARY_PATH=@executable_path/../libexec/
```
I don't know how to set this in platformio to get it all working.

maxgerhardt, 2022-09-29:
Well PlatformIO adds `<unpackaged package path>/lib` to the LYLD_LIBRARY_PATH, not `libexec`, so the package seems to be using the wrong folder names there.. Can you `pio pkg unpack`, rename the folder, `pio pkg pack` again and retry?

fietser28, 2022-09-29:
I tried did and it doesn't work. 

I noticed the error message doesn't claim to have searched the <unpackaged package path>/lib directory, so I think it is not set somehow:
```
Uploading .pio/build/pico/firmware.elf
dyld[68445]: Library not loaded: '/usr/local/opt/hidapi/lib/libhidapi.0.dylib'
  Referenced from: '/Users/jan/.platformio/packages/tool-openocd-rp2040-earlephilhower/bin/openocd'
  Reason: tried: '/usr/local/opt/hidapi/lib/libhidapi.0.dylib' (no such file), '/usr/local/lib/libhidapi.0.dylib' (no such file), '/usr/lib/libhidapi.0.dylib' (no such file)
*** [upload] Error -6
```

So I tried the following (/nowhere is deliberately not existing):
```
jan@fozzie PicoDebugTest % export DYLD_LIBRARY_PATH=/nowhere       
jan@fozzie PicoDebugTest % ~/.platformio/packages/tool-openocd-rp2040-earlephilhower/bin/openocd 
dyld[68476]: Library not loaded: '/usr/local/opt/hidapi/lib/libhidapi.0.dylib'
  Referenced from: '/Users/jan/.platformio/packages/tool-openocd-rp2040-earlephilhower/bin/openocd'
  Reason: tried: '/nowhere/libhidapi.0.dylib' (no such file), '/usr/local/opt/hidapi/lib/libhidapi.0.dylib' (no such file), '/usr/local/lib/libhidapi.0.dylib' (no such file), '/usr/lib/libhidapi.0.dylib' (no such file)
```
Now the path is shown in the error message.

I'm on OS X 12.5.1 on an M1 and it seems the LYLD_LIBRARY_PATH is not a normal environment variable. E.g.:
```
jan@fozzie PicoDebugTest % export DYLD_LIBRARY_PATH=/nowhere
jan@fozzie PicoDebugTest % env |grep LIBRARY                
jan@fozzie PicoDebugTest % echo $DYLD_LIBRARY_PATH          
/nowhere
jan@fozzie PicoDebugTest % 
``` 
This is not a 'normal' variable behaviour to me. The OS has some 'feature' here.

maxgerhardt, 2022-09-29:
Well, if you just want to get a working environment for now and no experimenting, better use [that posted command](https://github.com/earlephilhower/arduino-pico/issues/703#issuecomment-1192974035) to install the missing lib, maybe we'll figure something out later.

fietser28, 2022-09-30:
I installed the library in /usr/local/... and am in business for now. Feel free to ask if I need to test something.