earlephilhower/arduino-pico #706: Error at second build in the Arduino IDE

board707, 2022-07-24:
Hi

My config:
arduino-pico v2.3.2
Arduino IDE 1.8.3
Windows7

After opening the Arduino IDE, the first build of the sketch goes fine, but the second and all subsequent builds fail. The error diagnosis is as follows:
`arm-none-eabi-g++: error: D:\Users\DDU\AppData\Local\Temp\arduino_build_559677/..\arduino_cache_297517\core\core_rp2040_rp2040_rpipico_flash_2097152_0,freq_133,opt_Small,rtti_Disabled,stackprotect_Disabled,exceptions_Disabled,dbgport_Disabled,dbglvl_None,usbstack_picosdk_a86a8d467eb51e3fb26ce70ddc0c2805.a: No such file or directory
exit status 1`
I checked - this file is there, but the IDE does not find it. I investigated the situation and came to the conclusion that the reason is the length of the file path. In this case, the length is 281 characters, while on Windows7 the maximum path length is 260 characters and cannot be increased.
The filename becomes so long because the Arduino IDE creates it by concateating together all the options for a given board. In your package, the boards have a lot of options, for example, the Raspberry Pico has 12 of it. All of them are listed in the name:, as you can see at the error message above.
This is not a problem for Windows10 and Linux users, because these OS support longer paths and filenames, but can cause troubles for Win7 users as me.

earlephilhower, 2022-07-24:
Thanks for the description and background, but I think this is related to the version of the IDE you're running and there is no issue with the latest version.

1.8.3 was released in 2017, so over 5 years old at this point.

1.8.19 is available now from https://arduino.cc and it names the core file with just a hash, and does not list all the menu items.  That will drop 100 chars+ off of your filename, and I think avoid the issue completely.

For example, the `core.a` cached version for me is called `core_091639640ace3aac68c28c72463bf28f.a`, with no prefixes other then a path.

Can you please update to the latest IDE and see if it still fails?

board707, 2022-07-24:
Thanks for your reply. 
I tested the issue on Arduino IDE version 1.8.13 and for sure, there is no such problem.
I'm sorry for worrying you for nothing.
Dmitry

closed 2022-07-24 by board707