earlephilhower/arduino-pico #714: I2S: wrong frequency?

giampiero7, 2022-07-26:
Hi,
I'm testing the I2S library with a MEMS microphone, detecting the dominant frequency within sets of samples.
As results I'm getting 2x the expected frequency.
I had a quick look at the code:
https://github.com/earlephilhower/arduino-pico/blob/master/libraries/I2S/src/I2S.cpp#L81
and compared it to the Micropython code this lib is based on:
https://github.com/micropython/micropython/blob/master/ports/rp2/machine_i2s.c#L577
It looks like the calculated pio clock differs of a factor of 2.
As expected, if I double the frequency I get the correct results.
Is it possible there is a mistake?

Thank you!

earlephilhower, 2022-07-26:
The I2S code here assumes stereo input and output, so in that case the calculation would need to have 2x the # of bits (one for right, one for left).  I don't think I support mono, how are you driving the MEMS?  The WL output should be toggling high and low...

giampiero7, 2022-07-26:
I'm using the read32(l, r) method and only using the left samples. It works fine it seems.
A word clock period should include both L and R samples, so there should be no difference between mono and stereo wrt frequency (sample rate), no?

earlephilhower, 2022-07-26:
You've highlighted the bit clock, from which everything else is derived.

The frequency of the bit clock will need to be 2x if you are in stereo.  For example, if you have 44.1k, 16b, 1-channel then you need Fbit = 44.1K*16 = 705.6Khz bit clock.  But if you want stereo, you need to get 2x the number of samples out every period, so you need Fbit = 44.1K * 32b = 1.411MHz.

More simply, if the I2S was driving clocks 2x as fast as they should be, the audio out would be seriously messed up, which is not the case in my own testing.

Are you connecting the LRCLK to the microphone, or maybe just tying off its pin so it sends out every ?  Could there be data in the right sample you're dropping?

giampiero7, 2022-07-27:
As far as I understand the difference between mono and stereo is that stereo uses both the high and low side of the LRCLK, while mono uses only one. Mono is simply one of the two channels of a stereo configuration.
So the bit frequency should be the same in both cases.

Yes, I have the LRCLK connected to the mic and no data is in the right samples. If I were to connect a "right" mic I should get stereo stream without changing the code.

I've highlighted the bit clock, just because that's the first thing that caught my attention, as it differs from the Micropython  code with exactly the factor I was looking for... The issue might be somewhere else.
The Micropython library seem to work correctly in my case.

earlephilhower, 2022-07-27:
>  So the bit frequency should be the same in both cases.

Yes and no.  Each channel would get 44.1K samples/second, but the frequency that's being set is the bitclock, which need to pump in 16 bits for the left side and 16 bits for the right side.  So LRCLK = 44.1MHz, BITCLK = (16+16) * LRCLK.

In any case, maybe it's something else and not the clocks (double DMA or other weirdness).  Can you make a small sample that shows you getting 2x the samples/sec so I can run this weekend w/a logic analyzer and check the bus as well as the internal DMA stuff?

giampiero7, 2022-07-28:
Here it is:

```
#include <I2S.h>

I2S ics43432 = I2S(INPUT);

int bufferSamples = 10000;
int buffersN = 4;

long sampleRate = 40000;

int32_t l, r;
long ts;
long count;

void setup() {
  Serial.begin(9800);
  while(!Serial);

  ics43432.setBCLK(6);
  ics43432.setDATA(8);

  if (!ics43432.setBitsPerSample(32)) {
    Serial.println("ERROR: setBitsPerSample");
  }
  if (!ics43432.setBuffers(buffersN, bufferSamples / buffersN)) {
    Serial.println("ERROR: setBuffers");
  }
  if (!ics43432.begin(sampleRate)) {
    Serial.println("ERROR: begin");
  }

  Serial.println("Ready");

  ts = millis();
  count = 0;
}

void loop() {
  if (millis() - ts > 1000) {
    Serial.print("Samples/sec: ");
    Serial.println(count);
    count = 0;
    ts = millis();
  }
  if (ics43432.read32(&l, &r)) {
    count++;
    if (r != 0) {
      Serial.println("ERROR: R data");
    }
  } else {
    Serial.println("ERROR: read");
  }
}
```

I'm getting around 20000 samples/sec instead of the expected 40000, which is consistent with getting 2 x the dominant frequency in my original test.

The code also checks for no data on the R channel, which works as expected.

By the way, how to best choose number of buffers and buffers size? The doc doesn't say much about it.

Thank you!

earlephilhower, 2022-07-29:
Thanks for the simple sketch.  I'll try and find my clone LA and take a gander under the hood this weekend!

earlephilhower, 2022-07-29:
So on further code inspection I think the bitclock is the problem after all...it's not too fast, but too slow!  Maybe that's what  you were saying before (I thought you meant `*2.0` was too much, to too little), so apologies if I misunderstood.

The code sets the PIO clock to (bps * freq * 2 channels), **but** the PIO state machine needs 2 cycles to generate a single bitclock period!  (i.e. there is a set0 insn followed by a set1 insn).  So the PIO is running at bit clock, but the pin is only toggling at 1/2 that rate...

closed 2022-07-29 by earlephilhower

giampiero7, 2022-07-30:
Exactly, which was the `* PIO_INSTRUCTIONS_PER_BIT` (2) in the micropython code I suspected was missing.
Thank you for putting your time into this!

giampiero7, 2022-07-30:
On another note, how to best choose number of buffers and buffers size? The doc doesn't say much about it, does it make any difference?

earlephilhower, 2022-07-30:
TBH, I never looked at the MicroPython C++ code, only the PIO program because they had a nice way of handling the 1-bit-delay from WL to bit[32].  I was scratching my head a long time, and they had a very elegant solution.  I like my SW-configurable bit widths, though. :)

For buffers, it's up to how long you may go between processing them.  Because you can't use the buffer while DMA is using it, you need more than a couple to make sure you don't under/overrun.  They don't really add latency beyond a fixed 1-buffer latency.

PR #728 (2022-07-29): Fix I2S timing
BCLK was running at 50% of expected speed.

Fixes #714
commit 5ab19e9a (2022-07-29): Fix I2S timing (#728)

BCLK was running at 50% of expected speed.

Fixes #714