earlephilhower/arduino-pico #718: Ignore DTR on Serial USB

lu7did, 2022-07-28:
Hello,

I've to deal with programs that do not raise DTR and thus the rp2040 is not sending data to them. Is there any way to set the DTR value on the SerialUSB object?

Thanks in advance, Pedro

kholia, 2022-07-28:
Possibly related: https://github.com/hathach/tinyusb/discussions/932.

earlephilhower, 2022-07-29:
Right now, that is not supported by the Pico-SDK USB driver.  You may want to follow up on the issue @kholia listed in the TinyUSB repo, and if you have specific tests/etc. to help hathach understand and reproduce that would be very helpful.

For now, this isn't supported in the Pico-SDK which is what we use for basic USB processing, sorry,

closed 2022-07-29 by earlephilhower