earlephilhower/arduino-pico #720: Serial.print with Adafruit_TinyUSB problem

s-t-e-p-a-n-o-v, 2022-07-28:
There is a problem with Serial data transfer with Adafruit_TinyUSB.
Data goes to PC only on even loop iteration, odd iteration values are stored and concatenated.
Tested with Arduino 1.8.15 and 1.8.19, arduino-pico 2.2.0 and 2.3.3, behavior is the same.

```
#include <Adafruit_TinyUSB.h>

int led = 25;

// the setup routine runs once when you press reset:
void setup() {
  // initialize the digital pin as an output.
  pinMode(led, OUTPUT);
  Serial.begin(115200);
}

// the loop routine runs over and over again forever:
void loop() {
  Serial.println("blink!");
  digitalWrite(led, HIGH);   
  delay(200);               
  digitalWrite(led, LOW);    
  delay(200);               
}
``
![settings](https://user-images.githubusercontent.com/17668205/181439712-9da733a5-cefd-478e-b7c1-05d9a9bf7f70.png)
![output](https://user-images.githubusercontent.com/17668205/181439722-d8ab6848-2a24-4507-b1c9-2bce514a89f7.PNG)

https://user-images.githubusercontent.com/17668205/181439772-e2bdad74-3d59-40aa-862b-a585e3102858.mp4

`

maxgerhardt, 2022-07-28:
I've seen it before in https://github.com/earlephilhower/arduino-pico/issues/642#issuecomment-1176799088 that the Arduino-IDE uses a previously installed TinyUSB library version, and not the one in the core. Can you double check your Arduino IDE library manager and uninstall any TinyUSB versions?

earlephilhower, 2022-07-29:
This library includes the Adafruit_TinyUSB library, but makes no changes to it.  I think this is something you need to report at their repo.  The Pico-SDK TinyUSB we use by default does not show this issue in tests I just tried.

s-t-e-p-a-n-o-v, 2022-07-29:
Reinstalled Arduino 1.8.19 and arduino-pico 2.3.3 to fresh installed system and it still work as mentioned above.

Pico-SDK USB stack works fine, but I need a dual virtual port option, that is why I'm looking for Adafruit_TinyUSB library.

Also I tried to install latest Adafruit TinyUSB library 1.14.1 and Mbed OS RP2040 3.2.0 and this combination works fine.
I compared files hashes of src folders in standalone TinyUSB library 1.14.1 and in arduino-pico 2.3.3 and it is the same.
So it's unclear where difference come from.

s-t-e-p-a-n-o-v, 2022-08-02:
Some additional investigations:

cores/rp2040/main.cpp
__loop() called after every loop()
```
        while (true) {
            loop();
            __loop();
        }
```

__loop() calls yield() in case of USE_TINYUSB
```
extern void __loop() {
#ifdef USE_TINYUSB
    yield();
#endif
```

cores/rp2040/delay.cpp
yield() calls TinyUSB_Device_Task() and TinyUSB_Device_FlushCDC()
```
    void yield() {
#ifdef USE_TINYUSB
        TinyUSB_Device_Task();
        TinyUSB_Device_FlushCDC();
#endif
```

So after every loop iteration TinyUSB_Device_FlushCDC is called and data should be transferred.

I tried to add 1ms delay after FlushCDC and in this case data transferred successfully every loop cycle  

```
    void yield() {
#ifdef USE_TINYUSB
        TinyUSB_Device_Task();
        TinyUSB_Device_FlushCDC();
        delay(1);
#endif
```

Looks like problem can be in tud_cdc_n_write_flush() (src/class/cdc/cdc_device.c) there are some checks at the beginning which can skip data transfer. @hathach could you provide help at this point ?
```
  // Skip if usb is not ready yet
  TU_VERIFY( tud_ready(), 0 );

  // No data to send
  if ( !tu_fifo_count(&p_cdc->tx_ff) ) return 0;
```

s-t-e-p-a-n-o-v, 2022-08-02:
delay(1); was added after TinyUSB_Device_FlushCDC() not before.

Is it possible that there is some kind of race conditions and Serial.println("blink!") at the beginning of loop does not actually transfer data to usb tx buffer (not ready yet) ?

karelv, 2022-08-19:
I compiled on a windows 10 PC that had never Arduino installed before. ==> issue confirmed.

earlephilhower, 2022-08-31:
This is just a difference with the upstream Adafruit TinyUSB:

https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/5be0b21c0a5cf396b139bce0c6c0d403df051b0a/src/arduino/Adafruit_USBD_CDC.cpp#L219-L237

Compared to the core's ::write
https://github.com/earlephilhower/arduino-pico/blob/d2beb2da194e05e9213d348dc9f1f95504584c8d/cores/rp2040/SerialUSB.cpp#L119-L155

The core is calling the USB flush command after every write.  The Adafruit library is only calling flush (inside the core implementation of `yield()`) in the case when the USB buffer is full and it can't fit more data.

Adafruit's is less CPU intensive and higher bandwidth, but at the cost of latency (the host will poll the USB eventually to read back the buffers, but you may get multiple write()s in a single USB report).

There's not really anything we can do in this core, the Serial::write call is handled completely in the library.

If you *really* need each report to be separate, add a call to `Serial.flush()` after the print to ensure it's sent.  That works everywhere and is very explicit about what you're trying to do.

closed 2022-08-31 by earlephilhower