earlephilhower/arduino-pico #735: Hardware SerialUART available( ) sometimes returns wrong number

Haggarman, 2022-08-04:
I have a sketch that uses Serial2 to send and receive characters on the hardware pins at 115200 baud.
The program increases the FIFO size before calling Serial2.begin( ) with the Serial2.setFIFOSize(200); method.

The issue is, depending on how many characters are received and the fifo size, the Serial2.available( ) function call quite often reports the wrong number of characters in the buffer. I think this is due to how in a FIFO, the write index wraps around to the beginning and the read index ends up being a higher number?

So for this example, please forgive how busy this is about to look.
I am printing the ASCII Character, followed by the size reported by .available in brackets [ ] to the arduino terminal.
Basically echoing back what I read from Serial2 over to Serial.

For example, the first A[4] means that "A" is the next character read from the serial buffer, followed by a [4] in brackets indicating what the return value from .available() was just before reading. The value 4 is to my understanding is how many characters are available to be read in the FIFO buffer. Lines with just the brackets are linefeeds or carriage returns.

Below you can see the Serial2 receive is slightly outpacing how fast it can print to the Arduino terminal. The number is increasing as it is receiving the characters 'Manufacturer: u-blox' increasing from [3] up to [15] ). This part is normal and expected.

### Issue: Serial2.available( ) suddenly jumps from [53] characters to [154] characters, and then later on from [123] characters to [22] characters.

More serious, depending on how things align, the number of characters left reported by available( ) can be 0, even though the serial receive FIFO definitely has more characters in it, causing my program to get stuck. This example only looks nice because i chose to use a FIFOsize of 200, to hopefully make this as easy as possible to see what I'm seeing.

A[4] T[3] I[3] 
[3] 
[2] 
[2] M[3] a[3] n[2] u[3] f[4] a[4] c[4] t[4] u[5] r[6] e[7] r[8] :[9]  [9] u[10] -[10] b[11] l[12] o[13] x[15] 
[15] 
[16] M[17] o[19] d[20] e[20] l[20] :[21]  [22] S[24] A[25] R[26] A[27] -[28] R[29] 4[31] 1[32] 0[32] M[32] -[32] 0[33] 2[34] B[36] 
[37] 
[37] R[38] e[39] v[41] i[42] s[43] i[43] o[44] n[46] :[47]  [48] L[49] 0[50] .[51] `0[53] .[154]` 0[155] 0[155] .[156] 0[158] 0[160] .[161] 0[161] 5[162] .[163] 1[164] 1[164]  [163] [[162] J[161] u[160] n[159]  [158] 0[157] 8[156]  [155] 2[154] 0[153] 2[152] 0[151]  [150] 1[149] 9[148] :[147] 0[146] 9[145] :[144] 0[143] 8[142] ][141] 
[140] 
[139] S[138] V[137] N[136] :[135]  [134] 0[133] 4[132] 
[131] 
[130] I[129] M[128] E[127] I[126] :[125]  [124] `3[123] 3[22]` 3[21] 3[20] 3[19] 3[18] 3[17] 3[16] 3[15] 3[14] 3[13] 3[12] 3[11] 2[10] 2[9] 
[8] 
[7] 
[6] 
[5] O[4] K[3] 
[2] 
[1]

I believe this to be an issue in SerialUART.cpp where it says 

`295  return (_writer - _reader) % _fifoSize;`

The issue is that _writer can wrap and become a lower number than _reader, and in theory the modulo % should work but in practice it is very difficult to get it to behave with any sense. Due to the unpredictability of signed and unsigned numbers in C++, this is kind of hard to correct with a one-liner.

Here is my best shot at this?

`if (_writer >= _reader) {`
`return _writer - _reader;`
`} else {`
`return _fifoSize + _writer - _reader;`
`}`

Haggarman, 2022-08-05:
for the purpose of discussion, here is the least amount of code that produces the example result above.

```
#define TERM Serial

void setup() {
    TERM.begin(115200);

    Serial2.setFIFOSize(200);
    Serial2.begin(115200);
    //other modem init stuff
}

void loop() {
    int SARA_byte;
    int SARA_delta;

    while ( (SARA_delta=Serial2.available()) > 0) {
      SARA_byte = Serial2.read();
      TERM.write(SARA_byte);
      TERM.print("[");
      TERM.print(SARA_delta);
      TERM.print("] ");
    }
}
```

earlephilhower, 2022-08-05:
@Haggarman thanks very much for the concise bug report, debug, and solution.

I think you can achieve the fix by making sure the divide(mod) is unsigned instead of the `if...else`:
````
 return ((uint32_t)(_writer - _reader)) % _fifoSize;
````

Would you like to make a pull request with the fix and get your name included in the `git log` for posterity?  OTW, I'll do my best to get to it this weekend.

Haggarman, 2022-08-05:
@earlephilhower 
Thanks for looking into this.

I think the modulo method only going to work for certain powers of 2 even with the type casting.
I wrote a little test program, the 3rd column is using the divide(mod), the 4th column is the if-else way...

```
Example 1 (correct):
FIFO size is 255
        _writer        _reader      (w-r) % s         if_w<r
            254            255            255            255
            255            255              0              0
              0            255              1              1
              1            255              2              2
              2            255              3              3

Example 2 (off by 1):
FIFO size is 256
        _writer        _reader      (w-r) % s         if_w<r
            255            256              0            256
            256            256              0              0
              0            256              2              1
              1            256              3              2
              2            256              4              3

Example 3 (not correct):
FIFO size is 200
        _writer        _reader      (w-r) % s         if_w<r
            199            200             99            200
            200            200              0              0
              0            200            101              1
              1            200            102              2
              2            200            103              3
```

Haggarman, 2022-08-05:
Here is the program I used to generate those tables.
I tried to copy how internally it uses the FIFO size plus 1.
Use the first line with the **define** to change the size.
I appreciate all your hard work on the RP2040.

```

#define DEFAULT_FIFO (200)

void setup() {
  Serial.begin(115200);
  delay(4000);
  
  Serial.print("FIFO size is ");
  Serial.println(DEFAULT_FIFO);

}

void loop() {
  uint32_t i;
  uint32_t _writer = 0;
  
  uint32_t _reader = DEFAULT_FIFO;
  uint32_t _fifoSize = DEFAULT_FIFO + 1;

  uint32_t retval;

  Serial.printf("\n %14s %14s %14s %14s\n", "_writer", "_reader", "(w-r) % s", "if_w<r");   //table heading

  //iterate from just before wrap into just after the wraparound
  // for example if FIFO is 200, try 198, 199, 200, 0, and 1 as values for _writer
  //
  for (i = _fifoSize - 2; i <= _fifoSize + 2; i++) {
    _writer = i % _fifoSize;
    Serial.printf("%15u", _writer);
    Serial.printf("%15u", _reader);
    Serial.printf("%15u", ((uint32_t)(_writer - _reader)) % _fifoSize );

    //the admittedly uglier way using an if-then statement
    if (_writer >= _reader) {
      retval = _writer - _reader;
    } else {
      retval = _fifoSize + _writer - _reader;
    }
    Serial.printf("%15u\n", retval);
  }
  delay(4000);
  Serial.println(" ");
}
```

earlephilhower, 2022-08-05:
Yup, good call.  When the delta goes negative bad things happen, so we need to keep the delta positive like this:
````
return (_fifoSize + _writer - _reader) % _fifoSize;
````

earlephilhower, 2022-08-06:
Thanks again for the good bug report and debug!

PR #739 (2022-08-06): Fix UART wrong ::available() during wraparound
Fixes #735 .  Thanks to @ Haggarman for the find and fix.
commit cabb06d4 (2022-08-06): Fix UART wrong ::available() during wraparound (#739)

Fixes #735 .  Thanks to @ Haggarman for the find and fix.
closed 2022-08-06 by earlephilhower