earlephilhower/arduino-pico #738: NRF24 radio problem using arduino-pico

etrofla, 2022-08-06:
I ran into a problem with using Arduino-Pico and RF24 radio library.  I was using some code I developed using Arduino-mbed under platformIO that works fine for the target Arduino Nano RP2040 Connect board and nrf24 hardware.  The code compiles fine under Arduino-Pico but it seems like the pin settings for CE and CSN may not be getting set correctly when instantiating the radio object? 
Source code:

// TARGET BOARD: Arduino Nano RP2040 Connect
// RF24 library version: 1.4.5
#include <Arduino.h>
#include <SPI.h>
#include "printf.h"
#include "RF24.h"

// instantiate an object for the nRF24L01 transceiver
RF24 radio(9, 10);  // use pin 9 for the CE pin, and pin 10 for the CSN pin

void setup() {
  // put your setup code here, to run once:
   Serial.begin(115200);
  while (!Serial) {
    // wait to ensure access to serial over USB
  }

  // initialize NRF24 transceiver on the SPI bus
  if (!radio.begin()) {
    while (1) {
      Serial.println(F(">> radio NOT FOUND on SPI bus!"));
    }  // hold in infinite loop
  }

}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println(F(">> radio hardware found on SPI bus!!"));
}

earlephilhower, 2022-08-06:
Sorry, since this is both a 3rd party library and a piece of hardware I don't have, you'll have to do the debug on your own.

One guess would be that the pins being used for the SPI (spi0) don't match your wiring.  You can explicitly set them using the `SPI.setXXX` calls documented in the guide.

etrofla, 2022-08-07:
Checked the wiring again and it is correct. Runs the code using the standard Arduino/mbed/platformio environment.

earlephilhower, 2022-08-07:
OKay, you will need to get some waveforms or there's nothing that anyone but you can do since again, you are using your own hardware and your own third party library.

oddstr13, 2022-08-07:
Unable to reproduce on the Pico.

Once I found the SPI pins, your getting started snippet is working just fine.

Enable debugging on port `Serial`, and set debug level to `SPI`, and you should see a line similar to this (should be one of the first lines);
`SPI::begin(0), rx=16, cs=17, sck=18, tx=19`

- rx = MISO = 16
- sck = SCK = 18
- tx = MOSI = 19

etrofla, 2022-08-08:
Thanks I will try that. I am a newbie with Arduino-pico.  Standby...

etrofla, 2022-08-12:
Here is my latest code and screenshot of output:

// TARGET BOARD: Arduino Nano RP2040 Connect
// RF24 library version 1.4.5
// Arduino-Pico  release 2.3.3
// Arduino IDE 2.0.0.0-rc9.2
//    Debug level: SPI
//    Debug port: Serial
// 
#include <Arduino.h>
#include <SPI.h>
#include "printf.h"
#include "pico/binary_info.h"
#include "hardware/spi.h"
#include "RF24.h"

// instantiate an object for the nRF24L01 transceiver
RF24 radio(9, 10);  // use pin 9 for the CE pin, and pin 10 for the CSN pin

void setup() {
  // put your setup code here, to run once:
   Serial.begin(115200);
  while (!Serial) {
    // wait to ensure access to serial over USB
  }

  // Setup SPI bus pins
  // Defaults from official Arduino Nano RP2040 Connect pin diagram
  // Configuration works on same HW when using ArduinoCore-mbed
  SPI.setSCK(13);  // Clock signal
  SPI.setCS(9);    // Chip select
  SPI.setRX(12);   // MISO (CIPO) 
  SPI.setTX(11);   // MOSI (COPI)

  // initialize NRF24 transceiver on the SPI bus
  if (!radio.begin()) {
     Serial.println(F(">>> radio NOT FOUND on SPI bus!"));
    while (1) {
     }  // hold in infinite loop
  }
 Serial.println(F(">> radio hardware FOUND on SPI bus!!"));
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println(F(">> radio hardware FOUND on SPI bus!!"));
  delay(2000);
}

[image]

earlephilhower, 2022-08-12:
Well, that message says it all really.  SPI.SCK cannot be on pin 13.

I suggest looking at [this image](https://cdn-learn.adafruit.com/assets/assets/000/099/339/original/raspberry_pi_Pico-R3-Pinout-narrow.png?1612915004) which contains a breakdown of which pins are legal for which subsystems.

closed 2022-08-12 by earlephilhower