earlephilhower/arduino-pico #740: The 'softwire' library file cannot be used.

xiezhoubin, 2022-08-06:
Problem:
I use mpu6050 sensor and raspberry pie Pico development board. I can make sure the wiring is correct. However, the same code can run on Arduino uno and esp32, and can correctly read mpu6050 data. After the raspberry pie Pico runs, it cannot correctly read data.(Raspberry pie enables Pico to read normally using hardware IIC.)

SoftWire Library:
https://github.com/stevemarple/SoftWire

Picture:
[image]
Raspberry pie Pico failed to read with software IIC:
[image]
Arduino uno successfully read with software IIC:
[image]

Code:

#include <SoftWire.h>
SoftWire Wire(12, 13); // Raspberry pie Pico(SDA=12,SCL=13) ,ARDUINO UNO(SDA=2,SCL=3)

const int MPU6050_addr = 0x68;
int16_t AccX, AccY, AccZ, Temp, GyroX, GyroY, GyroZ;
uint8_t swTxBuffer[32];
uint8_t swRxBuffer[32];

void setup() {
  Wire.setTxBuffer(swTxBuffer, sizeof(swTxBuffer));
  Wire.setRxBuffer(swRxBuffer, sizeof(swRxBuffer));
  Wire.begin();
  Wire.beginTransmission(MPU6050_addr);
  Wire.write(0x6B);
  Wire.write(0);
  Wire.endTransmission(true);
  Serial.begin(9600);
}

void loop() {
  Wire.beginTransmission(MPU6050_addr);
  Wire.write(0x3B);
  Wire.endTransmission(false);
  Wire.requestFrom(MPU6050_addr, 14, true);
  AccX = Wire.read() << 8 | Wire.read();
  AccY = Wire.read() << 8 | Wire.read();
  AccZ = Wire.read() << 8 | Wire.read();
  Temp = Wire.read() << 8 | Wire.read();
  GyroX = Wire.read() << 8 | Wire.read();
  GyroY = Wire.read() << 8 | Wire.read();
  GyroZ = Wire.read() << 8 | Wire.read();
  
  Serial.print("AccX = ");
  Serial.print(AccX);
  Serial.print(" || AccY = ");
  Serial.print(AccY);
  Serial.print(" || AccZ = ");
  Serial.print(AccZ);
  Serial.print(" || GyroX = ");
  Serial.print(GyroX);
  Serial.print(" || GyroY = ");
  Serial.print(GyroY);
  Serial.print(" || GyroZ = ");
  Serial.print(GyroZ);
  Serial.print(" || Temp = ");
  Serial.println(Temp / 340.00 + 36.53);
  delay(500);
}

earlephilhower, 2022-08-06:
You're using a third party library and hardware that I don't have, so the debug is going to be your responsibility.

A quick look at the library doesn't seem to show anything that shouldn't work so I would suggest
a) Triple-checking the wiring.  You are using the RP2040 **chip** GPIO pins 12 and 13, but the **PCB** the chip is on may not label them the same.   So, check the PCB schematics and make sure you really are connected properly
b) Get a waveform capture with a logic analyzer.  This is a 2-pin interface so very simple to decode.  That will help point to the problem.

earlephilhower, 2022-08-13:
Closing due to lack of response

closed 2022-08-13 by earlephilhower