earlephilhower/arduino-pico #742: WiFiNINA FirmwareUpdater

sabas1080, 2022-08-06:
Hi

trying to use the example of FimwareUpdater for WiFiNINA I have the following error, it seems with the Serial3 port, I have tried to declare it but I get an error

```
arduino15/packages/rp2040/hardware/rp2040/2.3.3/variants/arduino_nano_connect/pins_arduino.h:94:21: error: 'Serial3' was not declared in this scope; did you mean 'Serial2'?
   94 | #define SerialNina  Serial3
      |                     ^~~~~~~
/Arduino/libraries/WiFiNINA/examples/Tools/FirmwareUpdater/ESP32BootROM.cpp:335:32: note: in expansion of macro 'SerialNina'
  335 | ESP32BootROMClass ESP32BootROM(SerialNina, NINA_GPIO0, NINA_RESETN);
```

Thanks

earlephilhower, 2022-08-09:
Looks like they implement a software serial port `Serial3` on the NANO to handle FW uploads.  There are only 2 physical UARTS, so it's got to be SW related.

Probably need to instantiate a SoftwareSerial object in the NANO variant and then add `extern SoftwareSerial Serial3;` to the variant header.  I don't own a NANO, but if you're using the git sources it should be straightforward.

earlephilhower, 2022-08-27:
I just installed the latest Arduino MBED Rp2040 core thru the boards manager, and the latest WiFiNINA library from the library manager.  This example does not compile on the official Arduino core, either.

The header makes me think this is expected as the comments don't list the RP2040:
````
/*
  FirmwareUpdater - Firmware Updater for the 
  Arduino MKR WiFi 1010, Arduino MKR Vidor 4000, and Arduino UNO WiFi Rev.2.
  
  Copyright (c) 2018 Arduino SA. All rights reserved.
  ````

Closing as expected behavior

sabas1080, 2022-08-29:
if you work on rp2040, just select the board first
[image]

earlephilhower, 2022-08-29:
Thanks for the clarification!

I actually just installed the "RP2040 boards" core from Arduino and did not install the "Nano" one.  I think they changed this recently, because I swear the RP2040 Connect and bare RP2040 were both in the same core, before.

I'll see what I can see w/the other Arduino core installed.

closed 2022-08-30 by earlephilhower