earlephilhower/arduino-pico #743: After flashing the pico, no life sign from the board

HexRebuilt, 2022-08-07:
Hi everyone,

i'm using a RP2040-Zero from waveshare [https://www.waveshare.com/product/raspberry-pi/boards-kits/raspberry-pi-pico-cat/rp2040-zero.htm](url). 

If i try to upload the code, even a simple "hello word" over serialUSB, nothing happens after the flashing is completed. It's like the board is not able boot or something because if i connect it over USB windows does not detect it and the code seems not running (tried to change a pin digital output to see if was something only related to the serialUSB but no changes there).

If i use platformio indicated platform (raspberry) even if it is missing a lot of key stuff that are present in the arduino-pico, the board is detected overt serial and i get the "hello word" print. So the board is ok and working from the hw point of view.

What am i doing wrong?

This is my platformio.ini file
`[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower`

thanks in advance

earlephilhower, 2022-08-07:
Most likely the issue is the flash.  A different bootloader is required for different flash vendors, and the Waceshare board may not use flash compatible with the original RPi Pico.

ry the Generic board, and use the Generic SPI / 4 (i.e. slowest, safest)

HexRebuilt, 2022-08-10:
Hi, thanks for the answer.

Even with the generic board nothing changes. still no life sign at any level.

'[env:generic]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = generic
framework = arduino
board_build.core = earlephilhower
lib_deps = claws/BH1750@^1.3.0'

The only way to have the board working is to use the "raspberry platform"

'[env:nanorp2040connect]
platform = raspberrypi
board = nanorp2040connect
framework = arduino'

But this platform is missing a lot of stuff to work with most of the arduino libraries online and does not allow changing the pins of the i2c or SPI. 

Any idea on what can cause the problem?

earlephilhower, 2022-08-10:
I was suggesting that it's not just the board you need to set, but the bootloader.  Not sure the PIO commands to make it happen, but you want to use this boot2.S file to slow things down and use the safest single-bit SPI:
````
build.boot2=boot2_generic_03h_4_padded_checksum
````

HexRebuilt, 2022-08-11:
I get what you are saying.

So the bootloader is missing and the board cannot execute any code. 
If i look into the "*.o" files builted while compiling, there is the presence of the Bootsel.cpp.o, which should be the responsible for gatting the board displayed as a usb drive to upload the ELF file. 

Still i don't know how to add the bootloader to the system, nor if it is possible to steal it from the raspberry platform

maxgerhardt, 2022-08-11:
Just to double check, when you use the Arduino IDE and the "Pico" or "Generic" board definition, the board also does nothing? If the Arduino IDE works, it's a fault on the PlatformIO side.

For PlatformIO, the default bootloader is decided in the board's JSON

https://github.com/maxgerhardt/platform-raspberrypi/blob/develop/boards/generic.json#L5

So if you open `C:\Users\<user>\.platformio\platforms\raspberrypi\boards\generic.json` and change that string to any of the other [available bootloaders](https://github.com/earlephilhower/arduino-pico/tree/master/boot2), you can see which one works.

maxgerhardt, 2022-08-11:
BTW, what is on written on the small flash chip on the top of the board? Same as in 

[image]
?

HexRebuilt, 2022-08-11:
> Just to double check, when you use the Arduino IDE and the "Pico" or "Generic" board definition, the board also does nothing? If the Arduino IDE works, it's a fault on the PlatformIO side.

in this case the board works. same with platformio by using the raspberry platform and MBED core. boots writes over serial etc. but i'm missing a lot of stuff that are present in this platform.

> BTW, what is on written on the small flash chip on the top of the board? Same as in
> 
> [image] ?

**_4P03Z0022Y_**

maxgerhardt, 2022-08-11:
>in this case the board works. same with platformio by using the raspberry platform and MBED core. boots writes over serial etc. but i'm missing a lot of stuff that are present in this platform.

Okay so to tripple check, Arduino IDE + Arduino-Pico (not Arduino IDE + ArduinoCore-mbed) works. What version of the Arduino-Pico core is install when you go into the Arduino IDE's tools->board->board manager? Can you make a screenshot of the exact tool menu with which it works in Arduino IDE + Arduino-Pico?

Andy2No, 2022-08-15:
I've got hold of a few Waveshare RP2040-Zero, and tried one with the NeoPixel example I first tried with the Adafruit ItsyBitsy RP2040: https://github.com/earlephilhower/arduino-pico/discussions/756

It's working fine with the Arduino IDE v1.8.19, with the board type set to Raspberry Pi Pico.   I haven't verified that all the pins are mapped where they should be, but the ADC pins seem to be (finger test, reading out the values to the serial monitor), and the WS2812 works on pin 16, as advertised.

I'm using it on Windows 7, so part of this may not apply.  Initially, it didn't connect so I couldn't  upload to it, but I used Zadig (see #96) to make Windows recognise it, after plugging it in with the Boot button held down.

After uploading the first sketch with the Boot button held down as I connected it, I haven't needed to use the Boot button again - it's just working the same as a Pico.

There could be different Flash versions though, I suppose, like there were with the ItsyBitsy RP2040.  I don't know if I have anything capable of reading the label on the chip.

I still have the PC I've been trying it with on arduino-pico 1.9.15, but that's probably not relevant to the issue @HexRebuilt is having.

Is the Zadig (issue #96) thing relevant to Windows 10/11?  That seems like the first thing to try.

Andy2No, 2022-08-16:
I found my cheap pair of watch repair glasses, and used a little torch to light the Flash chip.  I can't photograph it easily, but this is what I've read, on all four that I have:

1P128
0Q149D

I don't know if that's a zero or a capital O, on the second line, of course.

maxgerhardt, 2022-08-16:
Could you say something in regards to https://github.com/earlephilhower/arduino-pico/issues/743? Still trying to exactly understand the situation.

Andy2No, 2022-08-16:
@maxgerhardt My comments were intended to relate to this current issue, but I can't claim to fully understand what @HexRebuilt is saying is the problem.

I use Arduino, and only Arduino, in this case with Earle's arduino-pico core, and I was confirming that it works fine with those, giving the version numbers of both the IDE and the arduino-pico core I was using to test my new boards.

I can try newer versions of those two things if it helps in any way, but I had reasons for leaving it as it was on that PC, for the moment.

You asked for the marking on the Flash chip, recently, so I've given mine.  I have no way of knowing if the ones @HexRebuilt has are the same, of course.

maxgerhardt, 2022-08-16:
Ahh my bad, I confused you with the other user.

HexRebuilt, 2022-08-16:
> @maxgerhardt My comments were intended to relate to this current issue, but I can't claim to fully understand what @HexRebuilt is saying is the problem.

Hi guys sorry for the late reply. my problem is that, with platformio and arduino + MbdedOS the board works and do what is supposed to (except for the parts not implemented by mBedOS in platformio, requested by many off the shelf arduino libraries).
With platformio + Arduino-pico no life sign after the code is pushed into the board (over platformio or by copying the file into the board). i can still access the "mass storage" to upload a new fw but i'm not even able to change a pin state.

For sure i'm doing something wrong here but i don't know what it is.

maxgerhardt, 2022-08-16:
>With platformio + Arduino-pico no life sign after the code 

And with the Arduino IDE + latest Arduino-Pico, it's the same, no matter if choosing the generic pico or RP2040 Zero?

HexRebuilt, 2022-08-16:
> > With platformio + Arduino-pico no life sign after the code
> 
> And with the Arduino IDE + latest Arduino-Pico, it's the same, no matter if choosing the generic pico or RP2040 Zero?

works as well. this is why i'm thinking that it's my configuration of the platformio.ini
[image]

maxgerhardt, 2022-08-16:
>works as well. 

Can you take a screenshot of your exact Arduino IDE -> Tools menu with which it works with Arduino-Pico?

The pic you uploaded doesn't display.

maxgerhardt, 2022-08-16:
Can you [open a CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and execute 
```
pio pkg update -g -p "https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli"
``` 
to make sure you have all the latest versions? Your initial `platformio.ini` looks okay.

Remember to close the Arduino IDE serial monitor, then use the "Upload" task and then the "Monitor" task. (USB serial needs some time to show up).

HexRebuilt, 2022-08-16:
> Can you [open a CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and execute

>pio pkg update -g -p "https://github.com/maxgerhardt/platform-raspberrypi.git"

`Error: Could not find the package with 'https://github.com/maxgerhardt/platform-raspberrypi.git' requirements for your system 'windows_amd64'`

maxgerhardt, 2022-08-16:
Huh. Did you initially install it with a `pio platform update..`? 

In any case, please do a 

```
pio upgrade --dev
```
then remove all folders `C:\Users\<user>\.platformio\packages\raspberrypi*` and build + upload the project again as above.

HexRebuilt, 2022-08-17:
ok i've updated everything and now i'm not able to upload the code for some reason. 

- over platformio, with the upload button: it fails after opening up the folder where i can paste the UF2 file, by resetting the board.
- `Building in release mode
Retrieving maximum program size .pio\build\pico\firmware.elf
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
Checking size .pio\build\pico\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   3.1% (used 8052 bytes from 262144 bytes)
Flash: [          ]   2.8% (used 59564 bytes from 2093056 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picoprobe, picotool, raspberrypi-swd
CURRENT: upload_protocol = picotool
Looking for upload port...
Auto-detected: COM5
Forcing reset using 1200bps open/close on port COM5
Uploading .pio\build\pico\firmware.elf
rp2040load 1.0.1 - compiled with go1.15.8
.....................
*** [upload] Error 1
`

- pasting over the UF2 file: in this case nothing happens. usually the board reboots and runs the new fw but in this case it's like it's not reading the bult file from platformio. if i reset it, it runs the previous succesfully loaded FW.

maxgerhardt, 2022-08-17:
When you plug the boot in in bootloader mode, does Zadig still show WinUSB drivers loaded for the Interface 1? See https://github.com/platformio/platform-raspberrypi/pull/36#issuecomment-1196592406

HexRebuilt, 2022-08-17:
[image]
yes it does. it still recognise it. just to be sure i've used Zadig and followed the installation guide by putting USB Serial (CDC).

still the same error.

maxgerhardt, 2022-08-17:
Weird how it doesn't say "RP2 Boot (Interface 1)" but "RP2 Boot". Can you show a screenshot of Zadig with the RP2 Boot Interface 1 device selected?

HexRebuilt, 2022-08-17:
[image]
sure

HexRebuilt, 2022-08-17:
indeed. i've changed it again to USB Serial but nothing changes. Zadig reads it always as usbser. don't know if it's expected

maxgerhardt, 2022-08-17:
After loading "WinUSB" driver, it is indeed supposed to show "winusb" as the now loaded driver, if not the driver change has not work. I currently do not have access to my board to show the "counter screenshot", that needs a few hours more.

maxgerhardt, 2022-08-17:
Zadig should show

[image]

if not driver install failed. I'm using Zadig 2.7 here.

HexRebuilt, 2022-08-18:
i was following the guide here <https://arduino-pico.readthedocs.io/en/latest/install.html#windows-7-driver-notes> for the driver. that's why it was not showing winUSB in the screenshot. 

tried again with WinUSB but nothing changes. it just misses "_interface 1_" in the windows device manager.
The _interface 0_ is the same as your screenshot.
[image]

maxgerhardt, 2022-08-18:
Hm but Zadig shows the right driver is loaded at least, so it *should* work.. 

What happens in when you execute this in the VSCode terminal?
```
pio pkg exec -p "tool-rp2040tools"  -c "picotool info"
```

HexRebuilt, 2022-08-18:
> What happens in when you execute this in the VSCode terminal? (Pico board should be connected in bootloader mode of course)
> 
> ```
> pio pkg exec -p "tool-rp2040tools"  -c "picotool info"
> ```

i get this:
`Using tool-rp2040tools@1.0.2 package
Assertion failed: this_size, file /__w/rp2040tools/rp2040tools/picotool/main.cpp, line 809`

maxgerhardt, 2022-08-18:
https://github.com/raspberrypi/picotool/blob/a49dc0b3c211b8b73d4ec7967cf134710ce104c2/main.cpp#L807-L809

I'm literally baffled by what is going on your computer or chip, never seen this before. It's running into an assert when reading from the memory of the Pico, the calculated number of bytes it wants to read was 0.

There is a new version available (1.1.0), but PIO doesn't yet have it (https://registry.platformio.org/tools/platformio/tool-rp2040tools/versions).

The only thing I can recommend for now is to set 
```ini
upload_protocol = mbed
```
in the `platformio.ini` to select the "copy UF2 to bootloader drive" method. There is however no auto-reset there yet, so you would have to put the board in bootloader mode everytime.

I can work on getting the newer picotool software version and autoreset for the UF2 upload enabled (so it's pretty much equivalent to what this core does) in the meantime.

HexRebuilt, 2022-08-18:
ok i've tried it. i'm able to upload the file over usb `(upload_protocol = picotool)` but i'm back at the starting problem where the board does not boot.

to double check i've done another upload with the arduni IDE and with copying the UF2 file. still works. 

I've no idea on what i'm doing wrong with platformio. 

To triple check. i've created a new project from scratch with the classical "hello" over seial and it uploads over platformio and kinda works.
Same goes for creating variables and using them.

But in the instant want to use a BH1750 light sensor, then the board no longer boots, nor is recognised by windows. 
To reduce the problem scope i tried to define the I2C without using the sensor's library. It stops working. Same goes if i try to use different pins. the board is no longer recognised by windows, with associated message from the notification area.

it's not the _.begin()_ part that makes everything goes south but even doing:
`Wire.setSDA(D2);
  Wire.setSCL(D3);`
without calling `Wire.begin();`

the following snippet is enough to make the board mad:
```
#include <Arduino.h>
#include <string.h>
#include <stdlib.h>
#include <Wire.h>
//#include <BH1750.h>

int test = 0;
String letters;

//BH1750 lightMeter;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  Wire.setSDA(D2);
  Wire.setSCL(D3);

  /*
  //I2C to the light sensor
  Wire.setSDA(D2);
  Wire.setSCL(D3);
  Wire.begin();
  lightMeter.begin();
  */
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.print("hello + int def platformio: ");
  Serial.println(test);
  test++;

  /*
  float lux = lightMeter.readLightLevel();
  Serial.print("Light: ");
  Serial.print(lux);
  Serial.println(" lx");
  */
 
  delay(500);
}
```

i have no idea on what's causing this issue, also with other pins. Naturally if i remove those instruction, everything goes back to "normal".

earlephilhower, 2022-08-18:
````
  Wire.setSDA(D2);
  Wire.setSCL(D3);
````

This is a long and winding bug, but I can easily tell you why the above code is crashing:   Those pins are not legal for I2C0, only I2C1, and you are hitting a `panic()`.

[image]

HexRebuilt, 2022-08-18:
> This is a long and winding bug, but I can easily tell you why the above code is crashing: Those pins are not legal for I2C0, only I2C1, and you are hitting a `panic()`.

OH. i'm dumb. sorry. so now it works i think. if so i'll bother you again. thanks a lot for the help

closed 2022-08-18 by HexRebuilt

maxgerhardt, 2022-08-18:
Wait, does it work in PlatformIO too now? No more discrepency between PIO and Arduino?

HexRebuilt, 2022-08-18:
yes for the moment. i cannot use the old project tho. but that can be related to some of my programming error. i'm able to flash it with platformio