earlephilhower/arduino-pico #749: error: redefinition of 'class fs::File': "WiFi.h" and "SdFat.h" conflicts

Deemocean, 2022-08-10:
I was trying to use the WiFi lib and the Adafruit TinyUSB lib together as such:
```C++
#include "WiFi.h" 
#include "SdFat.h"
```
During compilation, the IDE(both Arduino&PlatformIO) gives:
```
In file included from .pio/libdeps/picow/SdFat - Adafruit Fork/src/FatLib/FatLib.h:27,
                 from .pio/libdeps/picow/SdFat - Adafruit Fork/src/SdFat.h:33,
                 from src/main.cpp:20:
.pio/libdeps/picow/SdFat - Adafruit Fork/src/FatLib/ArduinoFiles.h:122:7: error: redefinition of 'class fs::File'
  122 | class File : public FatFile, public Stream {
      |       ^~~~
In file included from /Users/deemo/.platformio/packages/framework-arduinopico/libraries/WiFi/src/CertStoreBearSSL.h:25,
                 from /Users/deemo/.platformio/packages/framework-arduinopico/libraries/WiFi/src/WiFiClientSecureBearSSL.h:30,
                 from /Users/deemo/.platformio/packages/framework-arduinopico/libraries/WiFi/src/WiFiClientSecure.h:23,
                 from /Users/deemo/.platformio/packages/framework-arduinopico/libraries/WiFi/src/WiFi.h:11,
                 from src/main.cpp:19:
/Users/deemo/.platformio/packages/framework-arduinopico/cores/rp2040/FS.h:51:7: note: previous definition of 'class fs::File'
   51 | class File : public Stream {
      |       ^~~~
In file included from .pio/libdeps/picow/SdFat - Adafruit Fork/src/FatLib/FatLib.h:28,
                 from .pio/libdeps/picow/SdFat - Adafruit Fork/src/SdFat.h:33,
                 from src/main.cpp:20:
.pio/libdeps/picow/SdFat - Adafruit Fork/src/FatLib/FatFileSystem.h: In member function 'fs::File FatFileSystem::open(const char*, oflag_t)':
.pio/libdeps/picow/SdFat - Adafruit Fork/src/FatLib/FatFileSystem.h:95:13: error: 'class fs::File' has no member named 'open'
   95 |     tmpFile.open(vwd(), path, oflag);
      |             ^~~~
```
For sanity check, in the Arduino IDE, I tried to compile the `#include` statements mentioned above with Arduino Uno configuration, which builds successfully.

Not sure if it is a problem with the Adafruit Fork SdFat or Arduino-pico.

My Environment:
	adafruit/Adafruit TinyUSB Library@^1.14.3
	adafruit/SdFat - Adafruit Fork@^1.5.1
	adafruit/Adafruit SPIFlash@^3.11.0

earlephilhower, 2022-08-10:
That library is incompatible with this core.  We already have a `File` class and include both the `SDFS` and the `SD` filesystems.  Use those if you need to access a SD card as they hook into the core's entire `File` ecosystem.

closed 2022-08-10 by earlephilhower

Deemocean, 2022-08-12:
What if I want to use the LittleFS to access the External onboard flash on the Pico? I am trying to use WiFi.h and letting the board behave like a mass storage device at the same time. Is there an alternative for the `Adafruit_SPIFlash`object? Because it is used in the callback function for msc in the Adafruit TinyUSB `msc_external_flash` example.

earlephilhower, 2022-08-12:
Sorry, I don't use that at all.  It's doubtful the host would understand LittleFS, anyway.

If you're just exporting flash memory, why not just use the flash read/write/erase calls directly?

Deemocean, 2022-08-12:
I want to do a USB flash drive using the onboard flash(so I can send files over wifi to the flash and transfer to the device it connects to (as msc)).   In the Adafruit TinyUSB's `msc_external_flash` example(which "expose on-board external Flash as USB Mass Storage."), it basically requires me to format the flash to FAT12 manually first(gives about 1Mb of FAT12 storage), then uses their SPIFlash to access the flash and run their fork of SdFAT on the flash. 
I guess LittleFS is really not meant to be used on anything related to MSC? (because no FAT support). 
Are there any other options other than me migrating some of the features needed from the Adafruit fork SdFAT to the ESP8266 SdFAT(not sure if gonna work)--and make the SPIFlash use that library, in order to turn pico into a flash drive while fitting in the ESP8266 SdFAT ecosystem?
Sorry if this is just a weird use case, if there is currently no way I think I can just buy a SD breakout board.

Deemocean, 2022-08-12:
I realized the Adafruit fork SdFAT is based on the original SdFat ver.1, which has classes like `BlockDriver.` As the Arduino-pico core uses SdFat ver.2 I should probably ask this on the SdFat repo...
Thanks for the replay tho!