earlephilhower/arduino-pico #751: FreeRTOS : configTIMER_TASK_STACK_DEPTH is too low which causing system hang

freeasabeer, 2022-08-11:
Currently configTIMER_TASK_STACK_DEPTH is set to 80 which is below configMINIMAL_STACK_SIZE which is set to 256.
This too low value is causing an exhaustion of the available stack for timer stuff resulting in a system hang while using FreeRTOS timers.
At least the timer stack should be set to 256 but I would recommend an even higher value like 1024 (for reference it is 2048 on the ESP32).

Great job done anyway on this project !

earlephilhower, 2022-08-11:
Good observation.  Each SIZE is actually 4-bytes, but even so 80 is probably a bit small especially if there are interrupts coming in.   1024 = 4K = the standard Pico SDK hardware stack size, so that seems like a reasonable size.

Would you like to do a PR changing this value?

freeasabeer, 2022-08-11:
Yes, I will do.

closed 2022-08-11 by earlephilhower