earlephilhower/arduino-pico #753: #include errors in Visual Studio Code

wachidsusilo, 2022-08-12:
I am using official Arduino by Microsoft in Visual Studio Code. There is red squiggle on every first line of file in my code. It is compiled and uploaded fine though. However, it's defeat the purpose of using VSCode in the first place, as i want to know if there is error in my code while writing it. 

The error says:
```
cannot open source file "lwip/init.h"
cannot open source file "lwip/ip_addr.h"
cannot open source file "lwip/ip4_addr.h"
cannot open source file "pico/mutex.h"
```
is there any workaround to get rid of this errors?

Here is the `c_cpp_properties.json` configuration in case it's needed.
```
{
    "version": 4,
    "configurations": [
        {
            "name": "Arduino",
            "compilerPath": "C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.4.0-c-0196c06\\bin\\arm-none-eabi-g++",
            "compilerArgs": [
                "-Werror=return-type",
                "-march=armv6-m",
                "-mcpu=cortex-m0plus",
                "-mthumb",
                "-ffunction-sections",
                "-fdata-sections",
                "-fno-exceptions"
            ],
            "intelliSenseMode": "gcc-x64",
            "includePath": [
                "C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.3.3\\include",
                "C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.3.3\\cores\\rp2040",
                "C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.3.3\\variants\\generic",
                "C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.3.3\\libraries\\LittleFS\\src",
                "c:\\users\\user\\appdata\\local\\arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.4.0-c-0196c06\\arm-none-eabi\\include\\c++\\10.3.0",
                "c:\\users\\user\\appdata\\local\\arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.4.0-c-0196c06\\arm-none-eabi\\include\\c++\\10.3.0\\arm-none-eabi\\thumb",
                "c:\\users\\user\\appdata\\local\\arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.4.0-c-0196c06\\arm-none-eabi\\include\\c++\\10.3.0\\backward",
                "c:\\users\\user\\appdata\\local\\arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.4.0-c-0196c06\\lib\\gcc\\arm-none-eabi\\10.3.0\\include",
                "c:\\users\\user\\appdata\\local\\arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.4.0-c-0196c06\\lib\\gcc\\arm-none-eabi\\10.3.0\\include-fixed",
                "c:\\users\\user\\appdata\\local\\arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.4.0-c-0196c06\\arm-none-eabi\\include"
            ],
            "forcedInclude": [
                "C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.3.3\\cores\\rp2040\\Arduino.h"
            ],
            "cStandard": "c11",
            "cppStandard": "c++17",
            "defines": [
                "CFG_TUSB_MCU=OPT_MCU_RP2040",
                "USB_VID=0x2e8a",
                "USB_PID=0xf00a",
                "USB_MANUFACTURER=\"Generic\"",
                "USB_PRODUCT=\"RP2040\"",
                "PICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1",
                "CYW43_LWIP=0",
                "LWIP_IPV6=0",
                "LWIP_IPV4=1",
                "LWIP_IGMP=1",
                "LWIP_CHECKSUM_CTRL_PER_NETIF=1",
                "SERIALUSB_PID=0xf00a",
                "USBD_MAX_POWER_MA=250",
                "F_CPU=133000000L",
                "ARDUINO=10819",
                "ARDUINO_GENERIC_RP2040",
                "BOARD_NAME=\"GENERIC_RP2040\"",
                "ARDUINO_ARCH_RP2040",
                "__DBL_MIN_EXP__=(-1021)",
                "__HQ_FBIT__=15",
                "__cpp_attributes=200809L",
                "__UINT_LEAST16_MAX__=0xffff",
                "__ARM_SIZEOF_WCHAR_T=4",
[... 408 lines cut ...]
                "__LDBL_MIN_10_EXP__=(-307)",
                "__cpp_generic_lambdas=201304L",
                "__SIZEOF_LONG_LONG__=8",
                "__ULACCUM_EPSILON__=0x1P-32ULK",
                "__cpp_user_defined_literals=200809L",
                "__SACCUM_IBIT__=8",
                "__GCC_ATOMIC_LLONG_LOCK_FREE=1",
                "__FLT_DECIMAL_DIG__=9",
                "__UINT_FAST16_MAX__=0xffffffffU",
                "__LDBL_NORM_MAX__=1.7976931348623157e+308L",
                "__GCC_ATOMIC_SHORT_LOCK_FREE=1",
                "__ULLFRACT_MAX__=0XFFFFFFFFFFFFFFFFP-64ULLR",
                "__UINT_FAST8_TYPE__=unsigned int",
                "__USFRACT_EPSILON__=0x1P-8UHR",
                "__ULACCUM_FBIT__=32",
                "__QQ_IBIT__=0",
                "__cpp_init_captures=201304L",
                "__ATOMIC_ACQ_REL=4",
                "__ATOMIC_RELEASE=3",
                "USBCON"
            ]
        }
    ]
}
```

earlephilhower, 2022-08-12:
I haven't used MSFT Visual Studio in a decade, so can't help you with the actual changers, but your `includePath` is missing all of the SDK files.  See if you can make it include every directory in the `lib/platform_inc.txt`.

closed 2022-08-12 by earlephilhower

wachidsusilo, 2022-08-12:
Thanks for the quick response.  Just to clarify, I'm using Visual Studio Code, not Visual Studio. Btw, The `includePath` is auto generated. It would be overwritten after the compiling process even if i try to change it. I think i should create another configuration just for the SDK files.

(**After Quick Try**) 
Ah, never mind, another configuration is just not recognized by the intelliSense. Luckily i have found the solution to force VSCode to not overwrite the existing configuration.

**First**, create file named `settings.json` under `.vscode` folder (if it's not already exist). Then, add below line of code:
```
{
    ...
    "arduino.disableIntelliSenseAutoGen": true
}
```
This will disable the auto generation of `c_cpp_properties.json`.
**NOTE**: This will disable auto generation for current project only.

**Finally**, add the `includePath` suggested by @earlephilhower. 
```
{
    ...
    "configurations": [
        {
            ...
            "includePath": [
                ...
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/cores/rp2040/api/deprecated-avr-comp/",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/include/pico_base/",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/lib/tinyusb/src/",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/boards/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/common/pico_base/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/common/pico_binary_info/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/common/pico_bit_ops/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/common/pico_divider/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/common/pico_stdlib/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/common/pico_sync/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/common/pico_time/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/common/pico_util/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2040/hardware_regs/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2040/hardware_structs/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/cmsis/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/cmsis/stub/CMSIS/Core/Include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/cmsis/stub/CMSIS/Device/RaspberryPi/RP2040/Include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/hardware_adc/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/hardware_base/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/hardware_claim/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/hardware_clocks/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/hardware_divider/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/hardware_dma/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/hardware_exception/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/hardware_flash/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/hardware_gpio/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/hardware_i2c/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/hardware_interp/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/hardware_irq/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/hardware_rtc/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/hardware_pio/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/hardware_pll/include",
[... cut ...]
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/pico_stdio_uart/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/src/rp2_common/pico_unique_id/include",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/lib/cyw43-driver/src",
                "C:/Users/{user}/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/2.3.3/pico-sdk/lib/lwip/src/include"
            ],
           ...
        }
    ]
}
```

Now the `#include` errors should have gone.