earlephilhower/arduino-pico #758: Setting pins for SPI freezes the Pico

bfgelectronics, 2022-08-13:
I want to store some data on a SD card an i used this snippet of code to initialise the SD card, but it seems like the Pico freezes/crashes here :
```
  SPI.setRX(12);
  SPI.setCS(13);
  SPI.setSCK(10);
  SPI.setTX(11);
  SD.begin(13);
```
```
I know that the Pico freezes here because i also have an I2C OLED display connected that shoud display something in normal conditions, but with that snippet it stays black
For context, here is my setup() :
void setup() {
  Wire.setSDA(16);
  Wire.setSCL(17);
  Wire.begin();

  pinMode(18, INPUT_PULLUP);
  pinMode(19, INPUT_PULLUP);

  Serial.begin(9600);

  SPI.setRX(12);
  SPI.setCS(13);
  SPI.setSCK(10);
  SPI.setTX(11);
  SD.begin(13);

  if (!display.begin(SSD1306_SWITCHCAPVCC, SCREEN_ADDRESS)) {
    Serial.println(F("SSD1306 allocation failed"));
    for (;;) {}
  }

  display.setTextColor(SSD1306_WHITE);

  display.clearDisplay();

  attachInterrupt(digitalPinToInterrupt(19), increment, FALLING);
  attachInterrupt(digitalPinToInterrupt(18), decrement, FALLING);
}
```

earlephilhower, 2022-08-13:
You are not setting legal pins for the `SPI` interface, and the Pico is `panic`ing.  You could probably use `SPI1` with those pins.  Fix the pinout per the usual diagram (borrowed from Adafruit):
[image]

Andy2No, 2022-08-13:
@bfgelectronics If the screen initialisation fails, you probably won't see the message on the serial console, because it happens too soon after booting up - it takes a second or two before the serial monitor starts responding, unlike when using things like the Uno or Mega 2560.

Try adding a short delay at the start of setup, maybe blinking the LED for a couple of seconds.  Also, make that part of the forever loop, after trying to report the error:

```
  if (!display.begin(SSD1306_SWITCHCAPVCC, SCREEN_ADDRESS)) {
    Serial.println(F("SSD1306 allocation failed"));
    for (;;) {}
  }
```

Change the for loop to something like this:

```
  pinMode(LED_BUILTIN, OUTPUT);

  if (!display.begin(SSD1306_SWITCHCAPVCC, SCREEN_ADDRESS)) {
    Serial.println(F("SSD1306 allocation failed"));
    for (;;) {
       wait(200); digitalWrite(LED_BUILTIN, 1);
       wait(300); digitalWrite(LED_BUILTIN, 0);
    }
  }
```

Andy2No, 2022-08-13:
@bfgelectronics Also, since those pins can only be used for SPI1, I think you need to change this:

```
  SPI.setRX(12);
  SPI.setCS(13);
  SPI.setSCK(10);
  SPI.setTX(11);
```

To this:

```
  SPI1.setRX(12);
  SPI1.setCS(13);
  SPI1.setSCK(10);
  SPI1.setTX(11);
```

I don't know if the standard SD library works with SPI1 though.  You might need to find a different library to do it, or just use the SPI0 pins instead.

bfgelectronics, 2022-08-14:
Thanks to everyone for support!
@Andy2No yep, i tryed the and the SD library doesnt work with the SPI1, but moved the SD card to some SPI0 pins and now it works!
```
  SPI.setRX(4);
  SPI.setCS(5);
  SPI.setSCK(2);
  SPI.setTX(3);
  SD.begin(5);
 ```

Andy2No, 2022-08-14:
@bfgelectronics Cool.  I'm glad you've got it working.  It's good to have it confirmed, that the SD library needs SPI0 too.

earlephilhower, 2022-08-14:
@bfgelectronics I have modified the SD.begin() call to include an optional filesystem.

You can give it as try with your original wiring, but you'd need to change the set calls to:
````
  SPI1.setRX(12);
  SPI1.setCS(13);
  SPI1.setSCK(10);
  SPI1.setTX(11);
````

and then call
````
SD.begin(13, SPI1);
````

commit 3b5c09de (2022-08-14): Allow selecting SPI port for SD/SDFS filesystem

Fixes #758
PR #759 (2022-08-15): Allow selecting SPI port for SD/SDFS filesystem
Fixes #758
commit 5787b4c0 (2022-08-15): Allow selecting SPI port for SD/SDFS filesystem (#759)

Fixes #758
closed 2022-08-15 by earlephilhower