earlephilhower/arduino-pico #760: RTClib/BusIO unable to detect I2C device with ItsyBitsy board

AcThPaU, 2022-08-15:
ItsyBitsy board has I2C muxed to GPIO 2 and 3.

RTClib and BusIO cannot detect DS3231 connected to I2C.

The hardware setup works fine in CircuitPython.

earlephilhower, 2022-08-15:
You have provided no code or MCVE, so there's not much to go on.

That said, GPIO2/3 is connected to the **second** I2C port on the RP2040, `Wire1`.  Make sure your code is using `Wire1` instead of `Wire`.  See the diagram below which shows the possible I/O muxings:
[image]

You can be even more explicit by calling
````
Wire1.setSDA(2);
Wire1.setSCL(3);
Wire1.begin(...
````

closed 2022-08-15 by earlephilhower

AcThPaU, 2022-08-15:
It's very easy to reproduce. 

```
#include "RTClib.h"

RTC_DS3231 rtc;

void setup() {

Serial.begin(115200);

if (! rtc.begin()) {
 Serial.println("Couldn't find RTC");
 while(1);
 }
}
```

The problem is that RTClib does not allow you to set specific I2C bus.

AcThPaU, 2022-08-15:
It turns out you can set I2C bus in RTClib

```
#include "RTClib.h"
#include <Wire.h>

RTC_DS3231 rtc;

void setup() {

Serial.begin(115200);
Wire1.setSDA(2);
Wire1.setSCL(3);
Wire1.begin();
if (! rtc.begin(&Wire1)) {
 Serial.println("Couldn't find RTC");
 while(1);
 }
}
```

This works, but rtc.begin() should have worked.