earlephilhower/arduino-pico #767: Do the wi-fi extensions support soft access points?

technoblogy, 2022-08-18:
On the ESP32 and ESP8266 you can set up a soft access point with **WiFi.softAP()** but this call doesn't seem to exist in the **Raspberry Pi Pico W** core. Is it there under a different name? If not, any plans to support it?

earlephilhower, 2022-08-18:
AP mode is supported, but not those ESP-specific APIs yet.  A simple wrapper will suffice for them.  Some things (like channel(!!)) are not supported by the CYW43 driver yet, but basic starting/stopping should be a 1-liner.

We're using the WiFi/WiFiNINA compatible `WiFi.beginAP()` as documented https://www.arduino.cc/reference/en/libraries/wifinina/wifi.beginap/ and elsewhere.

technoblogy, 2022-08-18:
Thanks! I'll try those out.

technoblogy, 2022-08-19:
That works nicely - thank you!

PR #777 (2022-08-20): Add ESP8266 compat functions for AP mode
Fixes #767
commit 7be47293 (2022-08-20): Add ESP8266 compat functions for AP mode (#777)

Fixes #767
closed 2022-08-19 by earlephilhower