earlephilhower/arduino-pico #773: Serial Debugging with Waveshare RP2040 Zero in Arduino IDE

IAmOrion, 2022-08-19:
I have a Waveshare RP2040 Zero - for the most part it works as expected, but for the life of me I can not get it to serial debug using the Arduino IDE! I've spent ages searching/googling but just can't find an answer.  What do I need to do / set in order to be able to use the Arduino IDE Serial Monitor and print data to the monitor? 

I've tried Serial.begin with numerous baud rates but no joy, I've tried Serial1. and Serial2. with no joy.  My Serial Monitor as always blank and no matter how many times I reset or re-plug my device, nothing ever appears in the serial monitor.

Has anyone come across this? 

I'm using macOS, IDE 1.8.19.
I used Thonny to install MicroPython, this in turn gave me a port (/dev/cu.usbmodem14101), after that I upload blink sketch, still had to reset + bootsel and then the example uploaded.  Now I can just click upload and it auto-resets the RP2040 Zero and uploads as expected.  I just can't get Serial.print etc to work :(
(I'm using the first Generic option)

Many thanks in advance for any help

earlephilhower, 2022-08-19:
`Serial` is the USB serial port.  That should be universal over all boards.  I believe the Mac handles the ACM serial port w/o any add'l drivers, so you should be good there.

The one gotcha is the USB serial is SW generated from inside the Pico.  So, the process of uploading a sketch may cause the serial port to unplug/plug back in from the PC's perspective.  That means the TTY device node can change.  Under Linux it can go from `/dev/ttyACM2` to `/dev/tyACM3`, or under Windows it can go from `COM5:` to COM6:`  for example.  The Mac also probably has a # after the devnode that increases, too.  So check you are looking at the right/latest TTY port after upload.

I also like to put a `sleep(5000)` after `Serial.begin()` in the sketches, since it may take several seconds for the PC to reconnect and you could lose data before that happens.

IAmOrion, 2022-08-20:
oof - it definitely seems to be playing funny buggers! I connected it to my Windows 11 machine - All I get is "COMx Port Busy".  Can't upload and can't open serial monitor.  I managed to get Thonny to see it and re-install MicroPython, I then tried to open serial monitor etc from Arduino (Closing thonny first obviously) but still no joy.  Spent hours trying to get something to happen on my Windows 11 machine with no joy.  Plugged it back into my Macbook, uploaded sketch and Serial Monitor was suddenly working! Went back to Windows machine - nothing, always com port busy.  Regardless, it's obviously nothing to do with this library etc so I'll close the issue, but hopefully if any others have had or resolved this they'll comment

closed 2022-08-20 by IAmOrion

earlephilhower, 2022-08-20:
@IAmOrion do you think you could do a WaveShare RP2040 Zero variant config for the core?  Basically make a pins_arduino.h which defines the GPIO mapping and default SPI/UART/etc. pins.

IAmOrion, 2022-08-21:
> @IAmOrion do you think you could do a WaveShare RP2040 Zero variant config for the core? Basically make a pins_arduino.h which defines the GPIO mapping and default SPI/UART/etc. pins.

I'll take a look and give it a go :)

IAmOrion, 2022-10-21:
@earlephilhower I realise I'm very late with this, but I *think* I've done this correctly - pins_arduino.h for Waveshare RP2040 Zero

[pins_arduino.zip](https://github.com/earlephilhower/arduino-pico/files/9838464/pins_arduino.zip)